mysensors/MySensors #872: chance to save a big chunk memory with signing

rclasen, 2017-06-04:
Hi,

as I keep running out of "dynamic memory" in quite a few node sketches, I took a look at whats using the memory.

```
avr-readelf -a /tmp/arduino_build_923073/node-light-tsl2951.ino.elf  | grep 'OBJ.* 3 ' |  sort -nk3
...
   360: 0080038f    33 OBJECT  GLOBAL HIDDEN     3 _msg
   433: 0080036e    33 OBJECT  GLOBAL HIDDEN     3 _msgTmp
   455: 008003b0    33 OBJECT  GLOBAL HIDDEN     3 _msgSign
   341: 00800445    42 OBJECT  GLOBAL HIDDEN     3 _ZL24_signing_verifying_n
   386: 00800416    42 OBJECT  GLOBAL HIDDEN     3 _ZL22_signing_signing_non
   406: 00800285    51 OBJECT  GLOBAL HIDDEN     3 _convBuf
   422: 008002bc   157 OBJECT  GLOBAL HIDDEN     3 Serial
   536: 008005b8   197 OBJECT  GLOBAL HIDDEN     3 _soft_sha256
   557: 0080046f   197 OBJECT  GLOBAL HIDDEN     3 _signing_sha256
```

_signing_sha256 is the Sha256Class used by the software signing backed. Ok. That's the price for not having hardware.

But if I look at _soft_sha256 in MySigning.cpp, I see it just being used for the EEPROM checksum validation... and the result is then ignored!!! by signerInit(). Wasting (more than) 197 bytes for a checksum that's ignored sounds questionable.

Actually, the checksum just seems to be verified within the SecurityPersonalizer sketch (... that's used to set the EEPROM values).

Removing _soft_sha256 (and related code) frees 229 Bytes and the node keeps working... it's just the Personalizer that needs it.

Though, just removing it sounds bad... and replacing it with some other Checksum will make the current EEPROM checksum invalid.

Are you interested in a patch/pull request to remove it? Or do you prefer a patch to defined the Sha256Class instances locally in signerValidatePersonalization() and pass the reference to signerSha256Init/-Update/-Final() (likely introduce memory issues in other situations)?

This is with the following defines:
```
#define MY_RADIO_NRF24

// signing (would use +600 bytes RAM with atsha)
#define MY_SIGNING_REQUEST_SIGNATURES
#define MY_SIGNING_GW_REQUEST_SIGNATURES_FROM_ALL
// ... but I use sw backend (+ another 100 bytes RAM)
#define MY_SIGNING_SOFT
```

fallberg, 2017-06-04:
Whitelisting use sha256 checksumming. 
I agree that ignoring the result privacy should be fixed. Without a valid personalization, signing won't work but I think there is a debug print that indicate this if result fails. So it is used in that sense.

rclasen, 2017-06-04:
Are you sure, it's used by whitelisting in a node sketch? I only found references to _soft_sha256 in signerSha256(Init|Update|Final) ... which are only used by signerValidatePersonalization (used and ignored by signerInit()) and by the Personalizer sketch.

fallberg, 2017-06-05:
Ok, that could be. You can place a pr to change it so that the backends handle the verifications if you like.
Or I fix it myself. But I cannot give a ETA on that.

fallberg, 2017-06-05:
As for the hash api itself, it should be left in there. It is a public interface for others to use. If not used, the linker should garbage collect it. The soft signing backend implements it's own instance. 
So there are two options. 
Either the public interface just pipe down to the backends to implement the generic interface. The problem with that is that the atsha204a seem to have problems with incrementally calculating sha256 hashes.
The other option is to let the backends handle the personalization validation, and keep the public hashing api in place and trust the linker to clean it up.

rclasen, 2017-06-05:
switching to a less memory-intensive checksum algorithm just for the eeprom verification likely is no option, is it? Surely this will create issues when updating a sketch... which is really going to hurt for OTA updates. Just asking as this would allow checking other parts of the eeprom without signing enabled. 

The linker in Arduino-1.8.2 in fact omits the bits when I just remove the signerValidatePersonalization() invocation.

fallberg, 2017-06-05:
I see no reason, as sha256 is used for signing to work at all. So re-using that algorithm for the checksum just makes sense. The problem here is that the instance used for that checksum is the public instance and not the one the backend provides. So switching to use the backend will save the bytes you consider wasted, until someone actually use the public API.

fallberg, 2017-06-05:
I made pr #873 to address the issue. Please test with that PR to verify the problem is solved (and also that checksumming still works for the software backend).

rclasen, 2017-06-05:
Thanks for the fix. Works as expected - memory footprint, checksum verification and message signing all look good.

So, I'm now curious to see if/when the public API gets used elsewhere and gets into my way, again ;)

fallberg, 2017-06-06:
It will most likely happen when proper SHA256 validation of OTA firmware is implemented.

fallberg, 2017-06-06:
Thanks for verifying. I will submit the PR once it passes all validations.

PR #873 (2017-06-06): Let signing backend validate personalization
For soft signing, let backend handle checksum verification of
personalization data. This ensures that the backends sha256 library is
used and avoids an unnecessary instanitation of the public library.
For atsha204 signing, no checksum verification is made, just chip
locking flags.

Fixes #872
commit 29fd6d63 (2017-06-06): Let signing backend validate personalization (#873)

For soft signing, let backend handle checksum verification of
personalization data. This ensures that the backends sha256 library is
used and avoids an unnecessary instanitation of the public library.
For atsha204 signing, no checksum verification is made, just chip
locking flags.

Fixes #872
closed 2017-06-06 by fallberg