mysensors/MySensors #876: Offline loop

blacksheepinc, 2017-06-07:
Hello.
I have an ESP8266 running mysensors, which has relays connected to it and switches.
It works both with manually with the switches or through mycontroller with MQTT.
However if there's a failure, router stops working or the MQTT or the mycontroller I would like to be able to use the ESP manually to switch the relays.
However in case of disconnect the loop is frozen and only reconnect is running.

Please advise!
Thanks

mfalkvidd, 2017-06-10:
See
* @def MY_TRANSPORT_WAIT_READY_MS
* @brief Timeout in MS until transport is ready during startup, set to 0 for no timeout

https://github.com/mysensors/MySensors/blob/master/MyConfig.h#L175

blacksheepinc, 2017-06-11:
Thank you I tried, but it still restarts (panic) then it just stuck at the wifi reconnect loop.
Here is the log:
`⸮Hd⸮⸮⸮$0x⸮lb8⸮⸮⸮0;255;3;0;9;MCO:BGN:INIT GW,CP=R-NGE--,VER=2.1.1
sta config unchangedscandone
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
.
connected with Xiaomi, channel 1
ip:192.168.1.202,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 0
wifi evt: 3
.IP: 192.168.1.202
0;255;3;0;9;MCO:BGN:STP
Booting
OTA server at: esp8266-30de20.local:8266
Ready
0;255;3;0;9;MCO:REG:NOT NEEDED
0;255;3;0;9;MCO:BGN:INIT OK,TSP=NA
IP: 192.168.1.202
0;255;3;0;9;Attempting MQTT connection...
:ref 1
:wr
:sent 22
:rn 4
:ww
:c0 1, 4
0;255;3;0;9;MQTT connected
0;255;3;0;9;Sending message on topic: mygateway-out/0/255/0/0/17
:wr
:sent 35
:ww
0;255;3;0;9;Sending message on topic: mygateway-out/0/255/3/0/11
:wr
:sent 54
:ww
0;255;3;0;9;Sending message on topic: mygateway-out/0/255/3/0/12
:wr
:sent 33
:ww
0;255;3;0;9;Sending message on topic: mygateway-out/0/0/0/0/3
:wr
:sent 27
:ww
0;255;3;0;9;Sending message on topic: mygateway-out/0/1/0/0/3
:wr
:sent 27
:ww
:wr
:sent 29
:rn 5
:ww
Request old state
0;255;3;0;9;Sending message on topic: mygateway-out/0/0/2/0/2
:wr
:sent 27
:ww
:c0 1, 5
0;255;3;0;9;Sending message on topic: mygateway-out/0/1/2/0/2
:wr
:sent 27
:ww
pm open,type:2 0
:wr
:sent 2
:rn 2
:ww
:c0 1, 2
bcn_timout,ap_probe_send_start  /////Here it lost the connection to the router
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
f r0, wifi evt: 1
STA disconnect: 200
:ur 1
:close
:del
:ur 1

Panic C:\Users\M⸮rk\Documents\Arduino\libraries\MySensors/core/MyMainESP8266.cpp:132 loop_task

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fffff80:  4021f572 3fffdab0 00000000 3ffeefa0  
3fffff90:  00000000 3fffdad0 3ffeef98 402023a1  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
⸮e)⸮L⸮⸮⸮) ?0;255;3;0;9;MCO:BGN:INIT GW,CP=R-NGE--,VER=2.1.1
sta config unchangedscandone
f 0, scandone
no Xiaomi found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
f -180, ....scandone
no Xiaomi found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
f r0, .....scandone
no Xiaomi found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
f 0, ....scandone
no Xiaomi found, reconnect after 1s`

closed 2017-09-21 by henrikekblad