mysensors/MySensors #878: OTA for GateWay

Shagrat2, 2017-06-12:
OTA for gateway not supported
https://www.openhardware.io/view/118/MDMSGate

tekka007, 2017-06-12:
@Shagrat2 Can you be a bit more specific?

Shagrat2, 2017-06-16:
I have gateway https://www.openhardware.io/view/118/MDMSGate
with Flash AT25DF512C and bootlaoder DualOptiBoot 
Set #define MY_OTA_FIRMWARE_FEATURE
But library not work with OTA on NodeID=0

matthew73210, 2017-06-23:
As far as I know, you can't OTA the gateway via that method, only nodes. Why would you want to as the gateway sketch is dead simple?

fallberg, 2017-06-23:
Who would OTA program firmware to the GW? OTA means over the air, so it implies that your controller would be radio capable. And if it is, why would you need a gateway?
It simply does not compute to have OTA capability on a gateway "by definition". It is the gateway that transmit OTA firmware to your nodes. The gateway receives the firmware through a wired protocol from the controller. Either USB or ethernet (or wifi, which correspond to ethernet).

closed 2017-07-06 by tekka007