mysensors/MySensors #886: Need help figuring out what to troubleshoot next...

BogusException, 2017-07-03:
Experts,

MyController
MySensors - stock except changed ch from 76 to 100
UNO R3 board w/DHT-11, code for voltage, but pin open.

log is up top, and sketch is way at bottom... Hope that is ok.,..

I could use a little help with an idea or 2 on what to test next. I've lost the 'musical components' game, so I can't swap anything else out! :)

Different UNO boards, different RF24s... I guess I need help narrowing this down. I have pro-minis & nanos working (as repeaters), and the auto-naming stuff aside, they are behaving. I went to add a UNO, and I have a debug log I can't understand:

(_note: I can't flag this text (log; below) as 'code', as this embedded editor strips out the newlines_)

0 MCO:BGN:INIT REPEATER,CP=RNNRA--,VER=2.1.1
3 TSM:INIT
4 TSF:WUR:MS=0
6 RF24:write register, reg=0, value=14
14 RF24:write register, reg=3, value=3
17 RF24:write register, reg=4, value=95
21 RF24:write register, reg=5, value=100
25 RF24:write register, reg=6, value=39
28 RF24:write register, reg=16, value=115
32 RF24:write register, reg=29, value=6
35 RF24:read register, reg=6, value=0
39 RF24:read register, reg=5, value=0
43 !RF24:INIT:SANCHK FAIL
45 !TSM:INIT:TSP FAIL
47 TSM:FAIL:CNT=1
49 TSM:FAIL:PDT
50 RF24:write register, reg=0, value=12
54 RF24:PD
10056 TSM:FAIL:RE-INIT
10058 TSM:INIT
10059 RF24:write register, reg=0, value=14
10068 RF24:write register, reg=3, value=3
10073 RF24:write register, reg=4, value=95
10076 RF24:write register, reg=5, value=100
10080 RF24:write register, reg=6, value=39
10084 RF24:write register, reg=16, value=115
10088 RF24:write register, reg=29, value=6
10092 RF24:read register, reg=6, value=0
10095 RF24:read register, reg=5, value=0
10099 !RF24:INIT:SANCHK FAIL
10102 !TSM:INIT:TSP FAIL
10104 TSM:FAIL:CNT=2
10106 TSM:FAIL:PDT
10108 RF24:write register, reg=0, value=12
10112 RF24:PD
20115 TSM:FAIL:RE-INIT
20117 TSM:INIT
20118 RF24:write register, reg=0, value=14
20127 RF24:write register, reg=3, value=3
20131 RF24:write register, reg=4, value=95
20134 RF24:write register, reg=5, value=100
20139 RF24:write register, reg=6, value=39
20143 RF24:write register, reg=16, value=115
20147 RF24:write register, reg=29, value=6
20151 RF24:read register, reg=6, value=0
20154 RF24:read register, reg=5, value=0
20158 !RF24:INIT:SANCHK FAIL
20161 !TSM:INIT:TSP FAIL
20163 TSM:FAIL:CNT=3
20165 TSM:FAIL:PDT
20167 RF24:write register, reg=0, value=12
20170 RF24:PD
30173 TSM:FAIL:RE-INIT
30175 TSM:INIT
30176 RF24:write register, reg=0, value=14
30185 RF24:write register, reg=3, value=3
[... 207 lines cut ...]
    // 3.44/1023 = Volts per bit = 0.003363075

    int batteryPcnt = sensorValue / 10;

#ifdef MY_DEBUG
    float batteryV  = sensorValue * 0.003363075;
    Serial.print("Battery Voltage: ");
    Serial.print(batteryV);
    Serial.println(" V");

    Serial.print("Battery percent: ");
    Serial.print(batteryPcnt);
    Serial.println(" %");
#endif

    if (oldBatteryPcnt != batteryPcnt) {
        // Power up radio after sleep
        sendBatteryLevel(batteryPcnt);
        oldBatteryPcnt = batteryPcnt;
    }

  // Sleep for a while to save energy
  sleep(UPDATE_INTERVAL); 
}
`

tekka007, 2017-07-03:
@BogusException Please also describe your radio wiring.

BogusException, 2017-07-04:
Thanks for writing, @tekka007 

[Here](https://lh3.googleusercontent.com/CMkRnMrGegjzeC8iVa_9JFGqg3r4gWH6wufzWd30rAhS9ynaoYXCDamRnpyfHp-KUy5JmtG9ULpo2Ms-aHtJ9rwYitjrfrEonOLuGXMCztL4GV2ATDXjLLSTnAatSQzfC0DZIv0KzT3mcgebYF3MKIRAL0RP2aguVFeOa8cT5PysSFk9Smc_o5PVPHsPpCPGhfUJwWD7KM_0RbxluZ__c5YLu9b-Wr5cfx8SVyQfeiqkxUQ-lKCoGBbieTzubVL4jrTjS4--RdnlSZhJjlROWSla7OPAIx0OfmgvzdEVO83w5-YIS9SRo6toPJoGtWF1v_dLxPV0aHgxx1gvmdvhVptfIlsmMwsM6bqp5uvW5dmnMbdKkx7I36HTiKNhyDEXWIsf7ddXS1r0LqK4Yj5mZIFE9pcF_vvmNtJw46QPRBRc5GDuLnpisx8o34WDbndUrWzsdHK_-kZd_4WuuRdUIb40cWTMHRvuWPwdWsQ6ZQBrsMwfgJKvDRe9AUhI4DHqPIVQwyb4n39-ReYpNGjwkU8JB069PajFJX4jN8SpyiQxoJsv1MC_3uTUbKm72JqyMnd_ZpWk1HEv8OvUWSgUfHXlNYvNyqbUbhC3mP_VQvKvrl0VeoulhK6_4ZJClgPs-Z5nmaebn1oS3L_fp9B6y6hi5B-E9d_BIY9gZhBfgvA=w1299-h731-no) is a picture as best I could, and at least all the wires aren't red! :) There is 3.3v at the nRF24 board.

I'll try this as well (see picture at link above):

```
UNO 	NRF24L01+
GND 	GND 	Black
3.3V 	VCC		Red
9 		CE		Orange
10 		CSN/CS	Yellow
13 		SCK		Purple
11 		MOSI	Blue
12 		MISO	Green
2 		IRQ		White

DHT-11
Gnd		Black
Vcc (3,3)	Red
Signal	Blue

(10uf across breadboard power rail)
```

I added to the config (used in compile below) these lines after seeing them in a search (_before the MySensors.h include_):

```
#define MY_RF24_CE_PIN 9
#define MY_RF24_CS_PIN 10
```

Also, I guess the compile log should have been included:

```
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Pat\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf"

         Using Port                    : COM15
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\Pat\AppData\Local\Temp\arduino_build_947523/UNO201_RepeaterNode_DHT11_Batt_v2.ino.hex"
avrdude: writing flash (18370 bytes):

Writing | ################################################## | 100% 2.62s

avrdude: 18370 bytes of flash written
avrdude: verifying flash memory against C:\Users\Pat\AppData\Local\Temp\arduino_build_947523/UNO201_RepeaterNode_DHT11_Batt_v2.ino.hex:
avrdude: load data flash data from input file C:\Users\Pat\AppData\Local\Temp\arduino_build_947523/UNO201_RepeaterNode_DHT11_Batt_v2.ino.hex:
avrdude: input file C:\Users\Pat\AppData\Local\Temp\arduino_build_947523/UNO201_RepeaterNode_DHT11_Batt_v2.ino.hex contains 18370 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 1.97s

avrdude: verifying ...
avrdude: 18370 bytes of flash verified

avrdude done.  Thank you.
```

tekka007, 2017-07-06:
@BogusException Everything looks fine to me, however, the logs clearily show a SPI communicaton and/or radio issue. A few things to test:
1. Power issue: Add/Exchange cap on radio, use at least 22uF
2. Noisy power supply: Exchange power supply, i.e. to a battery
3. Does the radio + cable work on a different MCU? 
4. Is the SPI port of your UNO board ok, e.g. using a SPI sensor?

BogusException, 2017-07-09:
Thank you for the awesome tips! I'll be sure to circle back w/results.

Thanks for writing, good Sir!

cimba007, 2017-07-11:
@BogusException: Have you connected the RF24 and the DHT-11 DataPin to the same pin? (blue)? If so try choosing a different pin fot the DHT-11. If I remember correctly I had a similar problem as both librarys initiate the pin in a different way and are not compatible with each other.

To narrow down this try to connect only the RF24 to your board.

closed 2017-09-21 by henrikekblad