mysensors/MySensors #891: Teensy with RFM95 does not go to sleep

micropet, 2017-07-11:
The command sleep (xx) does not work with the Teensy LC 3.2 / 3.6.
It is simply ignored.

trlafleur, 2017-07-11:
This is the same issues with the M0 processors.   
Sleep, WDT are not yet implemented.....

micropet, 2017-07-11:
OK thanks.
So that means wait ...

The Arduino Pro has too little memory and it would be nice to upgrade to more powerful CPUs (ESP32, SAMD21, Teensy).

trlafleur, 2017-07-11:
No, you can just implement the low power mode in your code.....

See code from rocketscream and others....  do a quick search....

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

micropet, 2017-07-12:
Yes Tom , I did. At the Teensys it goes with Snooze.h.

But it would be better if the basic functions are in the MySensors lib.

Peter

trlafleur, 2017-07-12:
Yes I agree , but the primary developer are still in the 8 bit world.   So it's do it yourself or wait.    Zero, M0 was added well over a year ago and still no direct support for sleep or WDT support.  

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

scalz, 2017-07-14:
@trlafleur said "the primary developer are still in the 8 bit world. So it's do it yourself or wait. Zero, M0 was added well over a year ago and still no direct support for sleep or WDT support."

Anthological! Hello !??

Do you consider yourself as a 32bits dev? because of using some libs with atsam??
why don't you help, instead of complains?? I don't understand this kind of frustration..
We have not all same priorities, simple ;)

This is not kind to behave like this whereas Core Team is using lot of his free time to feed users for free..

@micropet  Regarding this issue, i'll (or someone else in Core Team) take a look later, as it's not in my top priorities for the moment, unfortunately, i'm sorry.

micropet, 2017-07-14:
@scalz OK, thank you for your efforts.

trlafleur, 2017-07-14:
It was NOT my intent to complain, but to clarified that the support for some processors are missing some features. Nothing more. 

I do understand that the developers do have priority in their life's and on this project... again, it was NOT my intent to criticise their effort...  I greatly appreciate the effort of the developers, its a great product and has a great future.

Thank you for your effort.

d00616, 2017-08-26:
The sleep mode and WDT are implemented for nRF51(M0) and nRF52(M4) MCUs. Adding these functionality for all Cortex CPUs isn't simple because there are a lot of differences between Cortex-M MCUs.

Often the Arduino Ports are at an early stage like the arduino-nRF5 missing some functionality like WDT, but things becoming more and more stable.