mysensors/MySensors #893: RPI Gateway loses connection to the mqtt server

micropet, 2017-07-18:
If the network connection is interrupted, the connection to the mqtt server is interrupted and is no longer established.

Systemctl restart mysgw.service is required

The gateway must always be checked for safety-relevant applications such as Pir or a heating control.

`Jul 18 14:57:15 pi143 mysgw[7283]: connect: Connection refused
Jul 18 14:57:15 pi143 mysgw[7283]: failed to connect
Jul 18 14:57:15 pi143 mysgw[7283]: connect: Connection refused
Jul 18 14:57:15 pi143 mysgw[7283]: failed to connect
Jul 18 14:57:15 pi143 mysgw[7283]: connect: Connection refused
Jul 18 14:57:15 pi143 mysgw[7283]: failed to connect
Jul 18 14:57:15 pi143 mysgw[7283]: connect: Connection refused
Jul 18 14:57:15 pi143 mysgw[7283]: failed to connect
Jul 18 14:57:15 pi143 mysgw[7283]: connect: Connection refused
Jul 18 14:57:15 pi143 mysgw[7283]: failed to connect
`

mfalkvidd, 2017-07-18:
@micropet could you clarify in what way this is different from https://github.com/mysensors/MySensors/issues/788 ?

micropet, 2017-07-19:
Sorry, # 788 I had already forgotten.

The problem is worse than I had assumed.
The mqtt connection is also interrupted when the network connection is stable.

The ESP8266 problem from # 788 seems to be something else.
This is the WDT RESET which occurs again and again with the ESP8266 Chip when used the WLAN connection.

marceloaqno, 2018-03-24:
@micropet Could you please run a new test with the latest version of the development branch?
I couldn't replicate this error, [#1091](https://github.com/mysensors/MySensors/pull/1091) may have fixed it.

I used the following for the test:
- Raspberry Pi 1 with Raspbian (2017-11-29-raspbian-stretch-lite)
- ```./configure --my-transport=nrf24 --my-gateway=mqtt --my-controller-ip-address=192.168.7.12 --my-mqtt-publish-topic-prefix=mysensors-out --my-mqtt-subscribe-topic-prefix=mysensors-in --my-mqtt-client-id=mygatewa y01```
- Latest MySensors development branch (ea6d07c)

I unplugged the ethernet cable, waited 5 min and plugged it again. The gateway was able to reconnect to the mqtt broker:
```
Mar 24 15:19:00 DEBUG GWT:RMQ:MQTT RECONNECT
Mar 24 15:19:00 ERROR connect: Network is unreachable
Mar 24 15:19:00 ERROR failed to connect
Mar 24 15:19:00 DEBUG GWT:RMQ:MQTT RECONNECT
Mar 24 15:19:00 ERROR connect: Network is unreachable
Mar 24 15:19:00 ERROR failed to connect
Mar 24 15:19:00 DEBUG GWT:RMQ:MQTT RECONNECT
Mar 24 15:19:00 ERROR connect: Network is unreachable
Mar 24 15:19:00 ERROR failed to connect
Mar 24 15:19:00 DEBUG GWT:RMQ:MQTT RECONNECT
Mar 24 15:19:00 DEBUG connected to 192.168.7.12
Mar 24 15:19:00 DEBUG GWT:RMQ:MQTT CONNECTED
Mar 24 15:19:00 DEBUG GWT:TPS:TOPIC=mysensors-out/0/255/0/0/18,MSG SENT
Mar 24 15:19:00 DEBUG TSF:MSG:READ,5-5-0,s=255,c=3,t=33,pt=5,l=4,sg=0:5000
Mar 24 15:19:00 DEBUG GWT:TPS:TOPIC=mysensors-out/5/255/3/0/33,MSG SENT
Mar 24 15:19:00 DEBUG TSF:MSG:READ,5-5-0,s=1,c=1,t=37,pt=4,l=4,sg=0:102989
Mar 24 15:19:00 DEBUG GWT:TPS:TOPIC=mysensors-out/5/1/1/0/37,MSG SENT
Mar 24 15:19:00 DEBUG TSF:MSG:READ,5-5-0,s=2,c=1,t=0,pt=7,l=5,sg=0:28.1
Mar 24 15:19:00 DEBUG GWT:TPS:TOPIC=mysensors-out/5/2/1/0/0,MSG SENT
Mar 24 15:19:00 DEBUG TSF:MSG:READ,5-5-0,s=3,c=1,t=1,pt=7,l=5,sg=0:70.9
Mar 24 15:19:00 DEBUG GWT:TPS:TOPIC=mysensors-out/5/3/1/0/1,MSG SENT
```

Yveaux, 2019-01-05:
https://github.com/mysensors/MySensors/issues/788#issuecomment-430098469 mentions failure to reconnect on 2.3.1-alpha