mysensors/MySensors #894: TSM:FAIL:CNT gets stuck at 7

mfalkvidd, 2017-07-23:
When a node (using 2.2.0-beta / 3e75b52c) is unable to establish contact with the gateway, the failure count gets stuck at 7 (so 1-2-3-4-5-6-7-7-7-7...)

`
0 MCO:BGN:INIT NODE,CP=RRNNA---,VER=2.2.0-beta
4 TSM:INIT
4 TSF:WUR:MS=0
8 TSM:INIT:TSP OK
10 TSM:INIT:STATID=1
12 TSF:SID:OK,ID=1
14 TSM:FPAR
1255 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
3262 !TSM:FPAR:NO REPLY
3264 TSM:FPAR
4544 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
6551 !TSM:FPAR:NO REPLY
6553 TSM:FPAR
7948 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
9955 !TSM:FPAR:NO REPLY
9957 TSM:FPAR
11249 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
13256 !TSM:FPAR:FAIL
13258 TSM:FAIL:CNT=1
13260 TSM:FAIL:DIS
13262 TSF:TDI:TSL
23265 TSM:FAIL:RE-INIT
23267 TSM:INIT
23271 TSM:INIT:TSP OK
23273 TSM:INIT:STATID=1
23275 TSF:SID:OK,ID=1
23277 TSM:FPAR
24709 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
26716 !TSM:FPAR:NO REPLY
26718 TSM:FPAR
28039 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
30046 !TSM:FPAR:NO REPLY
30048 TSM:FPAR
31348 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
33355 !TSM:FPAR:NO REPLY
33357 TSM:FPAR
34625 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
36632 !TSM:FPAR:FAIL
36634 TSM:FAIL:CNT=2
36636 TSM:FAIL:DIS
36638 TSF:TDI:TSL
46641 TSM:FAIL:RE-INIT
46643 TSM:INIT
46647 TSM:INIT:TSP OK
46649 TSM:INIT:STATID=1
46651 TSF:SID:OK,ID=1
46653 TSM:FPAR
47925 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
49932 !TSM:FPAR:NO REPLY
49934 TSM:FPAR
51238 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
53248 !TSM:FPAR:NO REPLY
53250 TSM:FPAR
54503 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
56510 !TSM:FPAR:NO REPLY
56512 TSM:FPAR
57774 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
59781 !TSM:FPAR:FAIL
59783 TSM:FAIL:CNT=3
59785 TSM:FAIL:DIS
59787 TSF:TDI:TSL
69791 TSM:FAIL:RE-INIT
69793 TSM:INIT
69797 TSM:INIT:TSP OK
69799 TSM:INIT:STATID=1
69801 TSF:SID:OK,ID=1
69804 TSM:FPAR
71086 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
73093 !TSM:FPAR:NO REPLY
73095 TSM:FPAR
74387 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
76394 !TSM:FPAR:NO REPLY
76396 TSM:FPAR
77690 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
79699 !TSM:FPAR:NO REPLY
79702 TSM:FPAR
81031 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
83038 !TSM:FPAR:FAIL
83040 TSM:FAIL:CNT=4
83042 TSM:FAIL:DIS
83044 TSF:TDI:TSL
93048 TSM:FAIL:RE-INIT
93050 TSM:INIT
93054 TSM:INIT:TSP OK
93057 TSM:INIT:STATID=1
93059 TSF:SID:OK,ID=1
93061 TSM:FPAR
94316 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
96323 !TSM:FPAR:NO REPLY
96325 TSM:FPAR
97583 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
99590 !TSM:FPAR:NO REPLY
99592 TSM:FPAR
100835 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
102842 !TSM:FPAR:NO REPLY
102844 TSM:FPAR
104128 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
106135 !TSM:FPAR:FAIL
106137 TSM:FAIL:CNT=5
106139 TSM:FAIL:DIS
106141 TSF:TDI:TSL
116144 TSM:FAIL:RE-INIT
116146 TSM:INIT
116150 TSM:INIT:TSP OK
116152 TSM:INIT:STATID=1
116154 TSF:SID:OK,ID=1
116158 TSM:FPAR
117446 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
119453 !TSM:FPAR:NO REPLY
119455 TSM:FPAR
120717 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
122724 !TSM:FPAR:NO REPLY
122726 TSM:FPAR
124037 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
126044 !TSM:FPAR:NO REPLY
126046 TSM:FPAR
127330 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
129337 !TSM:FPAR:FAIL
129339 TSM:FAIL:CNT=6
129341 TSM:FAIL:DIS
129343 TSF:TDI:TSL
139347 TSM:FAIL:RE-INIT
139350 TSM:INIT
139354 TSM:INIT:TSP OK
139356 TSM:INIT:STATID=1
139358 TSF:SID:OK,ID=1
139362 TSM:FPAR
140648 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
142655 !TSM:FPAR:NO REPLY
142657 TSM:FPAR
143925 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
145932 !TSM:FPAR:NO REPLY
145934 TSM:FPAR
147251 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
149258 !TSM:FPAR:NO REPLY
149260 TSM:FPAR
150587 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
152594 !TSM:FPAR:FAIL
152596 TSM:FAIL:CNT=7
152598 TSM:FAIL:DIS
152600 TSF:TDI:TSL
212604 TSM:FAIL:RE-INIT
212606 TSM:INIT
212611 TSM:INIT:TSP OK
212613 TSM:INIT:STATID=1
212615 TSF:SID:OK,ID=1
212619 TSM:FPAR
213964 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
215971 !TSM:FPAR:NO REPLY
215973 TSM:FPAR
217280 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
219287 !TSM:FPAR:NO REPLY
219289 TSM:FPAR
220588 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
222595 !TSM:FPAR:NO REPLY
222597 TSM:FPAR
223911 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
225921 !TSM:FPAR:FAIL
225923 TSM:FAIL:CNT=7
225925 TSM:FAIL:DIS
225927 TSF:TDI:TSL
285931 TSM:FAIL:RE-INIT
285933 TSM:INIT
285937 TSM:INIT:TSP OK
285939 TSM:INIT:STATID=1
285941 TSF:SID:OK,ID=1
285945 TSM:FPAR
287328 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
289335 !TSM:FPAR:NO REPLY
289337 TSM:FPAR
290641 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
292651 !TSM:FPAR:NO REPLY
292653 TSM:FPAR
293986 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
295993 !TSM:FPAR:NO REPLY
295995 TSM:FPAR
297279 TSF:MSG:SEND,1-1-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
299286 !TSM:FPAR:FAIL
299288 TSM:FAIL:CNT=7
299290 TSM:FAIL:DIS
299292 TSF:TDI:TSL
`

mfalkvidd, 2017-07-23:
Maybe this is by design, since
`#define MY_TRANSPORT_MAX_TSM_FAILURES                (7u)            //!< Max. number of consecutive TSM failure state entries (3bits)`
but if that is the max number of failures, why does the node keep trying?

tekka007, 2017-07-23:
@mfalkvidd correct - the TSM goes then into extended re-init delay:
https://github.com/mysensors/MySensors/blob/development/core/MyTransport.cpp#L353, the timeout is  defined via `MY_TRANSPORT_TIMEOUT_EXT_FAILURE_STATE_MS` (default is 1min) and can be overridden.

mfalkvidd, 2017-07-24:
OK. I find it confusing that the counter does not increase, but since it is by design I'll just close this.

closed 2017-07-24 by mfalkvidd