mysensors/MySensors #902: Document capabilities in MCO:BGN:INIT CP=RNNG--Q-

mfalkvidd, 2017-07-31:
The listing of capabilities can be very useful and save several round-trips in the forum asking for more information. But I don't know how to interpret them :(

Would be good if they were documented (in Doxygen?) and added to the log parser.

mfalkvidd, 2017-09-22:
I don't know how to properly document the capabilities in Doxygen so I created this in the meanwhile: https://docs.google.com/spreadsheets/d/1fsJyOGhNGL9IeZMMSuVwluUpo9he43wcWjM_w6xTL5k/edit#gid=0

mysensors-jenkins, 2017-09-22:
Nice. I looked at documenting it briefly but quickly realized that it needs some thought and was to tired at the time. I will revisit it once I got jenkins tamed. But a table is probably the most straightforward way.

fallberg, 2017-09-22:
Nice. I looked at documenting it briefly but quickly realized that it needs some thought and was to tired at the time. I will revisit it once I got jenkins tamed. But a table is probably the most straightforward way.

commit 1f8d810f (2017-11-13): Add documentation for MyCapabilities (#989)

Also fixed spelling error in the MySigning documentation.

Fixes #902
closed 2017-11-13 by mfalkvidd