mysensors/MySensors #903: Document release process

mfalkvidd, 2017-07-31:
Documenting the release process would make it easier to be consistent, doing the same thing every time, and allowing anyone with the right privileges to create a release. It would also be easier to automate parts of the release process if it was documented, and it is easier to suggest improvements to the release process if people know what the release process looks like.

Below is what I have gathered so far, with great help from tbowmo and anticimex. It does not handle managing the 29 file merge conflict we currently have between master and development.

> 0. Clone MySensors if you haven't already
> git clone -o upstream git@github.com:mysensors/MySensors.git
> 1. Make sure you have the latest code
> git fetch upstream
> 2. git checkout remotes/upstream/master
> 3. git merge remotes/upstream/development
> 4. change relevant header files (for versioning)
> new_release=2.1.2CHANGEME
> old_release=2.1.1CHANGEME
> sed -i "s/$old_release/$new_release/" core/Version.h library.json library.properties README.md
> 5. git commit -am "MySensors $release release"
> 6. git push
> 7. Wait for Jenkins to be happy. Then create and push the release tag
> git tag -a $new_release
> git push origin $new_release
> 8. Create a new release at Github TODO: document this

The development branch will need to be updated afterwards as well. Something like this I think:
Update the development branch

> 1. git checkout remotes/upstream/development
> 2. change relevant header files
> new_devrelease=2.2.1-beta
> old_devrelease=2.2.0-beta
> sed -i "s/$old_devrelease/$new_devrelease/" core/Version.h library.json library.properties README.md
> 3. git commit -am "MySensors $release branch"
> 4. git push

fallberg, 2017-08-01:
I believe the line
`git commit -am "MySensors $release release"`
should be
`git commit -am "MySensors $new_release release"`
?

fallberg, 2017-08-01:
And fopr the development branch update
`git commit -am "MySensors $release branch"`
should be
`git commit -am "MySensors $new_devrelease branch"`

fallberg, 2017-08-01:
And probably the merge command from development to master could use arguments to enforce development overwrite master in case of conflicts (don't know those argument by heart though).

BogusException, 2017-08-02:
@fallberg At a few companies we've used the act/stage/term 'code freeze' in our release process. It may well be implied, but is the concept part of the way you guys release major/minor versions and/or periodic fix releases?

fallberg, 2017-08-03:
@BogusException This only document the technical aspects of the release process. Not the frequency.

mfalkvidd, 2018-01-25:
Release process is documented at https://github.com/mysensors/MySensors/wiki/Library-releases
It might need tweaking in the future, but let's open specific issues when we find out what is needed. Closing issue.

closed 2018-01-25 by mfalkvidd