mysensors/MySensors #918: GatewayESP8266 does not compile with MY_CONTROLLER_IP_ADDRESS

mfalkvidd, 2017-09-02:
On 2.1.1, using the vanilla gatewayESP8266 sketch and uncommenting
```
#define MY_CONTROLLER_IP_ADDRESS 192, 168, 178, 68
```
the sketch does not compile. This is the result:
```
"C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IR:\Scratch\ArduinoBuild/core" -c  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10608 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" "-IC:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" "-IR:\Documents\Arduino\libraries\MySensors" "-IC:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\EEPROM" "-IC:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI" "R:\Scratch\ArduinoBuild\sketch\GatewayESP8266.ino.cpp" -o "R:\Scratch\ArduinoBuild\sketch\GatewayESP8266.ino.cpp.o"
In file included from C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39:0,

                 from C:\Users\Micke\AppData\Local\Temp\arduino_modified_sketch_790718\GatewayESP8266.ino:132:

C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.h: In instantiation of 'size_t WiFiClient::write(T&, size_t) [with T = char*; size_t = unsigned int]':

R:\Documents\Arduino\libraries\MySensors/core/MyGatewayTransportEthernet.cpp:207:58:   required from here

C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.h:123:36: error: request for member 'available' in 'source', which is of non-class type 'char*'

     size_t left = source.available();

                                    ^

C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.h:127:5: error: request for member 'read' in 'source', which is of non-class type 'char*'

     source.read(buffer.get(), will_send);

     ^

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi 
Using library MySensors at version 2.1.1 in folder: R:\Documents\Arduino\libraries\MySensors 
Using library EEPROM at version 1.0 in folder: C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\EEPROM 
Using library SPI at version 1.0 in folder: C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```
This problem was reported by  Djames Suhanko in https://forum.mysensors.org/topic/7421/doubt-about-a-concept-in-gatewayesp8266

tmsd2001, 2017-09-03:
Here the same, Win10, Arduino IDE 1.8.4, Mysensors 2.1.1, ESP lib 2.3.0, NodeMPU 0.9

mfalkvidd, 2018-02-18:
I am unable to reproduce this problem with ESP8266 board definition version 2.4.0. Tried with development and 2.1.1. Libraries used:
```
Using library ESP8266WiFi at version 1.0
Using library MySensors at version 2.1.1
Using library EEPROM at version 1.0
Using library SPI at version 1.0
```

However, if I downgrade to ESP8266 board version 2.3.0 the error occurs. So it looks like the problem was in the ESP8266 board files, and has been fixed now.

Therefore, I'm closing this issue. Feel free to reopen if you have a different view.

closed 2018-02-18 by mfalkvidd