mysensors/MySensors #926: DNS serv

Eagle2373, 2017-09-11:
Currently it is not possible to specify the DNS server within MySensors and the ESP8266 falls back to the default DNS server (AFAIK opendns.com) as configured within the firmware.

The situation that I noticed is that opendns doesn't always resolve the domain name of my public webserver which I use to sent measure data to. By adding the DNS server in the WiFi.config call did resolve the issue for me. Could the DNS server be added as a #define and be added to the WiFi.config call?

mfalkvidd, 2018-02-18:
If I'm reading [the code](https://github.com/mysensors/MySensors/blob/af3452f91c15b60cc7666b6225fa7342f08a6f52/core/MyGatewayTransportEthernet.cpp#L126) correctly, MySensors will always use MY_IP_GATEWAY_ADDRESS as DNS server, if MY_IP_ADDRESS is defined.

The second parameter to WiFi-config is the dns server address according to https://www.arduino.cc/en/Reference/WiFiConfig

I'll add a pull request for this feature.

mfalkvidd, 2018-02-20:
Further test results. When using DHCP:
```
connected with Mr-IoT.com, channel 1
dhcp client start...
wifi evt: 0
ip:192.168.1.215,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 3
6142 GWT:TPC:CONNECTING...
6144 GWT:TPC:IP=192.168.1.215
6147 MCO:BGN:STP
6148 MCO:REG:NOT NEEDED
6151 MCO:BGN:INIT OK,TSP=NA
6153 GWT:TPC:IP=192.168.1.215
6156 GWT:RMQ:MQTT RECONNECT
[hostByName] request IP for: test.mosquitto.org
[hostByName] Host: test.mosquitto.org IP: 37.187.106.16
6266 GWT:RMQ:MQTT CONNECTED
6269 GWT:TPS:TOPIC=mygateway1-out/0/255/0/0/17,MSG SENT
```

with static IP, regardless of setting MY_DNS_ADDRESS:
```
connected with Mr-IoT.com, channel 1
ip:192.168.1.188,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.188,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 0
wifi evt: 3
6138 GWT:TPC:CONNECTING...
6140 GWT:TPC:IP=192.168.1.188
6143 MCO:BGN:STP
6145 MCO:REG:NOT NEEDED
6147 MCO:BGN:INIT OK,TSP=NA
6149 GWT:TPC:IP=192.168.1.188
6152 GWT:RMQ:MQTT RECONNECT
[hostByName] request IP for: test.mosquitto.org
[hostByName] Host: test.mosquitto.org lookup error: -6!
```

So it seems dns resolution does not work at all when using static configuration. I will need to do more digging.
Above output was generated by setting "Debug level" to "Wifi" on the Tools menu in the Arduino IDE and setting "Debug port" to "serial".

Eagle2373, 2018-02-20:
Thx for picking this up. It's weird as I incorporated AFAIK the DNS in the WiFi.config call in MyGatewayTransportEthernet.cpp. The only thing is that I can't check it not earlier then coming weekend.
I sniffed a bit in the ESP8266 board code  (version 2.3.0) and saw that in the file libraries/ESP8266Wifi/src/ESP8266WiFiSTA.h the following method declaration: 
bool config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1 = (uint32_t)0x00000000, IPAddress dns2 = (uint32_t)0x00000000);

Not sure how the compiler will resolve this if you provide a DNS address to the WiFi.config method...

mfalkvidd, 2018-02-20:
@Eagle2373 the ESp8266 config call uses some heuristics to reorder the parameters, since ESP8266 wifi config function uses a different order of parameters than the Arduino wifi config function. Reference: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h#L42

This makes it much harder to follow the code, which unfortunately has added a lot of troubleshooting time. My suspicion is that something goes wrong when trying to juggle around the parameters. I am not sure when I'll be able to pick this up again.

mfalkvidd, 2018-03-10:
I got it working! Took a few hours, and I need to clean up the code, but it works:
```
connected with Mr-IoT.com, channel 1
ip:192.168.1.188,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.188,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 0
wifi evt: 3
6210 GWT:TPC:CONNECTING...
6238 GWT:TPC:IP=192.168.1.188
6270 GWT:TPC:DNS=9.9.9.9
6296 MCO:BGN:STP
6314 MCO:REG:NOT NEEDED
6339 MCO:BGN:INIT OK,TSP=NA
6369 GWT:TPC:IP=192.168.1.188
6400 GWT:TPC:DNS=9.9.9.9
6426 GWT:RMQ:MQTT RECONNECT
[hostByName] request IP for: test.mosquitto.org
[hostByName] Host: test.mosquitto.org IP: 37.187.106.16
:ref 1
:wr 25 25 0
:wrc 25 25 0
:sent 25
:rn 4
:c0 1, 4
6740 GWT:RMQ:MQTT CONNECTED
6797 GWT:TPS:TOPIC=mygateway1-out/0/255/0/0/17,MSG SENT
```
Code is available in my branch.

mfalkvidd, 2019-05-30:
I am not able clean up the code. Tried a few times, but get stuck on the complexity. I'll leave my code in case someone else wants to try.