mysensors/MySensors #928: Receive corrupted ID when using multiple repeaters.

goodchip, 2017-09-15:
Hello,

I recently installed a mysensors network in a large building on two floors. Since I have placed several repeaters, I find random problems during the associations of sensors.

## Description of the problem :

During the association, **randomly**, the sensor receives an ID longer than the 3 characters of range 0-255 normally expected.

These are often numbers 255 or 100 placed after the ID, other numbers appear or changes the ID itself, but this is more rare.

I suspect that repeaters "overwrite" the value by repeating at the same time the payload. (not sure).

The fewer repeaters in the bridge area, the less problems are manifested. (no problem without repeaters).

### Here are my logs when the problem occurs:
> 0 MCO:BGN:INIT NODE,CP=RNNNA--,VER=2.1.1
> 40 MCO:BGN:BFR
> 57 TSM:INIT
> 69 TSF:WUR:MS=0
> 92 TSM:INIT:TSP OK
> 112 TSM:FPAR
> 126 TSF:MSG:SEND,255-255-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
> 686 TSF:MSG:READ,10-10-255,s=255,c=3,t=8,pt=1,l=1,sg=0:1
> 747 TSF:MSG:FPAR OK,ID=10,D=2
> 802 TSF:MSG:READ,111-111-255,s=255,c=3,t=8,pt=1,l=1,sg=0:1
> 2205 TSM:FPAR:OK
> 2222 TSM:ID
> 2236 TSM:ID:REQ
> 2252 TSF:MSG:SEND,255-255-10-0,s=255,c=3,t=3,pt=0,l=0,sg=0,ft=0,st=OK:
> 4329 TSM:ID
> 4341 TSM:ID:REQ
> 4360 TSF:MSG:SEND,255-255-10-0,s=255,c=3,t=3,pt=0,l=0,sg=0,ft=0,st=OK:
> 6434 TSM:ID
> 6447 TSM:ID:REQ
> 6465 TSF:MSG:SEND,255-255-10-0,s=255,c=3,t=3,pt=0,l=0,sg=0,ft=0,st=OK:
> 8540 TSM:ID
> 8552 TSM:ID:REQ
> 8570 TSF:MSG:SEND,255-255-10-0,s=255,c=3,t=3,pt=0,l=0,sg=0,ft=0,st=OK:
> 8644 TSF:MSG:READ,0-0-255,s=255,c=3,t=4,pt=0,l=6,sg=0:**197255**
> **8708 TSF:SID:OK,ID=135**
> 8732 TSF:MSG:READ,0-10-255,s=255,c=3,t=4,pt=0,l=6,sg=0:**197255**
> 8798 TSF:MSG:BC
> 8814 TSF:MSG:READ,0-111-255,s=255,c=3,t=4,pt=0,l=6,sg=0:**197255**
> 8880 TSF:MSG:BC
> 8898 TSF:MSG:READ,0-12-255,s=255,c=3,t=4,pt=0,l=6,sg=0:**197255**
> 8962 TSF:MSG:BC
> 8980 TSM:ID:OK
> 8996 TSM:UPL
> 9011 TSF:MSG:SEND,**135**-135-10-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=OK:1
> 9099 TSF:MSG:READ,0-10-**135**,s=255,c=3,t=25,pt=1,l=1,sg=0:2
> 9160 TSF:MSG:PONG RECV,HP=2
> 9191 TSM:UPL:OK
> 9207 TSM:READY:ID=**135**,PAR=10,DIS=2
> 9246 TSF:MSG:SEND,**135**-135-10-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
> 9326 TSF:MSG:READ,0-10-**135**,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
> 9392 TSF:MSG:SEND,**135**-135-10-0,s=255,c=0,t=17,pt=0,l=5,sg=0,ft=0,st=OK:2.1.1
> 9474 TSF:MSG:SEND,**135**-135-10-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:10
> 11552 TSF:MSG:SEND,**135**-135-10-0,s=255,c=3,t=11,pt=0,l=11,sg=0,ft=0,st=OK:sensor_test
> 11642 TSF:MSG:SEND,**135**-135-10-0,s=255,c=3,t=12,pt=0,l=4,sg=0,ft=0,st=OK:v0.1
> 11722 MCO:REG:REQ
> 11743 TSF:MSG:SEND,**135**-135-10-0,s=255,c=3,t=26,pt=1,l=1,sg=0,ft=0,st=OK:2
> 11821 TSF:MSG:READ,0-10-**135**,s=255,c=3,t=27,pt=1,l=1,sg=0:1
> 11884 MCO:PIM:NODE REG=1
> 11909 MCO:BGN:STP
> 11929 MCO:BGN:INIT OK,TSP=1
> 22341 TSF:MSG:READ,0-10-**135**,s=255,c=3,t=6,pt=0,l=1,sg=0:M

In the example below, the good ID sent by the gateway is 197, the sensor receives 197255 and takes ID 135 after conversion:
197255 = b110000001010000111, converted to uint8, only the last 8 bits on the right (b010000111) are retrieved, giving 135.

### For information, here is the topology of the network setup:

[image]

## Proposed amendment to the code:

I have not yet figured out the exact origin of the problem, however it seems to me it would be a good idea that the protocol is resistant to this ID corruption:

In the transportAssignNodeID function of MyTransport.cpp, the test:
```
// verify if ID valid
if (newNodeId! = GATEWAY_ADDRESS && newNodeId <= AUTO) {
```
Replacing "! =" With "<=" is not enough because the conversion to uint8 returns a pseudo-random number when the ID is greater than 255;

Change to uint32 type rather than uint8 works but is not optimized and could result many side-effects.

I propose this modification based on the length of the message which has the advantage of not touching the statement of the function:

### Before:
```
bool transportAssignNodeID(const uint8_t newNodeId)
{
	// verify if ID valid
	if (newNodeId != GATEWAY_ADDRESS && newNodeId != AUTO) {
		_transportConfig.nodeId = newNodeId;
		transportSetAddress(newNodeId);
		// Write ID to EEPROM
		hwWriteConfig(EEPROM_NODE_ID_ADDRESS, newNodeId);
		TRANSPORT_DEBUG(PSTR("TSF:SID:OK,ID=%d\n"),newNodeId);	// Node ID assigned
		return true;
	} else {
		TRANSPORT_DEBUG(PSTR("!TSF:SID:FAIL,ID=%d\n"),newNodeId);	// ID is invalid, cannot assign ID
		setIndication(INDICATION_ERR_NET_FULL);
		_transportConfig.nodeId = AUTO;
		return false;
	}
}
```

### After:
```
bool transportAssignNodeID(const uint8_t newNodeId)
{
	// verify if ID corrupted
	if(mGetLength(_msg) <= 3) {
		// verify if ID valid
		if (newNodeId != GATEWAY_ADDRESS && newNodeId <= AUTO) {
			_transportConfig.nodeId = newNodeId;
			transportSetAddress(newNodeId);
			// Write ID to EEPROM
			hwWriteConfig(EEPROM_NODE_ID_ADDRESS, newNodeId);
			TRANSPORT_DEBUG(PSTR("TSF:SID:OK,ID=%d\n"),newNodeId);	// Node ID assigned
			return true;
		}
		TRANSPORT_DEBUG(PSTR("!TSF:SID:FAIL,ID=%d\n"),newNodeId);	// ID is invalid, cannot assign ID
	} else {
		TRANSPORT_DEBUG(PSTR("!TSF:SID:FAIL,CORRUPT\n"),newNodeId);	// ID is corrupted, cannot assign ID
	}
	setIndication(INDICATION_ERR_NET_FULL);
	_transportConfig.nodeId = AUTO;
	return false;
}
```

### Log result example when ID corruction appears with this new transportAssignNodeID() function:
> 0 MCO:BGN:INIT NODE,CP=RNNNA--,VER=2.1.1
> 40 MCO:BGN:BFR
> 57 TSM:INIT
> 69 TSF:WUR:MS=0
> 92 TSM:INIT:TSP OK
> 112 TSM:FPAR
> 126 TSF:MSG:SEND,255-255-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
> 710 TSF:MSG:READ,111-111-255,s=255,c=3,t=8,pt=1,l=1,sg=0:1
> 772 TSF:MSG:FPAR OK,ID=111,D=2
> 985 TSF:MSG:READ,10-10-255,s=255,c=3,t=8,pt=1,l=1,sg=0:1
> 2205 TSM:FPAR:OK
> 2222 TSM:ID
> 2236 TSM:ID:REQ
> 2252 TSF:MSG:SEND,255-255-111-0,s=255,c=3,t=3,pt=0,l=0,sg=0,ft=0,st=OK:
> 4329 TSM:ID
> 4341 TSM:ID:REQ
> 4360 TSF:MSG:SEND,255-255-111-0,s=255,c=3,t=3,pt=0,l=0,sg=0,ft=0,st=OK:
> 6434 TSM:ID
> 6447 TSM:ID:REQ
> 6465 TSF:MSG:SEND,255-255-111-0,s=255,c=3,t=3,pt=0,l=0,sg=0,ft=0,st=OK:
> 8540 TSM:ID
> 8552 TSM:ID:REQ
> 8570 TSF:MSG:SEND,255-255-111-0,s=255,c=3,t=3,pt=0,l=0,sg=0,ft=0,st=OK:
> 8646 TSF:MSG:READ,0-10-255,s=255,c=3,t=4,pt=0,**l=6**,sg=0:**197255**
> **8710 !TSF:SID:FAIL,CORRUPT**
> 8730 TSF:MSG:READ,0-111-255,s=255,c=3,t=4,pt=0,**l=6**,sg=0:**197255**
> **8796 !TSF:SID:FAIL,CORRUPT**
> 8816 TSF:MSG:READ,0-111-255,s=255,c=3,t=4,pt=0,**l=3**,sg=0:**197**
> **8880 TSF:SID:OK,ID=197**
> 8904 TSM:ID:OK
> 8921 TSM:UPL
> 8935 TSF:MSG:SEND,**197**-197-111-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=OK:1
> 9029 TSF:MSG:READ,0-111-197,s=255,c=3,t=25,pt=1,l=1,sg=0:2
> 9091 TSF:MSG:PONG RECV,HP=2
> 9119 TSM:UPL:OK
> 9138 TSM:READY:ID=**197**,PAR=111,DIS=2
> 9177 TSF:MSG:SEND,**197**-197-111-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
> 9256 TSF:MSG:READ,0-111-**197**,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
> 9324 TSF:MSG:SEND,**197**-197-111-0,s=255,c=0,t=17,pt=0,l=5,sg=0,ft=0,st=OK:2.1.1
> 9408 TSF:MSG:SEND,**197**-197-111-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:111
> 9531 TSF:MSG:READ,0-111-**197**,s=255,c=3,t=6,pt=0,l=1,sg=0:M
> 9594 TSF:MSG:SEND,**197**-197-111-0,s=255,c=3,t=11,pt=0,l=11,sg=0,ft=0,st=OK:sensor_test
> 9684 TSF:MSG:SEND,**197**-197-111-0,s=255,c=3,t=12,pt=0,l=4,sg=0,ft=0,st=OK:v0.1
> 9764 MCO:REG:REQ
> 9785 TSF:MSG:SEND,**197**-197-111-0,s=255,c=3,t=26,pt=1,l=1,sg=0,ft=0,st=OK:2
> 9863 TSF:MSG:READ,0-111-**197**,s=255,c=3,t=27,pt=1,l=1,sg=0:1
> 9924 MCO:PIM:NODE REG=1
> 9949 MCO:BGN:STP
> 9967 MCO:BGN:INIT OK,TSP=1

Best Regards.

goodchip, 2017-09-15:
### Small additionals precisions:
The gateway and repeaters have the minimal example sketch provided on the mysensor site in version 2.1.1.

The sensor is flashed with the following test sketch:
```
#define MY_RADIO_NRF24
#define MY_BAUD_RATE 9600
#define MY_DEBUG

#include <MySensors.h>

void setup()
{
  
}

void before()
{

}

void presentation()
{
  
    sendSketchInfo("sensor_test", "v0.1");
    
}

void loop()
{

}
```

The nRF24L01 chips are versions without PA / LNA, and original (not Chinese copies, although I have also tried with).

henrikekblad, 2017-09-16:
Exemplary bug report!

The radio has a build CRC check, so getting a corrupt message is very improbable. As you say the most probably cause is that something happens with the message during relaying. A modification of length or overwrite.

fallberg, 2017-09-16:
It could also be interesting to see if signing the messages could affect the behavior. If the repeater modified the message inadvertently, the signature would be corrupt. I suspect this is the case, in which case adding signing would also serve as a validation of the end to end message.

tekka007, 2017-09-16:
The length field is corrupt (ID response msgs have a 1 byte payload) - would be intersting to see the repeater logs.

```
8644 TSF:MSG:READ,0-0-255,s=255,c=3,t=4,pt=0,**l=6**,sg=0:197255
8708 TSF:SID:OK,ID=135
8732 TSF:MSG:READ,0-10-255,s=255,c=3,t=4,pt=0,**l=6**,sg=0:197255
8798 TSF:MSG:BC
8814 TSF:MSG:READ,0-111-255,s=255,c=3,t=4,pt=0,**l=6**,sg=0:197255
8880 TSF:MSG:BC
8898 TSF:MSG:READ,0-12-255,s=255,c=3,t=4,pt=0,**l=6**,sg=0:197255
...
8816 TSF:MSG:READ,0-111-255,s=255,c=3,t=4,pt=0,**l=3**,sg=0:197
```

goodchip, 2017-09-19:
Thank you for your answers,

@henrikekblad: I put a repeater in debug mode, and actually the frame seems well corrumpuous at the source.

@fallberg & @ tekka007: the lenght field is always equal to the actual length of the payload (it is precisely on this parameter that my code is based to detect a corrupted frame). In short, the problem does not seem to come at all from the RF protocol.

So I disconnected all repeaters and the problem remains.

I suspected many repeaters because the problem occurred when I installed them.

I have refocused on the gateway, which seems to be the common denominator: by plugging a MySensors key (serial gateway on USB via FTDI) the problem disappears completely.

I suspect actually the serial link between RPi and the ATMEGA328p.

I look at that and keep you informed.

Best regards

goodchip, 2017-09-20:
Origin of the problem solved!

**It was a write too fast on the serial port** (C POSIX with RPi running Raspbian).

Would not it be wise to check the ID transferred in the payload?

If you do not see the interest, I can close this issue.

goodchip, 2017-09-20:
For information, here is the new routine I use (with usleep):

```
define SERIAL_TRANSMISSION_LENGTH 10      // 8N1 = 10 bits
define SERIAL_BAUDRATE 38400

writeSerial( const char* string ) {

    int myReturn = 0;
    serialMutex.lock();
    lastMutexLocked = MUTEX_WRITE;

    size_t myLength = strlen( string );

    myWriteLength = write( fileDescriptor, string, myLength );

    // waiting the end of wrote frame:
    usleep( (unsigned int)( ceil((myWriteLength * (SERIAL_TRANSMISSION_LENGTH)) / (double)(SERIAL_BAUDRATE)) ) );

    if( myWriteLength != myLength ) {
        myReturn = false;
    }

    serialMutex.unlock();

}
```

closed 2017-11-01 by henrikekblad