mysensors/MySensors #930: Splash Screen can break backward compatibility

LemarinelNet, 2017-09-18:
The splash-screen (included since #889) which is set ENABLED by default breaks the compatibility on my openhab binding:

It seems that the bindings always split the incoming frame with semi-column, but the splash lines doesn't contains the required semi-columns, and then the binding code (which is not robust enough) fails in a java unhandled error.

Would'nt it be more secure to output the splash-screen as a common frame (i.e. a comment or debug frame) ?

henrikekblad, 2017-09-18:
@tekka007, maybe the splash should be printed like normal debug messages on gateway-nodes so it doesn't interfere with the normal command parsing?

LemarinelNet, 2017-09-18:
PS : @tekka007,Sorry for this off-topic, I've dropped you a message in mysensors forum about OTA ;) Have you seen it ?

tekka007, 2017-09-18:
@hek This will most likely result in a distorted splash screen (due to the timer field).
I see two options: 
1. disable it if plugin has issues:
`#define MY_SPLASH_SCREEN_DISABLED
`
2. disable it on all GW nodes by default

LemarinelNet, 2017-09-18:
I vote for a disabled by default behavior.
Having this enabled by default might get many users in trouble if they doesn't understand what's happening ;)

henrikekblad, 2017-09-18:
@tekka, option 2 is probably best for backward compatibility.

tekka007, 2017-09-19:
Fixed in #931

closed 2017-09-19 by tekka007