mysensors/MySensors #935: Segfault after latest update 2.2.0

harryd100, 2017-09-22:
Hi, After building latest update of the 2.2.0 development branch (I need this for my RFM69), I get:

 /usr/local/bin/mysgw -d
mysgw: Starting gateway...
mysgw: Protocol version - 2.2.0-beta
mysgw: MCO:BGN:INIT GW,CP=RPNG----,VER=2.2.0-beta
mysgw: TSF:LRT:OK
mysgw: TSM:INIT
mysgw: TSF:WUR:MS=0
**Segmentation fault**

tekka007, 2017-09-22:
@harryd100 Please add additional information such as HW used, IDE version, board def version, GW sketch, etc.

harryd100, 2017-09-22:
It's just the RPI 2 gateway, it crashed when started after latest update. I have a backup image of last week, so i'm going to install that first this evening and try to restore it as all my sensors are dead now. I have tried to use a previous commit, but still the same result, maybe it's something else.

mfalkvidd, 2017-09-22:
@harryd100 could you post the full configure command? That would make reproducing and troubleshooting the problem easier.

harryd100, 2017-09-22:
I have already restored an old SD image and upgraded again to the latest 2.2.0 build, and guess what, it runs again. Don't know what happened before, but everything is working again. Will keep an eye on it with the next Beta update. Thanks all

closed 2017-09-22 by harryd100