mysensors/MySensors #941: NRF24 Radio does not link if MY_DEBUG_VERBOSE_RF24 is not defined.

SergioRius, 2017-10-01:
As the title describes, if the Radio debug flag is not defined, the node fails linking to the gateway.
I'm using the RaspberryPi NRF gateway.

The simplest Sketch:
`#define SKETCH_NAME "Development"
#define SKETCH_VERSION "1.0"

#define MY_DEBUG
#define MY_RADIO_RF24
//#define MY_DEBUG_VERBOSE_RF24

#include <MySensors.h>

void presentation() {
  Serial.println(F("<= presentation"));
  sendSketchInfo(SKETCH_NAME, SKETCH_VERSION);
}

void setup() {
  Serial.begin(MY_BAUD_RATE);
  Serial.println(F("<= setup"));
}

void loop() {
  Serial.println(F("<= loop"));
}

void receive(const MyMessage &message) {
  Serial.println(F("<= receive"));
}

void receiveTime(unsigned long ts) {
  Serial.println(F("<= receiveTime"));
}
`

The debug result:
`Opening port
Port open
 __  __       ____
|  \/  |_   _/ ___|  ___ _ __  ___  ___  _ __ ___
| |\/| | | | \___ \ / _ \ `_ \/ __|/ _ \| `__/ __|
| |  | | |_| |___| |  __/ | | \__ \  _  | |  \__ \
|_|  |_|\__, |____/ \___|_| |_|___/\___/|_|  |___/
        |___/                      2.2.0-beta

17 MCO:BGN:INIT NODE,CP=R-NNA---,VER=2.2.0-beta
26 TSM:INIT
27 TSF:WUR:MS=0
34 TSM:INIT:TSP OK
36 TSF:SID:OK,ID=6
37 TSM:FPAR
1637 TSF:MSG:SEND,6-6-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
3645 !TSM:FPAR:NO REPLY
3647 TSM:FPAR
5246 TSF:MSG:SEND,6-6-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
7254 !TSM:FPAR:NO REPLY
7256 TSM:FPAR
8855 TSF:MSG:SEND,6-6-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
10862 !TSM:FPAR:NO REPLY
10864 TSM:FPAR
12464 TSF:MSG:SEND,6-6-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
14472 !TSM:FPAR:FAIL
14473 TSM:FAIL:CNT=1
14475 TSM:FAIL:DIS
14477 TSF:TDI:TSL
24480 TSM:FAIL:RE-INIT
24482 TSM:INIT
24489 TSM:INIT:TSP OK
24492 TSF:SID:OK,ID=6
[... 2120 lines cut ...]
6956 RF24:WBR:REG=7,VAL=48
6958 RF24:STL
6960 RF24:WBR:REG=0,VAL=15
6962 RF24:WBR:REG=10,VAL=6
6965 TSF:MSG:SEND,6-6-0-0,s=255,c=3,t=26,pt=1,l=1,sg=0,ft=0,st=OK:2
6971 RF24:RBR:REG=8,VAL=17
6973 RF24:RBR:REG=23,VAL=16
6976 RF24:RXM:LEN=8
6978 RF24:WBR:REG=7,VAL=64
6980 TSF:MSG:READ,0-0-6,s=255,c=3,t=27,pt=1,l=1,sg=0:1
6985 MCO:PIM:NODE REG=1
6987 RF24:RBR:REG=23,VAL=17
6990 RF24:RBR:REG=23,VAL=17
6992 MCO:BGN:STP
<= setup
6994 MCO:BGN:INIT OK,TSP=1
6998 RF24:RBR:REG=23,VAL=17
<= loop
7000 RF24:RBR:REG=23,VAL=17
<= loop
7003 RF24:RBR:REG=23,VAL=17
<= loop
7007 RF24:RBR:REG=23,VAL=17
<= loop
7010 RF24:RBR:REG=23,VAL=17`

tekka007, 2017-10-01:
This could be a timing issue with fast GW / slow nodes, i.e. GW answer arrives before node radio is in RX.

SergioRius, 2017-10-02:
@tekka007 I'm just testing that version and still doesn't work. As soon as I activate MY_DEBUG_VERBOSE_RF24 it registers.
The HW is the same than in [Iss:942](https://github.com/mysensors/MySensors/issues/942#issuecomment-333391134)

> I've tested it on a Oh! MyRgbdimmer board (so Pro mini 3v3/8mhz with on board radio routing) and on an 8-port energy meter I'm developing ( pro-mini 5v/16mhz). And using different nrf24 modules.
Baud rate 9600 and 115200
Arduino IDE 1.8.4 - AVR Boards 1.6.20

> The gateway is a raspberry pi v2.1.1 compiled with:
./configure --my-transport=nrf24 --my-rf24-irq-pin=18 --my-gateway=ethernet --my-port=5003
The module on the raspberry is a PA+LNA with aditional regulated ps from 5v.
The controller is domoticz v3.8451 beta on the same pi 3.

SergioRius, 2017-10-02:
@tekka007 I'm still testing on this issue and I'm getting a strange behaviour.
Just one question: Mysensors saves anything related to radio management or else on eeprom of nodes or the gateway?

What I'm experiencing: After the post avobe, I reverted my library version, but still was failing. So I emptied all cache and temp folders for two IDEs but nothing worked. I checked connections on the NRF module (board, not wires) and where Ok so I replaced it but no way.
So I picked up a fresh board checked back you timing fix version, and tadá! started working. I then tried 2.20 beta and also worked. So 2.1.1.

Went back to one of the two other boards and also worked. The other one doesn't, even when connections and module is Ok, but I'm starting to think is something related to gateway and/or controller.

Does it make sense?

Or I'm having a problem with compilation caching...

andreypopov, 2017-10-17:
@tekka007  wow!!! man!
I updated to the latest version 2.1.1 and 2.2beta and thw whole my system failed. 
I spent almost whole day to fix it. Nothing worked.

Then I tried to load your fork to one of my devices and it helped!! It is magic. 
Should I update serial-arduino-nrf24-gateway too ?

bbrowaros, 2018-04-04:
Hi, 

I'm new to this topic so it might be that I've screwed something however looks like I'm hitting similar issue with my nodes: 
```
112959 RF24:RBR:REG=23,VAL=17
112962 RF24:RBR:REG=23,VAL=17
112964 RF24:RBR:REG=23,VAL=17
112967 RF24:RBR:REG=23,VAL=17
112970 RF24:RBR:REG=23,VAL=17
112973 RF24:RBR:REG=23,VAL=17
112975 RF24:RBR:REG=23,VAL=17
112978 RF24:RBR:REG=23,VAL=17
112982 RF24:RBR:REG=23,VAL=17
112985 !TSM:FPAR:FAIL
```

```
Connection to GW on PI via NRF24: 
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[32mINFO #033[0m Starting gateway...
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[32mINFO #033[0m Protocol version - 2.3.0-alpha
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m MCO:BGN:INIT GW,CP=RNNGL-Q-,VER=2.3.0-alpha
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m TSF:LRT:OK
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m TSM:INIT
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m TSF:WUR:MS=0
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m TSM:INIT:TSP OK
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m TSM:INIT:GW MODE
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m TSM:READY:ID=0,PAR=0,DIS=0
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m MCO:REG:NOT NEEDED
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m Listening for connections on 0.0.0.0:5003
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m MCO:BGN:STP
Apr  4 23:42:23 raspberrypi mysgw[27135]: Apr 04 23:42:23 #033[36mDEBUG#033[0m MCO:BGN:INIT OK,TSP=1
Apr  4 23:42:51 raspberrypi mysgw[27135]: Apr 04 23:42:51 #033[36mDEBUG#033[0m New connection from 127.0.0.1
Apr  4 23:42:51 raspberrypi mysgw[27135]: Apr 04 23:42:51 #033[36mDEBUG#033[0m GWT:TSA:C=0,CONNECTED
Apr  4 23:42:51 raspberrypi mysgw[27135]: Apr 04 23:42:51 #033[36mDEBUG#033[0m GWT:RFC:C=0,MSG=0;0;3;0;2;
Apr  4 23:42:52 raspberrypi mysgw[27135]: Apr 04 23:42:52 #033[36mDEBUG#033[0m GWT:RFC:C=0,MSG=0;0;3;0;2;Get Versio
```n

I've tried with development branch and master with GW however this is still not working. The Mysensors library for arduino is 2.2.0 (the same is happening on 2.1.0). 

Not sure what is going on. I've probably messed something up but cannot find what. 
I see that this fixed the problem: https://github.com/tekka007/MySensors/tree/TimingIssue  
Is this integrated with development branch?

brenard, 2018-12-09:
Hello I'm new in this issue and I just start using MySensors. I have a global problem on my MySensors network and all my nodes (only 1 gateway and 2 nodes) seem to have this problem. _FPAR_ "never" occurred and when I enable _MY_DEBUG_VERBOSE_RF24_ I have (in loop) the following message in the serial console (gateway and node) between messages : 
`RF24:RBR:REG=23,VAL=17 `

I try different solutions this last weeks to eventually solve other problems : 

- Check/recheck (a lot) my wires
- Increase/Decrease _MY_RF24_PA_LEVEL_
- Test with different value of _MY_RF24_CHANNEL_
- I had some LED to show network activity : messages are sent, sometime received and error LED frequently blinked...
- I had Decoupling-Capacitor (4.7uF and 47uF) on all RF24
- I had a 5v->3.3V regulator to powering radio
- I had a external power source to ensure power

I also try to set static node ID on my nodes and it firstly mostly work with a lot of errors. When I reboot node, it's not always reconnect. Finally, my two nodes never reconnect.

I also try the timing branch (https://github.com/tekka007/MySensors/tree/RF24STDBYTiming) as recommend : no change.

Do you have found any solution / work-around that I have not tried yet ? As things stand, I'm completely stuck.

bbrowaros, 2018-12-09:
Hi,

I was fighting with this a long time and finally gave up. I've return to this project after some time and wired back radio using new msgw library (2.3.0). This time all worked fine. I've tried to check what difference I've made with previous setup and looks like this was a wiring issue bettwen arduino and radio and lack of capacitor on nrf24 radio. Nrf24 on raspberry was correctly wired. 

Good luck. Hope this helps a bit.

brenard, 2018-12-09:
Hi,
Thanks you for your return ! Firstly, you have given me the courage to restart working on my installation and a path to explore : wiring. I disconnected everything, checked all my welds and tested my connections one by one. Finally, I turn on my gateway and my nodes one by one and all worked fine!

If somebody have the same issue, or at least the _"RF24:RBR:REG=23,VAL=17"_ error, I check in my case : I always have this message if I enable RF24 debugging, but it's not blocking.

flatsiedatsie, 2019-05-09:
This issue has re-appeared here:
https://forum.mysensors.org/topic/10327/rf-nano-nano-nrf24-for-just-3-50-on-aliexpress/29

mfalkvidd, 2019-05-09:
@flatsiedatsie sorry I don't follow. Could you clarify how this is related? I don't see any linking problems in the thread youposted.