mysensors/MySensors #942: MySensors crashes just after the logo.

SergioRius, 2017-10-01:
If MY_DEBUG is not defined, MySensors crashes after or during the logo print.
At slower baud rates the node seems to be slowing down during the logo print, in some cases doesn't end printing it.

The minimal sketch that reproduces:
```Arduino
#define SKETCH_NAME "Development"
#define SKETCH_VERSION "1.0"

#define MY_RADIO_RF24
//#define MY_DEBUG
//#define MY_DEBUG_VERBOSE_RF24

#include <MySensors.h>

void presentation() {
  Serial.println(F("<= presentation"));
  sendSketchInfo(SKETCH_NAME, SKETCH_VERSION);
}

void setup() {
  Serial.begin(MY_BAUD_RATE);
  Serial.println(F("<= setup"));
}

void loop() {
  Serial.println(F("<= loop"));
}
```

henrikekblad, 2017-10-01:
Do you get the same crash if you define

#define MY_SPLASH_SCREEN_DISABLED 

?

tekka007, 2017-10-01:
@SergioRius Negative, just tested the most recent dev branch with your sketch and everything is running just fine.
```
 __  __       ____
|  \/  |_   _/ ___|  ___ _ __  ___  ___  _ __ ___
| |\/| | | | \___ \ / _ \ `_ \/ __|/ _ \| `__/ __|
| |  | | |_| |___| |  __/ | | \__ \  _  | |  \__ \
|_|  |_|\__, |____/ \___|_| |_|___/\___/|_|  |___/
        |___/                      2.2.0-beta

<= presentation
<= setup
<= loop
<= loop
<= loop
<= loop
<= loop
<= loop
<= loop
<= loop
<= loop
...
```

Arduino Pro Mini, 3V3, 8Mhz
Baud rate 9600 and 115200
Arduino IDE 1.8.4
AVR board defs 1.6.20
MySensors 2.2.0-beta, current dev [3ca9b56](https://github.com/mysensors/MySensors/commit/3ca9b56380967265a064c5b9cc6757bbb01eac51)

What are your specs / HW / debug output?

SergioRius, 2017-10-02:
@henrikekblad Yes, I just tried and also crashes.

@tekka007 I've triple checked it today, in case I mis-configured something or the board kind (mhz) was wrong, but no. I can reproduce it by only changing the MySensors folder for the devel one.
I've cleaned the IDE caches folders on every switch.

I've tested it on a Oh! MyRgbdimmer board (so Pro mini 3v3/8mhz with on board radio routing) and on an 8-port energy meter I'm developing ( pro-mini 5v/16mhz). And using different nrf24 modules.
Baud rate 9600 and 115200
Arduino IDE 1.8.4 - AVR Boards 1.6.20
The IDE also outputs this warning for 2.2.0-beta:

> WARNING: Spurious .mystools folder in 'MySensors' library

And Visual Studio throws the following error by just changing the Mys version:

> E1696	can't open the source file (error code "limits.h").	SimpleTest	c:\Program Files (x86)\Arduino\hardware\tools\avr\avr\include\avr\boot.h	107	

But this can be anything else.

The gateway is a raspberry v2.1.1 compiled with:
`./configure  --my-transport=nrf24  --my-rf24-irq-pin=18  --my-gateway=ethernet --my-port=5003`
The module on the raspberry is a PA+LNA with aditional regulated ps from 5v.
The controller is domoticz v3.8451 beta.

If I have some time I'll do a diff between 21.1 and 2.2.0 and rolling back some sections on the new version until it doesn't crash anymore.

SergioRius, 2017-10-02:
@henrikekblad Please take a look at: [NM: 223](https://github.com/mysensors/NodeManager/issues/223)
It seems that the first time I tested for that bug, something went wrong with my gateway/controller for two of my testing nodes. I managed to recover them and I found that your suggestion on `MY_SPLASH_SCREEN_DISABLED` works for that.
Now without first running NM on the node, seems that v2.2.0-beta works.

closed 2017-10-06 by henrikekblad