mysensors/MySensors #944: Can't connect to radio on raspPi with latest stable 2.1.1 build, but works with 2.2.beta but with beta mqtt gateway very unstable...

jkoenig72, 2017-10-02:
Hi, just wondering if I simple do something wrong.

I try to setup a mysensors gateway on raspPi. What I did:

git clone https://github.com/mysensors/MySensors.git --branch master

./configure --my-gateway=mqtt --my-controller-ip-address=127.0.0.1 --my-mqtt-publish-topic-prefix=mysensors-out --my-mqtt-subscribe-topic-prefix=mysensors-in --my-mqtt-client-id=mygateway1 --my-rf24-ce-pin=22 --my-rf24-cs-pin=24 --my-rf24-irq-pin=15

Build seems ok, but when I tried to start the gw, it seems radio can not be initialized. I got

i@pi:~/msg_stable/MySensors $ sudo bin/mysgw -d
mysgw: Starting gateway...
mysgw: Protocol version - 2.1.1
mysgw: MCO:BGN:INIT GW,CP=RNNG--Q,VER=2.1.1
mysgw: TSM:INIT
mysgw: TSF:WUR:MS=0
mysgw: !TSM:INIT:TSP FAIL
mysgw: TSM:FAIL:CNT=1
mysgw: TSM:FAIL:PDT

When I use same kind of install with the development branch 2.2.beta it all starts up, however the gateway seems very unstable (not surprisingly, as its dev build), it cashes a lot.

So - how to get the stable 2.1.1 connecting to the radio (NRF24L01+ )? I guess the gateway on 2.1.1 is stable...

Regards
Joerg

bilbolodz, 2017-10-03:
Same for me: 2.1.1 no connection to radio, 2.2 beta is working.

mfalkvidd, 2017-10-23:
Seems like this was fixed in https://github.com/mysensors/MySensors/pull/827
Some more information: https://forum.mysensors.org/post/77973

How about closing this as fixed by #827 ?

SunSeaAndPalms, 2017-10-28:
Just faced the same issue in the `master` branch, but I see the `release` branch with `2.2.0` seems to fix that.

fallberg, 2017-11-18:
If the release branch (now deleted) fixes it, it is fixed on development.

closed 2017-11-18 by fallberg

jkoenig72, 2018-01-05:
Folks,

I tested the current development branch - still the same, this issue is not solved.

the gateway on Rpi goes down often with segmentation fault. HIGHLY unreliable.

Anything I can do? Some advice? How can I assist in finding the root of this issue.

Joerg

jkoenig72, 2018-01-05:
I guess this one needs to be reopened then...

mfalkvidd, 2018-01-05:
Troubleshooting is going on in these threads
https://forum.mysensors.org/topic/8836/raspberry-pi-stability-problem/
https://forum.mysensors.org/post/82711

fallberg, 2018-01-07:
Is this related to #1025? Can probably be closed then (it is merged so just try with the latest development changes).