mysensors/MySensors #945: MySensors MQTT GW -> AWS IoT

juampe, 2017-10-03:
I use MySensors 2.2.0-beta from git as a gateway (onto raspberri pi) for my RF MySensors network and tested succesfully with domoticz controller via My Sensors gateway LAN interface.

Now, I want to test and use a cloud controller like AWS IoT

The AWS IoT SDK provides the folowing certificates to log into the service

Sample from SDK
```
#define AWS_IOT_MQTT_HOST              "a22j5sm6o3yzc5.iot.us-east-1.amazonaws.com"
#define AWS_IOT_MQTT_PORT              8883
#define AWS_IOT_MQTT_CLIENT_ID         "MyRaspberryPi"
#define AWS_IOT_MY_THING_NAME          "MyRaspberryPi"
#define AWS_IOT_ROOT_CA_FILENAME       "root-CA.crt"
#define AWS_IOT_CERTIFICATE_FILENAME   "4bbdc778b9-certificate.pem.crt"
#define AWS_IOT_PRIVATE_KEY_FILENAME   "4bbdc778b9-private.pem.key"
```
But I realize that the MySensors MQTT gateway only have simple auths methods

```
root@kelsier:~/arduino/libraries/MySensors# ./configure --help|grep mqtt
    --my-gateway=[none|ethernet|serial|mqtt]
                                If gateway is set to mqtt, it sets the broker port.
    --my-mqtt-client-id=<ID>    MQTT client id.
    --my-mqtt-user=<UID>        MQTT user id.
    --my-mqtt-password=<PASS>   MQTT password.
    --my-mqtt-publish-topic-prefix=<PREFIX>
    --my-mqtt-subscribe-topic-prefix=<PREFIX>
```

Sure I miss something, maybe some MQTT featues is not supported yet or maybe AWS is too contemporary , for in MQTT protocol in general.

mfalkvidd, 2017-10-04:
MySensors is using [knolleary's mqtt library](https://pubsubclient.knolleary.net/). There seems to have been [some work](https://github.com/knolleary/pubsubclient/issues/84) to get certificate-based communication working, with mixed results.

For the raspberry pi gateway, it might be easier to try to find a non-arduino mqtt library that can handle whatever aws iot needs. Knolleary's library needs the cpu and ram power of an esp8266 for tls support so it won't be useful on atmega328 anyway.

@juampe would you be able to investigate the possibilites and work needed to use another library?

henrikekblad, 2017-10-04:
I would suggest setting up a MQTT broker which forward the data to AWS. E.g.. mosquitto.

http://www.steves-internet-guide.com/mosquitto-bridge-configuration/

henrikekblad, 2018-02-18:
Seems someone successfully got certificates up and working on ESP8266.
https://github.com/knolleary/pubsubclient/issues/84
See comment on on 10 Dec 2017. 

Closing this as it's outside the MySensors library scope.

closed 2018-02-18 by henrikekblad