mysensors/MySensors #946: ESP8266Gateway do not connect to Access Point

tmsd2001, 2017-10-06:
Hello,

the ESP8266Gateway on NodeMCU V0.9 do not connect to the Access Point.
I was able to correct the error with change the wait(500) to delay(500) after the WIFI.begin in the 
MyGatewayTransportEthernet.cpp

regards

mfalkvidd, 2017-10-06:
@tmsd2001 which version/branch are you using? Do you have the possibility to try on other esp-based boards? I'm using Wemos D1 mini as ethernet gateway on MySensors 2.1.1 and haven't needed to change the library. I don't have a nodemcu 0.9 unfortunately.

tmsd2001, 2017-10-06:
I have only the NodeMCU 0.9 I use the 2.2.0-Dev, downloaded e few hours ago. 
The MySensors debug print an error, but I don't have a log. It was some like Err Eth cannot connect.
Which is always repeated.
After changing, the Gateway run, but I have "LmacRxBlk:1" in the debug screen.
Have you seen the Note on the ESP Git site to use delay?
In Arduino IDE for ESP, the delay is not a standard delay, it is like "wait" in Mysensors.

mfalkvidd, 2017-10-06:
Thanks for the additional information.

I'm not that deep into the MySensors hardware abstraction for ESP but from what I can read from the code, MySensors wait() calls esp_schedule() and esp_yield(), just like ESP's own delay [does](https://github.com/esp8266/Arduino/blob/2126146e20042878026f03a19107555f32e3431c/cores/esp8266/core_esp8266_wiring.c#L43). So there seems to be no difference. If there is a difference, it seems a bit strange that my 3 gateways work but it could be a change from 2.1.1 to development. I can test with development tomorrow.

Could you clarify which if the esp sites have a note on delay? I don't remember seeing that note, and googling didn't give me any useful results :( I found [this](https://github.com/esp8266/Arduino/issues/2186), but it doesn't seem related. I might be wrong though.

tmsd2001, 2017-10-06:
[http://esp8266.github.io/Arduino/versions/2.0.0/doc/reference.html#timing-and-delays](url)

mfalkvidd, 2017-10-06:
Thanks. Then there should not be any difference, since esp_schedule() and esp_yield() are called by MySensors wait().

tmsd2001, 2017-10-07:
OK, wait() call the same as esp_schedule() and esp_yield() , but "wait ()" also has to do Mysensor things, right? 
delay() do only ESP things, I would assume.
I remember recently having created a gateway with version 2.1.1, there were no problems.
The 2.1.1 Gateway was created with Windows IDE the new 2.2.0 with Linux.
I switch to 2.2.0 because of the static IP problem but also has a 2.1.1 version with static IP fix.
Afterwards I will try again, because I still want to make changes at the gateway.

mfalkvidd, 2017-10-07:
Yes, MySensors wait() checks for incoming (radio) messages. Just like ESP delay() does the Arduino delay plus some small additional tasks, MySensors wait() does the ESP delay() plus some small additional tasks.

mfalkvidd, 2017-10-07:
I'm unable to reproduce the problem on an NodeMCU 1.0. I used commit 8cb9b2692098 which currently is the latest commit in the MySensors development branch.

Output from the gateway:

> 5997 MCO:BGN:INIT GW,CP=R-NGE---,VER=2.2.0-beta
> scandone
> f 0, scandone
> state: 0 -> 2 (b0)
> state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 3
> cnt 
> 
> connected with Mr-IoT.com, channel 2
> dhcp client start...
> .....ip:192.168.1.215,mask:255.255.255.0,gw:192.168.1.1
> .IP: 192.168.1.215

Arduino IDE version 1.6.8 on Windows 7.
ESP8266 board version 2.3.0

Do you have any idea why this problem occurs on your board?

mfalkvidd, 2017-10-07:
@tmsd2001 could it be that 500ms sometimes is too short (depending on status of the wifi router for example)? How many times in a row did it fail with wait, and does delay always result in success?

tmsd2001, 2017-10-07:
The China NodeMCU V0.9 has some timing problems. If I use the ESP8266 example sketch the Serial output doesn't work at the first attempt, I need to insert a delay (1000) behind the serial.begin statement then the serial output works. I have found the evidence on StackOverflow from other users. So there are several users who have problems.

tmsd2001, 2017-10-09:
Small update, the problem is only on TCP Gateway.
Change from 500ms to more does not improve.
delay(1000) improves the problem, but does not eliminate it.
I will continue to test and then report back.

tmsd2001, 2017-10-09:
Here the debug log:
wait(500) original 2.2.0:

> load 0x4010f000, len 1384, room ⸮
> chksum 0x2d
> csum 0x2d
> v3de0c112
> ~ld
> 17186 MCO:BGN:INIT GW,CP=RNNGE---,VER=2.2.0-beta
> 17191 TSF:LRT:OK
> 17192 TSM:INIT
> 17194 TSF:WUR:MS=0
> 17201 TSM:INIT:TSP OK
> 17203 TSM:INIT:GW MODE
> 17205 TSM:READY:ID=0,PAR=0,DIS=0
> 17208 MCO:REG:NOT NEEDED
> scandone
> f 0, 17211 Eth: Failed to connect
> 17214 Eth: Failed to connect
> 17217 Eth: Failed to connect
> 17219 Eth: Failed to connect
> 17222 Eth: Failed to connect
> 17224 Eth: Failed to connect
> 17227 Eth: Failed to connect
> 17229 Eth: Failed to connect
> 17232 Eth: Failed to connect
> 17234 Eth: Failed to connect
> 17237 Eth: Failed to connect

wait(1000) after WIFI.begin

> load 0x4010f000, len 1384, room ~ld
> 17185 MCO:BGN:INIT GW,CP=RNNGE---,VER=2.2.0-beta
> 17190 TSF:LRT:OK
> 17191 TSM:INIT
> 17192 TSF:WUR:MS=0
> 17200 TSM:INIT:TSP OK
> 17202 TSM:INIT:GW MODE
> 17204 TSM:READY:ID=0,PAR=0,DIS=0
> 17207 MCO:REG:NOT NEEDED
> scandone
> f 0, 17210 Eth: Failed to connect
> 17213 Eth: Failed to connect
> 17216 Eth: Failed to connect
> 17218 Eth: Failed to connect
> 17221 Eth: Failed to connect
> 17223 Eth: Failed to connect
> 17226 Eth: Failed to connect
> 17228 Eth: Failed to connect
> 17231 Eth: Failed to connect
> 17233 Eth: Failed to connect
> 17236 Eth: Failed to connect
> 17238 Eth: Failed to connect
> 17241 Eth: Failed to connect
> 17244 Eth: Failed to connect
> 17246 Eth: Failed to connect
> 17249 Eth: Failed to connect
> 17251 Eth: Failed to connect

delay(1000) work with suspend

> load 0x4010f000, len 1384, room +⸮⸮j
> ~ld
> 17189 MCO:BGN:INIT GW,CP=RNNGE---,VER=2.2.0-beta
> 17194 TSF:LRT:OK
> 17195 TSM:INIT
> 17196 TSF:WUR:MS=0
> 17204 TSM:INIT:TSP OK
> 17206 TSM:INIT:GW MODE
> 17208 TSM:READY:ID=0,PAR=0,DIS=0
> 17211 MCO:REG:NOT NEEDED
> scandone
> f 0, scandone
> state: 0 -> 2 (b0)
> state: 2 -> 0 (0)
> .reconnect
> f -180, scandone
> state: 0 -> 2 (b0)
> state: 2 -> 0 (0)
> .reconnect
> f r0, scandone
> state: 0 -> 2 (b0)
> state: 2 -> 0 (0)
> .reconnect
> f 0, scandone
> state: 0 -> 2 (b0)
> state: 2 -> 0 (0)
> .reconnect
> f -180, scandone
> state: 0 -> 2 (b0)
> state: 2 -> 0 (0)
> .reconnect
> f r0, scandone
> state: 0 -> 2 (b0)
> state: 2 -> 0 (0)
> .reconnect
> f 0, scandone
> state: 0 -> 2 (b0)
> state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 5
> cnt 
> 
> connected with AP-TD-01, channel 1
> ip:192.168.0.246,mask:255.255.255.0,gw:192.168.0.254
> .IP: 192.168.0.246
> 24229 Eth: Failed to connect
> 24232 MCO:BGN:INIT OK,TSP=1
> 24239 Eth: Failed to connect
> 24242 TSF:MSG:READ,25-17-0,s=0,c=1,t=38,pt=7,l=5,sg=0:4.080
> 24256 Eth: Failed to connect
> 24259 TSF:MSG:READ,25-17-0,s=1,c=1,t=38,pt=3,l=2,sg=0:593
> 24268 Eth: Failed to connect

tmsd2001, 2017-10-09:
Can it be the NodeMCU have the wrong Firmware?

tmsd2001, 2017-10-31:
Hello, I did a few more tests.
I have not found the error yet.
But a few reasons for delay () instead of wait ().
During the WIFI startup, it is not good to jump into the send routine, wait () tries to send, although the connection is not established yet.
WIFI connects to the AP, but then can not connect to the controller, here must be the problem.
A WIFI trace says unexpected connection reset.