mysensors/MySensors #955: Debug prints portable commit causing build failure on Linux

madimadi, 2017-10-27:
Hi,
[The commit](https://github.com/mysensors/MySensors/commit/0ed558f1d077e68b69c2fe1cb0056c526ae3023e)
[https://github.com/mysensors/MySensors/commit/0ed558f1d077e68b69c2fe1cb0056c526ae3023e](The commit) looks to be causes compiler failure.

mfalkvidd, 2017-10-28:
I'm able to reproduce this error on Rasbpian Wheezy but not on Raspbian Stretch
Configure command: `./configure --extra-cxxflags="-DMY_DEBUG" && make`

Error on the following configuration:
```
gcc -v |& grep 'gcc version'
gcc version 4.7.2 (Debian 4.7.2-5+rpi1)
grep PRETTY_NAME /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"
```
No error, compile runs fine on the following configuration:
```
gcc -v |& grep 'gcc version'
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1)
grep PRETTY_NAME /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
```

tekka007, 2017-10-28:
Also no error on this configuration:
```
gcc -v |& grep 'gcc version'
gcc version 4.9.2 (Raspbian 4.9.2-10)
grep PRETTY_NAME /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
```

mfalkvidd, 2017-10-29:
This works on Wheezy/ gcc 4.7.2:
`./configure --extra-cxxflags="-DMY_DEBUG -D__STDC_FORMAT_MACROS" && make`

closed 2017-10-29 by tekka007