mysensors/MySensors #968: Please support ESP32

micropet, 2017-11-02:
Hello,

I have been using the ESP32 fork from tekka007 for months.
https://github.com/tekka007/MySensors/commits/ESP32HW
The software runs very stable on about 20 ESP32 boards.

I would be happy, if the ESP32 extension flows into the official sources.

Greetings Peter

tekka007, 2017-11-11:
@micropet thanks for your feedback. PR for dev: see #865.

closed 2018-04-07 by mfalkvidd

mfalkvidd, 2018-08-18:
@knopserl both [esp32](https://github.com/mysensors/MySensors/blob/15351cdd5b922480e9b3b0ec4060f62e2b0e89de/hal/architecture/ESP32/MyHwESP32.cpp#L94) and esp8266 will return MY_SLEEP_NOT_POSSIBLE when calling sleep and return immediately.

mfalkvidd, 2018-08-19:
@knopserl the light sleep mode f the esp32 consumes 0.8 mA (according to the datasheet) which means about 3 months battery life on 2xAA. That's less than 10% battery life compared to an atmega328-based node, so I'm not sure it makes sense to implement sleep using light sleep.

As you mentioned, the deep sleep will require a reboot at wakeup, which means the sleep function would not return. Such a radical change in sleep behavior would probably confuse a lot of users, and would require refactoring the structure of the sketches.

I'd say this rules out battery-powered nodes based on esp32. But if someone is eager enough to get it working and prepared to maintain document and it, we'd probably be able to merge code that supports sleep on esp32.