mysensors/MySensors #971: Configuration doesn't work with Rpi 1B+ and default value

pierrepinon, 2017-11-04:
Hello,

I was follow the tuto on official website to configure and install mysensors on my Rpi 1B+. 

When I configure the build before installation step : 
``
./configure --spi-driver=SPIDEV --spi-spidev-device=/dev/spidev0.0 --my-transport=nrf24 --my-rf24-irq-pin=15
``

The program return 
``
[SECTION] Detecting target machine.
./configure: line 111: warning: command substitution: ignored null byte in input
  [OK] machine detected: SoC=unknown, Type=unknown, CPU=armv6l.
[SECTION] Checking GPIO Sysfs.
  [OK] /sys/class/gpio/export found
[SECTION] Detecting init system.
  [OK] init system detected: systemd.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.
``

And the gateway doesn't work 
``
mysgw: Starting gateway...
mysgw: Protocol version - 2.1.1
mysgw: MCO:BGN:INIT GW,CP=RNNG--Q,VER=2.1.1
mysgw: TSM:INIT
mysgw: TSF:WUR:MS=0
mysgw: !TSM:INIT:TSP FAIL
mysgw: TSM:FAIL:CNT=1
mysgw: TSM:FAIL:PDT
mysgw: TSM:FAIL:RE-INIT
``

To fix it, I force the soc parameter : 
``
--spi-driver=SPIDEV --spi-spidev-device=/dev/spidev0.0 --my-transport=nrf24 --soc=BCM2835
``

Can you fix the configuration program to auto detect this device ?

Thx

marceloaqno, 2018-02-18:
@pierrepinon Could you please run a new test with the latest version?

marceloaqno, 2018-03-24:
@pierrepinon Could you please run the command below on your RPi 1B + and send the result?
```
cat /proc/cpuinfo
```

uberflyx, 2018-06-10:
I have a similar issue where everthing build fine but SPI comms doesn't happen on my orange pi zero. It works with armbian legacy kernel but does not work with mainline. I suspect it's something along the lines of whats in this thread: https://forum.odroid.com/viewtopic.php?f=115&t=25920

I get the following output:
Jun 10 21:30:12 INFO  Starting gateway...
Jun 10 21:30:12 INFO  Protocol version - 2.3.0-alpha
Jun 10 21:30:12 DEBUG MCO:BGN:INIT GW,CP=RNNGL-Q-,VER=2.3.0-alpha
Jun 10 21:30:12 DEBUG TSF:LRT:OK
Jun 10 21:30:12 DEBUG TSM:INIT
Jun 10 21:30:12 DEBUG TSF:WUR:MS=0
Jun 10 21:30:12 DEBUG !TSM:INIT:TSP FAIL
Jun 10 21:30:12 DEBUG TSM:FAIL:CNT=1
Jun 10 21:30:12 DEBUG TSM:FAIL:DIS
Jun 10 21:30:12 DEBUG TSF:TDI:TSL
Jun 10 21:30:22 DEBUG TSM:FAIL:RE-INIT
Jun 10 21:30:22 DEBUG TSM:INIT
Jun 10 21:30:22 DEBUG !TSM:INIT:TSP FAIL
Jun 10 21:30:22 DEBUG TSM:FAIL:CNT=2
Jun 10 21:30:22 DEBUG TSM:FAIL:DIS
Jun 10 21:30:22 DEBUG TSF:TDI:TSL
^CJun 10 21:30:23 NOTICE Received SIGINT

pierrepinon, 2018-06-18:
@marceloaqno I bought a new RPI3. I try to find my old RPI and send you the result of cpuinfo

marceloaqno, 2018-09-04:
@uberflyx the problem with spidev in mainline kernels was fixed in #1200 

According to the raspberry pi official [site](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md), the revision code for the RPi 1B+ is 0010 or 900032.

I did a quick test and both codes are detected as:
SoC=BCM2835, Type=rpi1, CPU=armv6l.
in the latest development branch, so I'm closing this issue.

@pierrepinon feel free to reopen if these revision codes are not the same as your old RPi.

closed 2018-09-04 by marceloaqno