mysensors/MySensors #980: Problem compiling mysgw on rpi3

tchoblond59, 2017-11-09:
Hi !

Seems like the same issue in #955 
When i try to compile from development branch i get this error
 `drivers/BCM/bcm2835.c: In function ‘bcm2835_delayMicroseconds’:
drivers/BCM/bcm2835.c:441:40: error: expected ‘)’ before ‘PRIu64’
   printf("bcm2835_delayMicroseconds %" PRIu64 "\n", micros);`

I am running a fresh raspbian 9.1. I also try `./configure --extra-cxxflags="-DMY_DEBUG -D__STDC_FORMAT_MACROS"`

gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1)

Am i missing something ?

Thanks

res1mp7q, 2017-11-09:
Same issue here running dev version 
error: expected ‘)’ before ‘PRIu64’
   printf("bcm2835_delayMicroseconds %" PRIu64 "\n", micros);

mfalkvidd, 2017-11-09:
@res1mp7q are you using the same gcc and rasbian versions as @tchoblond59 ?

MissionWebUK, 2017-11-10:
I'm running on a rpi3 model B fresh install of raspbian stretch lite: 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

Master branch had issues communicating with the nrf24 so I moved to dev branch. Same error as above but I didn't run the configure command that tchoblond59 ran, just the standard ethernet config

mfalkvidd, 2017-11-10:
Forum threads on this subject:
https://forum.mysensors.org/topic/7944/rpi-gateway-tsm-init-tsp-fail/7
https://forum.mysensors.org/topic/7955/can-t-get-past-first-base-arduino-serial-gateway/8

mfalkvidd, 2017-11-18:
Fix merged to development, will be included in upcoming 2.2.0 release

closed 2017-11-18 by mfalkvidd