mysensors/MySensors #1003: Water Meter Pulse Sensor - generates a lot of counts on one interrupt

alcor82, 2017-11-26:
Hello,
when i  run water meter pulse sensor manually it generates a lot of counts on one interrupt. 
But it should be generate only one. Shouldnt it?
It is not possible to define minimum and maximum pulse length to filter noise. I think it could be very useful.

LemarinelNet, 2017-11-27:
What kind of pulse counter device do you use ? 
If you are using a reed switch or a push button for test, there may be bouncing, which must be debounced (via software or hardware)
If you are using a hall effect sensors, there will be no bounce and so the counter should work correctly.

Tell us more about your testing hardware.

alcor82, 2017-11-27:
At this moment just push button. 
Will be connected to heat meter Actaris CF echo II with option card MBUSREP which uses galvanically isolated photo coupler, bipolar output. Pulse lenght is 250ms+-8%. 
I will let you know how works with this device.

LemarinelNet, 2017-11-27:
Ok, 
according to what you say and what I understand, this counter shouldn't be mechanic but electronic, so it shouldn't bounce.

mfalkvidd, 2017-12-23:
@alcor82 have you had a chance to verify using the heat meter?

alcor82, 2018-01-03:
Hello, yes. I tried and it was perfect.

closed 2018-01-03 by alcor82