mysensors/MySensors #1007: GatewaySerial not compiling

szymex73, 2017-11-30:
I am trying to compile GatewaySerial for Arduino Leonardo and it gives me this error:
```
In file included from /home/szymex/Arduino/libraries/MySensors/hal/architecture/AVR/MyHwAVR.cpp:20:0,
    from /home/szymex/Arduino/libraries/MySensors/MySensors.h:64,
    from /home/szymex/Arduino/libraries/MySensors/examples/GatewaySerial/GatewaySerial.ino:86:
/home/szymex/Arduino/libraries/MySensors/core/MyTransport.cpp: In function 'bool transportAssignNodeID(uint8_t)':
/home/szymex/Arduino/libraries/MySensors/hal/architecture/AVR/MyHwAVR.h:54:82: error: 'eeprom_update_byte' was not declared in this scope
 #define hwWriteConfig(__pos, __val) eeprom_update_byte((uint8_t*)(__pos), (__val))
                                                                                  ^
/home/szymex/Arduino/libraries/MySensors/core/MyTransport.cpp:503:3: note: in expansion of macro 'hwWriteConfig'
   hwWriteConfig(EEPROM_NODE_ID_ADDRESS, newNodeId);
   ^
/home/szymex/Arduino/libraries/MySensors/core/MyTransport.cpp: In function 'void transportSetRoute(uint8_t, uint8_t)':
/home/szymex/Arduino/libraries/MySensors/hal/architecture/AVR/MyHwAVR.h:54:82: error: 'eeprom_update_byte' was not declared in this scope
 #define hwWriteConfig(__pos, __val) eeprom_update_byte((uint8_t*)(__pos), (__val))
                                                                                  ^
/home/szymex/Arduino/libraries/MySensors/core/MyTransport.cpp:1065:2: note: in expansion of macro 'hwWriteConfig'
  hwWriteConfig(EEPROM_ROUTES_ADDRESS + node, route);
  ^
/home/szymex/Arduino/libraries/MySensors/core/MySensorsCore.cpp: In function 'void _begin()':
/home/szymex/Arduino/libraries/MySensors/hal/architecture/AVR/MyHwAVR.h:54:82: error: 'eeprom_update_byte' was not declared in this scope
 #define hwWriteConfig(__pos, __val) eeprom_update_byte((uint8_t*)(__pos), (__val))
                                                                                  ^
/home/szymex/Arduino/libraries/MySensors/core/MySensorsCore.cpp:140:2: note: in expansion of macro 'hwWriteConfig'
  hwWriteConfig(EEPROM_PARENT_NODE_ID_ADDRESS, MY_PARENT_NODE_ID);
  ^
/home/szymex/Arduino/libraries/MySensors/core/MySensorsCore.cpp: In function 'bool _processInternalMessages()':
/home/szymex/Arduino/libraries/MySensors/hal/architecture/AVR/MyHwAVR.h:56:114: error: 'eeprom_update_block' was not declared in this scope
 #define hwWriteConfigBlock(__buf, __pos, __length) eeprom_update_block((void*)(__buf), (void*)(__pos), (__length))
                                                                                                                  ^
/home/szymex/Arduino/libraries/MySensors/core/MySensorsCore.cpp:415:4: note: in expansion of macro 'hwWriteConfigBlock'
    hwWriteConfigBlock((void*)&_coreConfig.controllerConfig, (void*)EEPROM_CONTROLLER_CONFIG_ADDRESS,
    ^
/home/szymex/Arduino/libraries/MySensors/core/MySensorsCore.cpp: In function 'void saveState(uint8_t, uint8_t)':
/home/szymex/Arduino/libraries/MySensors/hal/architecture/AVR/MyHwAVR.h:54:82: error: 'eeprom_update_byte' was not declared in this scope
 #define hwWriteConfig(__pos, __val) eeprom_update_byte((uint8_t*)(__pos), (__val))
                                                                                  ^
/home/szymex/Arduino/libraries/MySensors/core/MySensorsCore.cpp:499:2: note: in expansion of macro 'hwWriteConfig'
  hwWriteConfig(EEPROM_LOCAL_CONFIG_ADDRESS+pos, value);
  ^
exit status 1```
How to fix it? I tried adding eeprom lib to those files but it didn't work

mfalkvidd, 2017-11-30:
Adding cross-reference to thread in the MySensors forum (in case the problem is discussed there https://forum.mysensors.org/topic/8045/problem-with-compiling-gatewayserial )

per1234, 2017-12-02:
Also posted at https://arduino.stackexchange.com/q/47200

tekka007, 2017-12-03:
@szymex73 Are you using the most recent Arduino IDE and AVR board defs? No issues here when compiling the GatewaySerial example for the Arduino Leonardo board, tested with the most recent master and dev branches.

szymex73, 2017-12-03:
I updated arduino IDE to 1.8.5, checked the Arduino IDE version of MySensors lib and the version from github repo and it still gives me this error

mfalkvidd, 2017-12-03:
The serial gateway compiles cleanly on my computer as well.

[image]
Maybe see if removing and re-instaling the board definitions helps?

szymex73, 2017-12-03:
It looks like it worked, I reinstalled board definitions and lib to be sure.
Thanks for everything

closed 2017-12-03 by szymex73