mysensors/MySensors #1010: Serial GW reboots when sending a message to an unreachable node

papylhomme, 2017-12-07:
Hello,

I'm building a Serial Gateway as instructed [here](https://www.mysensors.org/build/serial_gateway), but the gateway reboots when I try to send a message to a node, except when the node is online.

The gateway runs on an Arduino Nano powered via the USB connector, and the nrf24 module is powered using a 3.3v regulator (LM1085IT3.3), itself powered by the `5V` and `GND` pins on the Arduino.

Any ideas about this ?
 

GW log

    0;255;3;0;9;MCO:BGN:INIT GW,CP=RNNGA--,VER=2.1.1
    0;255;3;0;9;TSM:INIT
    0;255;3;0;9;TSF:WUR:MS=0
    0;255;3;0;9;TSM:INIT:TSP OK
    0;255;3;0;9;TSM:INIT:GW MODE
    0;255;3;0;9;TSM:READY:ID=0,PAR=0,DIS=0
    0;255;3;0;9;MCO:REG:NOT NEEDED
    0;255;3;0;14;Gateway startup complete.
    0;255;0;0;18;2.1.1
    0;255;3;0;11;Serial Gateway
    0;255;3;0;12;0.1
    0;0;0;0;6;Temperature
    0;1;0;0;7;Humidity
    0;255;3;0;9;MCO:BGN:STP
    0;255;3;0;9;MCO:BGN:INIT OK,TSP=1
    0;0;1;0;0;26.20
    0;1;1;0;1;44.20
    0;0;1;0;0;26.20
    0;1;1;0;1;44.70
    0;255;3;0;9;!TSF:MSG:SEND,0-0-1-1,s=1,c=1,t=48,pt=0,l=1,sg=0,ft=0,st=NACK:2
    0;255;3;0;9;MCO:BGN:INIT GW,CP=RNNGA--,VER=2.1.1
    0;255;3;0;9;TSM:INIT
    0;255;3;0;9;TSF:WUR:MS=0
    0;255;3;0;9;TSM:INIT:TSP OK
    0;255;3;0;9;TSM:INIT:GW MODE
    0;255;3;0;9;TSM:READY:ID=0,PAR=0,DIS=0
    0;255;3;0;9;MCO:REG:NOT NEEDED
    0;255;3;0;14;Gateway startup complete.
    0;255;0;0;18;2.1.1
    0;255;3;0;11;Serial Gateway
    0;255;3;0;12;0.1
    0;0;0;0;6;Temperature
    0;1;0;0;7;Humidity
    0;255;3;0;9;MCO:BGN:STP
    0;255;3;0;9;MCO:BGN:INIT OK,TSP=1
    0;0;1;0;0;26.20
    0;1;1;0;1;46.80
    0;0;1;0;0;26.30
    0;1;1;0;1;47.00

papylhomme, 2017-12-16:
I tried using separate power for the regulator, same results.

Configuration is now Arduino Nano powered via USB, and another USB port powering the nrf24 via a LM1085IT3.3 regulator

papylhomme, 2018-01-12:
Problem is solved using the dev branch

closed 2018-01-12 by papylhomme

asoltesz, 2018-02-15:
I also have this issue and it is very severe.

Switching to 2.2-alpha has solved it but I think this should be backported to 2.1.x

fallberg, 2018-02-16:
With 2.2.0 released, backporting to 2.1.x makes little sense. Just update to the latest release.