mysensors/MySensors #1017: wait function datatype

danielefrigo, 2017-12-20:
in this page [https://www.mysensors.org/download/sensor_api_20](url) wait function seems to accept an unsigned long to indicate the length of the delay in milliseconds.

I noticed some problems when using delays longer than 1 minute, so I  guess there's something in the code  using an unsigned int instead of an unsigned long.

I'm not very experienced, but in MySensorsCore.h I noticed that the wait function is declared as 

`void wait(const uint32_t waitingMS);`

mfalkvidd, 2017-12-21:
Thanks for reporting this @danielefrigo 
Could you expand on how you got to this conclusion? What code did you use? What Arduino board are you using? Which version of MySensors?

I am unable to reproduce this. My code (I added it to the GatewayESP8266 example)
```
void loop()
{
  testWait(500);
  testWait(1000);
  testWait(2000);
  testWait(8000);
  testWait(30000);
  testWait(60000);
  testWait(120000);
  testWait(300000);
}

void testWait(unsigned long waitTime) {
  uint32_t startTime = millis();
  wait(waitTime);
  Serial.print("wait took ");
  Serial.print(millis() - startTime);
  Serial.println (" milliseconds");
}
```
Result:
```
14262 MCO:BGN:INIT OK,TSP=NA
wait took 500 milliseconds
wait took 1000 milliseconds
wait took 2000 milliseconds
pm open,type:2 0
wait took 8000 milliseconds
wait took 30000 milliseconds
wait took 60000 milliseconds
wait took 120000 milliseconds
wait took 300000 milliseconds
```
I tested with MySensors at version 2.2.0-rc.2.

danielefrigo, 2017-12-21:
I'm using MySensors Library v2.1.1 an Arduino nano (Chinese clone).
I simply noticed that when in my loop I added a wait shorter than 60000 ms, everything was working fine.
As soon as I increased it, I started having random behaviors (let's say, sensor value reported every 10'' instead of 10' or reported only once).

I'm really new to MySensors, so I can surely have misunderstood something.

mfalkvidd, 2017-12-21:
Strange. Could you post a sketch that reproduces this, and the corresponding debug output?

This is the result I get on Arduino Nano (based on the Gateway Serial example, just disabling radio, replacing loop and adding testWait as before) with MySensors at version 2.1.1:
```
0;255;3;0;9;MCO:BGN:INIT GW,CP=R-NGA--,VER=2.1.1
0;255;3;0;14;Gateway startup complete.
0;255;0;0;17;2.1.1
0;255;3;0;9;MCO:REG:NOT NEEDED
0;255;3;0;9;MCO:BGN:INIT OK,TSP=NA
wait took 500 milliseconds
wait took 1000 milliseconds
wait took 2000 milliseconds
wait took 8000 milliseconds
wait took 30000 milliseconds
wait took 60000 milliseconds
wait took 120000 milliseconds
```
(I'll add more output when it arrives, but it will take a while :))

danielefrigo, 2017-12-22:
Sorry for the unfamiliar notation!
' = minutes
'' = seconds
This is the sketch I'm using:
`
// Enable debug prints to serial monitor
#define MY_DEBUG

// Enable and select radio type attached
#define MY_RADIO_NRF24
#define MY_RF24_PA_LEVEL   RF24_PA_HIGH
#define MY_RF24_DATARATE   RF24_250KBPS
#define MY_RF24_CHANNEL 125

#define MY_OTA_FIRMWARE_FEATURE
#define MY_REPEATER_FEATURE

#define MY_NODE_ID 5
//#define MY_PARENT_NODE_ID 0
//#define MY_PARENT_NODE_IS_STATIC

#define TEMP_ID 1
#define HUM_ID 2
#define RELAY_ID 3
#define RELAY_ID_C 4

#define TEMP_PIN 8

#define RELAY_PIN 2
#define RELAY_ON 1  // GPIO value to write to turn on attached relay
#define RELAY_OFF 0 // GPIO value to write to turn off attached relay

#include <MySensors.h>

#include <DHT.h>
#define DHTTYPE DHT22   // DHT 22  (AM2302)
DHT dht(TEMP_PIN, DHTTYPE); //// Initialize DHT sensor for normal 16mhz Arduino

MyMessage msgTEMP(TEMP_ID, V_TEMP);
MyMessage msgHUM(HUM_ID, V_HUM);
MyMessage msgRELAY(RELAY_ID_C, V_STATUS);

static const uint8_t FORCE_UPDATE_N_READS = 4;

float temp;
float temp_offset = -1.5;
float last_temp; //per forzare primo invio
unsigned int nNoUpdatesTEMP;

float hum;
float last_hum; //per forzare primo invio
unsigned int nNoUpdatesHUM;

unsigned int relay;
unsigned int last_relay;
unsigned int nNoUpdatesRELAY;

void before()
{
  pinMode(RELAY_PIN, OUTPUT);
  digitalWrite(RELAY_PIN, loadState(RELAY_ID)?RELAY_ON:RELAY_OFF);

  pinMode(TEMP_PIN, INPUT);
  digitalWrite(TEMP_PIN, LOW);
}

void setup() {

  last_temp = 0;
  last_hum = 0;
  last_relay = -1;

  nNoUpdatesTEMP = 0;
  nNoUpdatesHUM = 0;
  nNoUpdatesRELAY = 0;

}

void presentation()
{
  sendSketchInfo("P2 - Termostato", "1.2");

  present(TEMP_ID, S_TEMP);
  present(HUM_ID, S_HUM);
  present(RELAY_ID, S_BINARY);
  present(RELAY_ID_C, S_BINARY);

}

void loop() {

  temp = dht.readTemperature();
  hum = dht.readHumidity();

  if (temp != last_temp || nNoUpdatesTEMP == FORCE_UPDATE_N_READS) {
    last_temp = temp;
    nNoUpdatesTEMP = 0;
    send(msgTEMP.set(temp,1));

    #ifdef MY_DEBUG
    Serial.print("T: ");
    Serial.println(temp);
    #endif
  } else {
    nNoUpdatesTEMP++;
  }

  if (hum != last_hum || nNoUpdatesHUM == FORCE_UPDATE_N_READS) {
    last_hum = hum;
    nNoUpdatesHUM = 0;
    send(msgHUM.set(hum,1));

    #ifdef MY_DEBUG
    Serial.print("H: ");
    Serial.println(hum);
    #endif
  } else {
    nNoUpdatesHUM++;
  }

  if (relay != last_relay || nNoUpdatesRELAY == FORCE_UPDATE_N_READS) {
    last_relay = relay;
    nNoUpdatesRELAY = 0;
    send(msgRELAY.set(relay));

    #ifdef MY_DEBUG
    Serial.print("R: ");
    Serial.println(relay);
    #endif
  } else {
    nNoUpdatesRELAY++;
  }

  wait(900000);

}

void receive(const MyMessage &message)
{
  Serial.println("*** Receiving ***");

  if (message.type == V_STATUS && message.sensor == RELAY_ID) {
    // Change relay state
    digitalWrite(RELAY_PIN, message.getBool()?RELAY_ON:RELAY_OFF);
    relay = message.getBool()?RELAY_ON:RELAY_OFF;

    // Store state in eeprom
    saveState(message.sensor, message.getBool());
    
    send(msgRELAY.set(relay));
    last_temp = temp;

    // Write some debug info
    #ifdef MY_DEBUG
    Serial.print("Incoming change for sensor:");
    Serial.print(message.sensor);
    Serial.print(", New status: ");
    Serial.println(message.getBool());
    #endif
  }
  
}

`

mfalkvidd, 2017-12-22:
Thanks.
The sketch looks good to me. Could you post the debug output?

danielefrigo, 2017-12-22:
Per my (little) experience, the script is good: if I change the wait(900000) into a loop with 15 cycles of wait(60000) I have no issues (it's running this way in this moment, and after 2 days I've never lost a message).

I'm at work in this moment, I can try retrieving the serial monitor output this evening or tomorrow.

mfalkvidd, 2018-01-03:
@danielefrigo have you had a chance to capture the debug output?

danielefrigo, 2018-01-03:
I didn't have much time, I tried quickly but I've not been able to
reproduce the issue.

You can close the issue for the time being, I'll open it again if I manage
to reproduce it again.

mfalkvidd, 2018-01-03:
Sounds good. Thanks for replying.

closed 2018-01-03 by mfalkvidd