mysensors/MySensors #1022: Option to disable `isatty` check for Linux gateway

ConnorRigby, 2017-12-29:
I'm developing a controller in Erlang/Elixir, And i would like to use the Raspberry Pi SPI bus directly, rather than using an external Arduino gateway. I've hit a small problem where spawning programs from the Erlang Beam VM fails the `isatty` check, so i don't get logs from `mysgw`.  

https://github.com/mysensors/MySensors/blob/development/hal/architecture/Linux/MyMainLinux.cpp#L415

mfalkvidd, 2017-12-29:
@ConnorRigby what configure command did you use when configuring mysgw?

Does --my-serial-is-pty help?

mfalkvidd, 2017-12-29:
I used https://raw.githubusercontent.com/skuhl/sys-prog-examples/master/simple-examples/isatty.c to test the following:
```
pi@raspberrypi:~ $ ./a.out
stdin is a terminal.
stdout is a terminal.
stderr is a terminal.
pi@raspberrypi:~ $ touch /tmp/file
pi@raspberrypi:~ $ ./a.out < /tmp/file
stdin is not a terminal.
stdout is a terminal.
stderr is a terminal.
root@raspberrypi:/home/pi# ./a.out < /dev/ttyUSB42
stdin is a terminal.
stdout is a terminal.
stderr is a terminal.
```

mysgw configuration:
`./configure --my-transport=none --my-gateway=serial --my-serial-is-pty --my-serial-pty=/dev/ttyUSB42`

So it seems like --my-serial-is-pty does the trick.

ConnorRigby, 2017-12-29:
Will try that out right now!

ConnorRigby, 2017-12-29:
Unfortunately that still did not work. I'm using the Network Gateway by the way. I thought `--my-serial-is-pty` was only for a `serial` gateway
```bash
./configure \
--c_compiler=/home/connor/.nerves/artifacts/nerves_toolchain_armv6_rpi_linux_gnueabi-0.12.1.linux-x86_64/bin/armv6-rpi-linux-gnueabi-gcc\
--cxx_compiler=/home/connor/.nerves/artifacts/nerves_toolchain_armv6_rpi_linux_gnueabi-0.12.1.linux-x86_64/bin/armv6-rpi-linux-gnueabi-g++ \
--my-transport=nrf24 \
--my-config-file=/tmp/mysensors.dat \
--my-debug=enable\ 
--bindir=/home/connor/oss/home_controller/deleteme/priv/my_sensors \
--build-dir=/home/connor/oss/home_controller/deleteme/_build/my_sensors \
--prefix=/home/connor/oss/home_controller/deleteme/priv/my_sensors \
--cpu-flags="-march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp" --extra-cflags="-DLINUX_ARCH_RASPBERRYPI" \
--extra-cxxflags="-std=c++98" --soc="BCM2835" \
--spi-driver="BCM"

mfalkvidd, 2017-12-29:
ah, now I understand. You want an option to let the gateway output debug data on the TCP stream, intermingled with the MySensors API data?

To my knowledge, that would break all controllers. Doesn't seem like a good idea.

What prevents you from using syslog for the debug data?

ConnorRigby, 2017-12-29:
sorry. I'm not trying to use any serial data at all. Just using the Ethernet Gateway mode. I just don't get any output (debug messages, not node/sensor data) that gets printed.
so messages that say `mysgw: some useful information`. 

Im not trying to put debug data on the TCP stream, only sensor data gets exchanged there.

mfalkvidd, 2017-12-29:
In that case, I don't think the isatty if clause is relevant. It only prevents the gateway from printing debug logs.
What console output did you get when you [tested](https://www.mysensors.org/build/raspberry#test) the gateway?

ConnorRigby, 2017-12-29:
With the above ^ `configure` the only output i get is the output of `--help` because it uses `printf`
and `mysgw: Config file /tmp/mysensors.dat does not exist, creating new config file.`
(because it uses `printf`)

if i apply this patch:
```
diff --git a/c_src/MySensors/hal/architecture/Linux/MyMainLinux.cpp b/c_src/MySensors/hal/architecture/Linux/MyMainLinux.cpp
index f67ea2f..fc0444b 100644
--- a/c_src/MySensors/hal/architecture/Linux/MyMainLinux.cpp
+++ b/c_src/MySensors/hal/architecture/Linux/MyMainLinux.cpp
@@ -412,7 +412,7 @@ int main(int argc, char *argv[])
 	}

 	log_opts = LOG_CONS;
-	if (foreground && isatty(STDIN_FILENO)) {
+	if (foreground) {
 		// Also print syslog to stderror
 		log_opts |= LOG_PERROR;
 	}

```

all is fine. without that patch, everything works great, i just don't get any debug logs.

ConnorRigby, 2017-12-29:
im running `mysgw -d` btw

mfalkvidd, 2017-12-29:
Sorry for the confusion. Seems like I'm in way over my head here. Hopefully someone more qualified can assist you.

ConnorRigby, 2017-12-29:
Thanks for the help! Sorry this is kind of a strange use case i think.

mfalkvidd, 2017-12-29:
The check was inserted in 09103e9c042eb4c0c90b7ecba8227baaf9f78b63 but I can't see why the check is done. (Is it needed at all? What does it prevent?) Maybe @marceloaqno remembers?

marceloaqno, 2017-12-29:
isatty() is used to check if mysgw is running from a terminal, which apparently isn't @ConnorRigby case.

I added it to prevent the gateway from printing to stderr when it ran as a service.

I think we should add a option to disable the check.

ConnorRigby, 2017-12-29:
An option to disable it would be cool. You're right, im not running inside a "real" terminal.

Somewhat off topic, but is there a way of getting _only_ packet info on stdin/stdout of `mysgw`.
so packets the gateway would write packets out to stdout, and read packets in on stdin. This would remove a layer of complexity from my project if possible. It looks like all the pieces are there, (SerialSimulator, etc) things just need to be wired up in a probably really weird way

marceloaqno, 2017-12-30:
@ConnorRigby I'm not sure if I got it right, a way for the gateway to communicate with the controller using **stdin/stdout** to send and receive packets? Instead of using ethernet?

ConnorRigby, 2017-12-30:
Yes. For the project im working on, im starting up `mysgw`, then connecting to `localhost:5003` over tcp sockets. It would be nice to skip all the socket stuff, and just use `mysgw`'s **stdin/stdout** directly. That is out of scope of this question anyway. I was just wondering if that was something that had already been done. Guess its another strange case :)

marceloaqno, 2017-12-30:
You could try to configure the mysgw as a serial gateway:
`./configure --my-gateway=serial ...`

and make the following changes so that the SerialSimulator class is used for stdin and stdout operations:
```
--- a/hal/architecture/Linux/MyHwLinuxGeneric.h
+++ b/hal/architecture/Linux/MyHwLinuxGeneric.h
@@ -25,15 +25,7 @@
 #include "SerialPort.h"
 #include "SerialSimulator.h"

-#ifdef MY_GATEWAY_SERIAL
-#ifdef MY_LINUX_IS_SERIAL_PTY
-SerialPort Serial = SerialPort(MY_LINUX_SERIAL_PTY, true);
-#else
-SerialPort Serial = SerialPort(MY_LINUX_SERIAL_PORT);
-#endif
-#else
 SerialSimulator Serial = SerialSimulator();
-#endif
```

You may need to make some adjustments on the read() of the SerialSimulator class

Also, the communications between the gateway and the controller will use the serial protocol defined here:
https://www.mysensors.org/download/serial_api_20

mfalkvidd, 2018-03-23:
Fixed by #1061

closed 2018-03-23 by mfalkvidd