mysensors/MySensors #1034: [FEATURE REQUEST] Support RF (RFM69) decoder and parser for IKEA Sparsnas electricity meter (sensor)?

Hedda, 2018-01-23:
Would it be possible to add this decoder for IKEA Sparsnäs (Ikea Sparnas electricity/energy meter) sensor?

There is already an existing project for ESP8266 that does this with RFM69 but wondering if decoder code could be ported to MySensors?

https://github.com/bphermansson/EspSparsnasGateway

Some more information and links are posted here which I copy below:

http://www.letscontrolit.com/forum/viewtopic.php?f=10&t=3726

+1 request for IKEA Sparsnäs (Ikea Sparsnas) electricity meter with wireless display

http://www.ikea.com/se/sv/catalog/products/90292645/

At least in Sweden these are often on sale every so often with a large rebate via IKEA FAMILY and as such they can also be found relativly cheap from rebate scalpers on eBay (and its Tradera action-site in Sweden) following closely aften a sale.

http://www.ikea.com/se/sv/manuals/sparsnas-energidisplay__AA-1201480-1.pdf

"[i]SPARSNÄS energy display assists you to monitor the total usage of electricity in your home. The energy display consists of a transmitter and a display. The transmitter, which you attach to your electricity meter, continuously registers the usage of electricity and transfers the data wireless to the display. Place the display where it becomes a natural part of daily life in your home"

Again, there is an open source RTL-SDR decoder and parser for it:

https://github.com/strigeus/sparsnas_decoder

"SPARSNÄS uses a CC1101 chip configured for GFSK modulation at 868MHz. The two FSK symbol frequencies once downconverted to baseband are roughly 67kHz and 105kHz. (EDIT: It seems like with a different RTL-SDR the baseband frequencies are 12.5khz and 50khz. Maybe my first RTL-SDR was inaccurate). The packet length is 18 bytes, including the length byte, and a 16-bit CRC is appended to the end. The CC1101 sync word is 0xD201. The packet payload byte 5 to 17 are encrypted using a key derived from the sender device's ID. The encryption key is a repeating XOR key"

tubalainen also has a fork of the above RTL-SDR project which convert and output as JSON to be used to pipe data to MQTT broker:

https://github.com/tubalainen/sparsnas_decoder

"The packet data is encrypted using your sender's ID. The sender ID is the last 6 digits of the serial number located under the battery."

"A BIG thanks to the https://www.facebook.com/groups/SHgruppen on Facebook"

Hedda, 2018-01-25:
@mfalkvidd Tip is that Ikea does currently have Sparnäs on sale for only 99SEK in some selected stores:

https://www.sweclockers.com/forum/trad/1506976-tradlos-display-for-99-pa-ikea

These types on tips on sale of Ikea Sparnäs have recently been posted often in this other thread:

http://elektronikforumet.com/forum/viewtopic.php?f=2&t=85006

It is also in that thread that some one the developers working on Sparnäs based solutions have posted.

Hedda, 2018-01-25:
To reverse this suggestion;  here is code projects to instead send a signal to Sparnäs wireless display:

https://github.com/MagnusThome/sparsnas-esp8266-01

https://github.com/MagnusThome/sparnas-raspberry

Hedda, 2018-01-30:
FYI; here are more detailed information information on how exactly Sparnäs work with power-meters :

https://github.com/kodarn/Sparsnas

@kodarn include a "LedBlinkerHelperToo" to emulate power-meter to help develop without Sparnäs.

Other people have also created similar LED blinkers for for it to do send a value to Sparnäs display, ex: 

https://github.com/MagnusThome/sparsnas-esp8266-01
https://github.com/MagnusThome/sparnas-raspberry

This and other projects for the Sparnäs transmitter and the receiving display in this Swedish forum thread:

http://elektronikforumet.com/forum/viewtopic.php?f=2&t=85006

I think that Sparnäs is still very interesting to people because its really cheap and profesionally buildt.

fallberg, 2018-02-25:
Thank you for your suggestion. However, the Sparsnäs uses it's own RF protocol, and it is incompatible with MySensors which uses it's own protocol. There is currently no plan to support multiple RF protocols on the same radio modules in the MySensors framework.
However, if you are interested in developing and _maintaining_ such support, you are welcome to file a pull request.

closed 2018-02-25 by fallberg