mysensors/MySensors #1045: cppcheck warnings

mfalkvidd, 2018-01-28:
Are these something we should fix?
```
$ find . -name "*.h" -print0 | xargs -0 .mystools/cppcheck/run.sh
[core/MySensorsCore.h:95]: (style) struct member 'controllerConfig_t::isMetric' is never used.
[core/MySensorsCore.h:104]: (style) struct member 'coreConfig_t::nodeRegistered' is never used.
[core/MySensorsCore.h:105]: (style) struct member 'coreConfig_t::presentationSent' is never used.
[core/MySensorsCore.h:106]: (style) struct member 'coreConfig_t::reserved' is never used.
[core/MySigning.h:598]: (style) struct member 'whitelist_entry_t::nodeId' is never used.
[core/MySigning.h:599]: (style) struct member 'whitelist_entry_t::serial' is never used.
[core/MyTransport.h:236]: (style) struct member 'transportConfig_t::nodeId' is never used.
[core/MyTransport.h:237]: (style) struct member 'transportConfig_t::parentNodeId' is never used.
[core/MyTransport.h:238]: (style) struct member 'transportConfig_t::distanceGW' is never used.
[core/MyTransport.h:239]: (style) struct member 'transportConfig_t::passiveMode' is never used.
[core/MyTransport.h:240]: (style) struct member 'transportConfig_t::reserved' is never used.
[core/MyTransport.h:268]: (style) struct member 'transportSM_t::currentState' is never used.
[core/MyTransport.h:269]: (style) struct member 'transportSM_t::stateEnter' is never used.
[core/MyTransport.h:271]: (style) struct member 'transportSM_t::lastUplinkCheck' is never used.
[core/MyTransport.h:273]: (style) struct member 'transportSM_t::findingParentNode' is never used.
[core/MyTransport.h:274]: (style) struct member 'transportSM_t::preferredParentFound' is never used.
[core/MyTransport.h:275]: (style) struct member 'transportSM_t::uplinkOk' is never used.
[core/MyTransport.h:276]: (style) struct member 'transportSM_t::pingActive' is never used.
[core/MyTransport.h:277]: (style) struct member 'transportSM_t::transportActive' is never used.
[core/MyTransport.h:278]: (style) struct member 'transportSM_t::stateRetries' is never used.
[core/MyTransport.h:280]: (style) struct member 'transportSM_t::failedUplinkTransmissions' is never used.
[core/MyTransport.h:281]: (style) struct member 'transportSM_t::failureCounter' is never used.
[core/MyTransport.h:282]: (style) struct member 'transportSM_t::msgReceived' is never used.
[core/MyTransport.h:284]: (style) struct member 'transportSM_t::pingResponse' is never used.
[core/MyTransport.h:285]: (style) struct member 'transportSM_t::uplinkQualityRSSI' is never used.
[core/MyTransport.h:292]: (style) struct member 'routingTable_t::route' is never used.
[drivers/ATSHA204/sha256.h:11]: (style) union member '_buffer::b' is never used.
[drivers/ATSHA204/sha256.h:12]: (style) union member '_buffer::w' is never used.
[drivers/ATSHA204/sha256.h:15]: (style) union member '_state::b' is never used.
[drivers/ATSHA204/sha256.h:16]: (style) union member '_state::w' is never used.
[drivers/AVR/DigitalIO/attic/MCP355X.h:12]: (style) The class 'MCP355X' does not have a constructor.
[drivers/RFM69/new/RFM69_new.h:304]: (style) struct member 'rfm69_internal_t::address' is never used.
[drivers/RFM69/new/RFM69_new.h:306]: (style) struct member 'rfm69_internal_t::txSequenceNumber' is never used.
[drivers/RFM69/new/RFM69_new.h:307]: (style) struct member 'rfm69_internal_t::powerLevel' is never used.
[drivers/RFM69/new/RFM69_new.h:308]: (style) struct member 'rfm69_internal_t::ATCtargetRSSI' is never used.
[drivers/RFM69/new/RFM69_new.h:311]: (style) struct member 'rfm69_internal_t::dataSent' is never used.
[drivers/RFM69/new/RFM69_new.h:312]: (style) struct member 'rfm69_internal_t::dataReceived' is never used.
[drivers/RFM69/new/RFM69_new.h:313]: (style) struct member 'rfm69_internal_t::ackReceived' is never used.
[drivers/RFM69/new/RFM69_new.h:314]: (style) struct member 'rfm69_internal_t::ATCenabled' is never used.
[drivers/RFM69/new/RFM69_new.h:315]: (style) struct member 'rfm69_internal_t::listenModeEnabled' is never used.
[drivers/RFM69/old/RFM69_old.h:140]: (warning) Member variable 'RFM69::_SPCR' is not initialized in the constructor.
[drivers/RFM69/old/RFM69_old.h:140]: (warning) Member variable 'RFM69::_SPSR' is not initialized in the constructor.
[drivers/RFM69/old/RFM69_old.h:140]: (warning) Member variable 'RFM69::_SREG' is not initialized in the constructor.
[drivers/RFM95/RFM95.h:288]: (style) struct member 'rfm95_internal_t::address' is never used.
[drivers/RFM95/RFM95.h:290]: (style) struct member 'rfm95_internal_t::txSequenceNumber' is never used.
[drivers/RFM95/RFM95.h:291]: (style) struct member 'rfm95_internal_t::powerLevel' is never used.
[drivers/RFM95/RFM95.h:292]: (style) struct member 'rfm95_internal_t::ATCtargetRSSI' is never used.
[drivers/RFM95/RFM95.h:295]: (style) struct member 'rfm95_internal_t::cad' is never used.
[drivers/RFM95/RFM95.h:296]: (style) struct member 'rfm95_internal_t::ATCenabled' is never used.
[drivers/RFM95/RFM95.h:297]: (style) struct member 'rfm95_internal_t::ackReceived' is never used.
[drivers/RFM95/RFM95.h:298]: (style) struct member 'rfm95_internal_t::dataReceived' is never used.
[drivers/RFM95/RFM95.h:299]: (style) struct member 'rfm95_internal_t::reserved' is never used.
[hal/architecture/NRF5/MyHwNRF5.h:175]: (style) struct member 'nrf_ecb_t::key' is never used.
[hal/architecture/NRF5/MyHwNRF5.h:178]: (style) struct member 'nrf_ecb_t::cleartext' is never used.
[hal/architecture/NRF5/MyHwNRF5.h:181]: (style) struct member 'nrf_ecb_t::ciphertext' is never used.
```

tekka007, 2018-01-28:
nah, looks like false alarm - they are in use. How did you get this list?

mfalkvidd, 2018-01-28:
@tekka007 look at the first line :-)

tekka007, 2018-01-28:
mkay, then cppcheck needs adjustments - good catch 👍

mfalkvidd, 2018-01-28:
Could be me running the wrong settings. Jenkins doesn't complain. I'm ok with closing this.

closed 2018-01-29 by mfalkvidd