mysensors/MySensors #1051: Add more configuration for new RFM69 driver

MiloshCZ, 2018-02-01:
Hi,
is it possible to add some more configuration options for new RFM69 driver?

For example:
```
MY_RFM69_FDEVLSB
MY_RFM69_FDEVMSB
MY_RFM69_REG_RXBW
```
Thanks

mfalkvidd, 2018-02-01:
Could you describe the value people would get from these settings, and/or which use cases would be helped? (So we can add proper documentation on how and when to use them)

tekka007, 2018-02-01:
@MiloshCZ Have a look at #1049, there I'm exposing these settings:

see here: https://github.com/mysensors/MySensors/pull/1049/files#diff-cf14b6301beb3a4dbaded9b95bb190feR772
and here: https://github.com/mysensors/MySensors/pull/1049/files#diff-37d5dce0f1c49c4798c609cd5749b07eR214

MiloshCZ, 2018-02-02:
@tekka007 Yes, it is better than expected!

PR #1049 (2018-02-18): RFM69: Updates and optimizations
- use IRQ flags
- reduce RAM usage (-50 bytes)
- reduce flash usage (-200 bytes)
- minor speed improvement
- implement CSMA
- Fixes #1051, #824
closed 2018-02-18 by fallberg