mysensors/MySensors #1055: 2.2. wrong instructions for security from Rpi

sineverba, 2018-02-04:
I use the software security and signing.

Launching `sudo mysgw --set-soft-hmac-key=CA425435D5765F2A96B97A10B38` Rpi gives following instructions:

```
The next line is intended to be used in SecurityPersonalizer.ino:
#define MY_SOFT_HMAC_KEY	0XCA,0X42,0X54,0X35,0X5F,0X2A,0XB,0X38
```

In SecurityPersonalizer.ino there is no MY_SOFT_HMAC_KEY but `#define MY_HMAC_KEY`

Seems need to edit PI/Linux instructions.

referrer also with this topic: https://forum.mysensors.org/topic/8968/wrong-instructions-on-gateway-for-personalizer

Have a nice day!

pepeEL, 2018-02-04:
Hi
Please help me. Is any place where i can found all info step by step how secure my gateway on RPi3 and nodes on Arduino MIni Pro with radio RFM69HW ? Please

fallberg, 2018-02-06:
Fixed in #1056

fallberg, 2018-02-06:
@pepeEL It is currently not implemented. PR #1056 implements support for activating encryption on RFM69HW but it is untested. Please test.

commit 8a1f5f4c (2018-02-11): Revamp the simple password flag and rPi fixes (#1056)

MY_SIGNING_SIMPLE_PASSWD is now
called MY_SECURITY_SIMPLE_PASSWD.
MY_SIGNING_SIMPLE_PASSWD only affects signing,
and a new flag, MY_ENCRYPTION_SIMPLE_PASSWD
only affects encryption.
MY_SECURITY_SIMPLE_PASSWD enable both these flags.

The rPi port has been improved to support more
security settings and it is now also possible to
enable encryption for RFM69 transport.

Documentation has been improved to provide a
"Security" section under "Library customisation"
which has a table that summarize all security
related configuration flags, and th [... cut]
commit c18f472a (2018-06-18): Revamp the simple password flag and rPi fixes (#1056)

MY_SIGNING_SIMPLE_PASSWD is now
called MY_SECURITY_SIMPLE_PASSWD.
MY_SIGNING_SIMPLE_PASSWD only affects signing,
and a new flag, MY_ENCRYPTION_SIMPLE_PASSWD
only affects encryption.
MY_SECURITY_SIMPLE_PASSWD enable both these flags.

The rPi port has been improved to support more
security settings and it is now also possible to
enable encryption for RFM69 transport.

Documentation has been improved to provide a
"Security" section under "Library customisation"
which has a table that summarize all security
related configuration flags, and th [... cut]
closed 2018-02-11 by tekka007