mysensors/MySensors #1057: Splash screen not redirected to MY_DEBUGDEVICE

tekka007, 2018-02-07:
See forum:
https://forum.mysensors.org/topic/8989/splash_screen-with-my_debugdevice

nelsonov, 2018-10-08:
I have a fix for this.  It's only a couple lines in MySplashScreen.cpp.  However, in working on this I uncovered something related.  displaySplashScreen is called before the sketch's setup().  If someone is using a soft serial port, that port will not be started until after the splash screen is displayed.  This makes the splash screen disappear into a black hole.

Should I start a PR for this issue, #1057, as originally stated?

Should I create a separate issue for the problem of the order of displaySplashScreen and setup()?

nelsonov, 2018-10-09:
As noted in #1207 , if using a software serial device as MY_DEBUGDEVICE, then it needs to initialized in preHwInit() for the splash screen to work.  Later functions like before() and setup() happen after the splash screen has already been displayed.

```
void preHwInit()
{
  SoftSerial.begin(9600);
}
```