mysensors/MySensors #1066: SAMD device cannot clear external i2c eeprom

tbowmo, 2018-02-18:
Reported on the forum: 
https://forum.mysensors.org/topic/8814/cleareprom-does-not-work-for-sensebender-gw

When clearing eeprom, operation is halted.

tbowmo, 2018-02-18:
Writes is halted in SERCOM module under arduino, when writing single bytes to the eeprom.

tbowmo, 2018-02-18:
We are not initializing HW propperly, when using MY_CORE_ONLY, so I2C bus is not working

GregerLindblom, 2018-03-04:
Seems to still be an issue when running SensebenderGatewaySerial sketch from development branch. Execution halts when `Wire.endTransmission();` is hit in extEEPROM.cpp.

I2C hardware probably still not initialized correctly.

pbolduc, 2019-07-24:
I am using a [Moteino M0](https://lowpowerlab.com/guide/moteino/moteinom0/). A Moteino M0 uses a [ATSAMD21G18](https://www.microchip.com/wwwproducts/en/ATsamd21g18) processor. When I try to run a simple serial gateway, I found during initialization when MySensors tries to write the parent node id to eeprom, the process hanges.  I put debug statements before and after each line to isolate the place it hung.   According to the [SAM D21 datasheet](https://cdn.sparkfun.com/datasheets/Dev/Arduino/Boards/Atmel-42181-SAM-D21_Datasheet.pdf), the eeprom is emulated.  Here is the line where it hangs,

https://github.com/mysensors/MySensors/blob/3ae4fb657f82ec7c10a22a8dd8c334f9d8f1f4b4/core/MySensorsCore.cpp#L165

I also found this [arduino issue](https://github.com/arduino/ArduinoCore-samd/issues/145 ) related to the eeprom on SAMD. 

I think the current SAMD configuration was developed around the [Sensebender-Gateway](https://www.openhardware.io/view/12/Sensebender-Gateway). Based on the [BOM](https://www.openhardware.io/view/12/Sensebender-Gateway#tabs-bom), the Sensebender-Gateway has an extra "32Kb (4096x8) Serial EEPROM with Unique Serial Number"

In my local copy of MySensors, I #ifdef wrapped the calls to hwRead/hwWrite and the gateway runs as expected.  In the MyHwSAMD.h, there is currently no way to opt out of the external EEPROM,

https://github.com/mysensors/MySensors/blob/3ae4fb657f82ec7c10a22a8dd8c334f9d8f1f4b4/hal/architecture/SAMD/MyHwSAMD.h#L48-L50

In my case, I can think of the following options for persisting the configuration,

1. Use the emulated eeprom
2. Use external flash memory
3. Do not persist the configuration

I would like some help coming up with a proposed solution that can be merged into the main MySensors.  The very first thing that comes to mind is adding a new define,

```cpp
// if defined, the use of external eeprom will be disabled,
// calls to read / write configuration not occur
#define MY_EXT_EEPROM_DISABLED
```

PR #1070 (2018-02-20): Fix MY_CORE_ONLY: Initialize HW
- Fixes #1066