mysensors/MySensors #1077: RPI3 GW poor RSSI on RFM69HW

SimoneBnc, 2018-03-04:
I have tried compiling the development version on my RPI3 with a RFM69HW radio same as the node (pro mini 3.3v) but I am getting -90dbm RSSI on node with 100% tx power and a bunch of NACKs, while just compiling the stable version on the RPI3 makes the node reporting a -70dbm RSSI with 0% tx power. I also tried both stable and development version on node but the result is the same so it is something wrong on the GW side.
GW is compiled as follow: ./configure --my-transport=rfm69 --my-rfm69-frequency=433 --my-is-rfm69hw --my-gateway=ethernet --my-port=5003, I only added #define MY_RFM69_NETWORKID (27) to the mysgw.cpp

mfalkvidd, 2018-03-04:
Could you compare the two versions when using the same output power? That would help us determine if the problem is caused by a difference in output power or a difference in code.

SimoneBnc, 2018-03-04:
The output power is set automatically by ATC, I'm just reading it from node side.

mfalkvidd, 2018-03-04:
ok. Thanks.

SimoneBnc, 2018-03-04:
I'll post some rfm69 logs as soon as I can

SimoneBnc, 2018-03-10:
Here are the logs with latest dev on GW and stable on node as it doesn't make much of a difference. 
[log.zip](https://github.com/mysensors/MySensors/files/1799473/log.zip)

tekka007, 2018-03-11:
Looks like there is noise/constant activity on the channel:
```
mysgw: TSF:MSG:GWL OK
mysgw: RFM69:SWR:SEND,TO=2,SEQ=0,RETRY=0
mysgw: RFM69:CSMA:RSSI=-90
mysgw: RFM69:CSMA:RSSI=-85
mysgw: RFM69:CSMA:RSSI=-87
mysgw: RFM69:CSMA:RSSI=-90
...
mysgw: RFM69:CSMA:RSSI=-89
mysgw: RFM69:CSMA:RSSI=-90
mysgw: RFM69:CSMA:RSSI=-90
mysgw: RFM69:CSMA:RSSI=-90
mysgw: RFM69:CSMA:RSSI=-90
mysgw: RFM69:CSMA:RSSI=-91
mysgw: RFM69:CSMA:RSSI=-90
mysgw: RFM69:CSMA:RSSI=-90
mysgw: RFM69:CSMA:RSSI=-89
...
mysgw: RFM69:CSMA:RSSI=-89
mysgw: RFM69:CSMA:RSSI=-89
mysgw: RFM69:CSMA:RSSI=-88
mysgw: RFM69:CSMA:RSSI=-90
mysgw: RFM69:CSMA:RSSI=-90
...
mysgw: RFM69:CSMA:RSSI=-89
mysgw: RFM69:CSMA:RSSI=-89
mysgw: RFM69:CSMA:RSSI=-91
mysgw: RFM69:CSMA:RSSI=-90
mysgw: RFM69:CSMA:RSSI=-88
mysgw: RFM69:CSMA:RSSI=-88
```
CSMA threshold is set to -95dBm (by default), i.e. RSSI above this value are considered as channel activity - you could try setting the threshold lower, e.g. to -85dBm and see if communication/retries improve:
```
#define MY_RFM69_CSMA_LIMIT_DBM (-85)
```

SimoneBnc, 2018-03-11:
mysgw: Starting gateway...
mysgw: Protocol version - 2.3.0-alpha
mysgw: MCO:BGN:INIT GW,CP=RPNGL---,VER=2.3.0-alpha
mysgw: TSF:LRT:OK
mysgw: TSM:INIT
mysgw: TSF:WUR:MS=0
mysgw: RFM69:INIT
mysgw: RFM69:INIT:PIN,CS=24,IQP=22,IQN=25
mysgw: RFM69:PTX:LEVEL=5 dBm
mysgw: TSM:INIT:TSP OK
mysgw: TSM:INIT:GW MODE
mysgw: TSM:READY:ID=0,PAR=0,DIS=0
mysgw: MCO:REG:NOT NEEDED
mysgw: Listening for connections on 0.0.0.0:5003
mysgw: MCO:BGN:STP
mysgw: MCO:BGN:INIT OK,TSP=1
mysgw: New connection from 192.168.0.118
mysgw: GWT:TSA:C=0,CONNECTED
mysgw: TSF:MSG:READ,2-2-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
mysgw: TSF:MSG:BC
mysgw: TSF:MSG:FPAR REQ,ID=2
mysgw: TSF:PNG:SEND,TO=0
mysgw: TSF:CKU:OK
mysgw: TSF:MSG:GWL OK
mysgw: RFM69:SWR:SEND,TO=2,SEQ=0,RETRY=0
mysgw: RFM69:CSMA:RSSI=-92
mysgw: RFM69:SWR:ACK,FROM=2,SEQ=1,RSSI=-67
mysgw: TSF:MSG:SEND,0-0-2-2,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
mysgw: TSF:MSG:READ,2-2-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
mysgw: TSF:MSG:BC
mysgw: TSF:MSG:FPAR REQ,ID=2
mysgw: TSF:CKU:OK,FCTRL
mysgw: TSF:MSG:GWL OK
mysgw: RFM69:SWR:SEND,TO=2,SEQ=1,RETRY=0
mysgw: RFM69:CSMA:RSSI=-93
mysgw: RFM69:SWR:ACK,FROM=2,SEQ=2,RSSI=-67
mysgw: TSF:MSG:SEND,0-0-2-2,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-95
mysgw: RFM69:CSMA:RSSI=-94
mysgw: TSF:MSG:READ,2-2-0,s=255,c=3,t=24,pt=1,l=1,sg=0:1
mysgw: TSF:MSG:PINGED,ID=2,HP=1
mysgw: RFM69:SWR:SEND,TO=2,SEQ=3,RETRY=0
mysgw: RFM69:CSMA:RSSI=-96
mysgw: !RFM69:SWR:NACK
mysgw: RFM69:SWR:SEND,TO=2,SEQ=4,RETRY=1
mysgw: RFM69:CSMA:RSSI=-91
mysgw: RFM69:SWR:ACK,FROM=2,SEQ=4,RSSI=-68
mysgw: TSF:MSG:SEND,0-0-2-2,s=255,c=3,t=25,pt=1,l=1,sg=0,ft=0,st=OK:1
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-94
mysgw: RFM69:CSMA:RSSI=-90
mysgw: TSF:MSG:READ,2-2-0,s=255,c=4,t=0,pt=6,l=10,sg=0:010003007807C8CF0300
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-95
mysgw: RFM69:CSMA:RSSI=-92
mysgw: TSF:MSG:READ,2-2-0,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
mysgw: RFM69:SWR:SEND,TO=2,SEQ=6,RETRY=0
mysgw: RFM69:CSMA:RSSI=-91
mysgw: !RFM69:SWR:NACK
mysgw: RFM69:SWR:SEND,TO=2,SEQ=7,RETRY=1
mysgw: RFM69:CSMA:RSSI=-87
mysgw: RFM69:SWR:ACK,FROM=2,SEQ=7,RSSI=-68
mysgw: TSF:MSG:SEND,0-0-2-2,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-91
mysgw: RFM69:CSMA:RSSI=-90
mysgw: TSF:MSG:READ,2-2-0,s=255,c=0,t=17,pt=0,l=5,sg=0:2.2.0
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-90
mysgw: RFM69:CSMA:RSSI=-95
mysgw: TSF:MSG:READ,2-2-0,s=255,c=3,t=6,pt=1,l=1,sg=0:0
mysgw: GWT:RFC:C=0,MSG=2;255;3;0;6;M
mysgw: RFM69:SWR:SEND,TO=2,SEQ=9,RETRY=0
mysgw: RFM69:CSMA:RSSI=-90
mysgw: RFM69:SWR:ACK,FROM=2,SEQ=10,RSSI=-67
mysgw: TSF:MSG:SEND,0-0-2-2,s=255,c=3,t=6,pt=0,l=1,sg=0,ft=0,st=OK:M
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-91
mysgw: RFM69:CSMA:RSSI=-89
mysgw: TSF:MSG:READ,2-2-0,s=255,c=3,t=11,pt=0,l=12,sg=0:TEMP HUM dev
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-94
mysgw: RFM69:CSMA:RSSI=-94
mysgw: TSF:MSG:READ,2-2-0,s=255,c=3,t=12,pt=0,l=3,sg=0:0.1
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-91
mysgw: RFM69:CSMA:RSSI=-89
mysgw: TSF:MSG:READ,2-2-0,s=0,c=0,t=7,pt=0,l=0,sg=0:
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-93
mysgw: RFM69:CSMA:RSSI=-92
mysgw: TSF:MSG:READ,2-2-0,s=1,c=0,t=6,pt=0,l=0,sg=0:
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-91
mysgw: RFM69:CSMA:RSSI=-90
mysgw: TSF:MSG:READ,2-2-0,s=202,c=0,t=33,pt=0,l=0,sg=0:
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-92
mysgw: RFM69:CSMA:RSSI=-91
mysgw: TSF:MSG:READ,2-2-0,s=203,c=0,t=33,pt=0,l=0,sg=0:
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-95
mysgw: RFM69:CSMA:RSSI=-89
mysgw: TSF:MSG:READ,2-2-0,s=204,c=0,t=16,pt=0,l=0,sg=0:
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-85
mysgw: RFM69:CSMA:RSSI=-86
mysgw: TSF:MSG:READ,2-2-0,s=201,c=0,t=30,pt=0,l=0,sg=0:
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-87
mysgw: RFM69:CSMA:RSSI=-85
mysgw: TSF:MSG:READ,2-2-0,s=255,c=3,t=26,pt=1,l=1,sg=0:2
mysgw: RFM69:SWR:SEND,TO=2,SEQ=19,RETRY=0
mysgw: RFM69:CSMA:RSSI=-82
mysgw: RFM69:CSMA:RSSI=-79
mysgw: RFM69:CSMA:RSSI=-79
mysgw: RFM69:CSMA:RSSI=-81
mysgw: RFM69:CSMA:RSSI=-81
mysgw: RFM69:CSMA:RSSI=-80
mysgw: RFM69:CSMA:RSSI=-78
mysgw: RFM69:CSMA:RSSI=-78
mysgw: RFM69:CSMA:RSSI=-79
mysgw: RFM69:CSMA:RSSI=-80
mysgw: RFM69:CSMA:RSSI=-82
mysgw: RFM69:CSMA:RSSI=-82
mysgw: RFM69:CSMA:RSSI=-85
mysgw: !RFM69:SWR:NACK
mysgw: RFM69:SWR:SEND,TO=2,SEQ=20,RETRY=1
mysgw: RFM69:CSMA:RSSI=-93
mysgw: RFM69:SWR:ACK,FROM=2,SEQ=20,RSSI=-67
mysgw: TSF:MSG:SEND,0-0-2-2,s=255,c=3,t=27,pt=1,l=1,sg=0,ft=0,st=OK:1
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-93
mysgw: RFM69:CSMA:RSSI=-89
mysgw: TSF:MSG:READ,2-2-0,s=1,c=1,t=0,pt=7,l=5,sg=0:22.40
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-94
mysgw: RFM69:CSMA:RSSI=-90
mysgw: TSF:MSG:READ,2-2-0,s=0,c=1,t=1,pt=7,l=5,sg=0:46.0
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-92
mysgw: RFM69:CSMA:RSSI=-91
mysgw: TSF:MSG:READ,2-2-0,s=202,c=1,t=37,pt=2,l=2,sg=0:-94
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-95
mysgw: RFM69:CSMA:RSSI=-93
mysgw: TSF:MSG:READ,2-2-0,s=203,c=1,t=37,pt=2,l=2,sg=0:-67
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-95
mysgw: RFM69:CSMA:RSSI=-89
mysgw: TSF:MSG:READ,2-2-0,s=204,c=1,t=23,pt=2,l=2,sg=0:100
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-89
mysgw: RFM69:CSMA:RSSI=-93
mysgw: TSF:MSG:READ,2-2-0,s=201,c=1,t=38,pt=7,l=5,sg=0:3.26
mysgw: RFM69:SAC:SEND ACK,TO=2,RSSI=-97
mysgw: RFM69:CSMA:RSSI=-89
mysgw: TSF:MSG:READ,2-2-0,s=255,c=3,t=32,pt=5,l=4,sg=0:500

that improved a little, but the node is still using 100% tx power

SimoneBnc, 2018-03-11:
here is node log 

0 MCO:BGN:INIT NODE,CP=RPONA---,VER=2.3.0-alpha
4 TSM:INIT
6 TSF:WUR:MS=10000
8 RFM69:INIT
8 RFM69:INIT:PIN,CS=10,IQP=2,IQN=0
14 RFM69:PTX:LEVEL=5 dBm
16 TSM:INIT:TSP OK
18 TSF:SID:OK,ID=2
20 TSM:FPAR
20 RFM69:SWR:SEND,TO=255,SEQ=0,RETRY=0
26 RFM69:CSMA:RSSI=-102
32 TSF:MSG:SEND,2-2-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
1071 RFM69:SAC:SEND ACK,TO=0,RSSI=-72
1075 RFM69:CSMA:RSSI=-97
1081 TSF:MSG:READ,0-0-2,s=255,c=3,t=8,pt=1,l=1,sg=0:0
1087 TSF:MSG:FPAR OK,ID=0,D=1
2041 TSM:FPAR:OK
2041 TSM:ID
2043 TSM:ID:OK
2045 TSM:UPL
2048 RFM69:SWR:SEND,TO=0,SEQ=2,RETRY=0
2052 RFM69:CSMA:RSSI=-96
2064 RFM69:SWR:ACK,FROM=0,SEQ=3,RSSI=-91
2068 RFM69:ATC:ADJ TXL,cR=-91,tR=-78..-82,TXL=5
2074 RFM69:PTX:LEVEL=6 dBm
2076 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=OK:1
2269 RFM69:SAC:SEND ACK,TO=0,RSSI=-72
2275 RFM69:CSMA:RSSI=-99
2279 TSF:MSG:READ,0-0-2,s=255,c=3,t=25,pt=1,l=1,sg=0:1
2285 TSF:MSG:PONG RECV,HP=1
2289 TSM:UPL:OK
2291 TSM:READY:ID=2,PAR=0,DIS=1
2293 RFM69:SWR:SEND,TO=0,SEQ=4,RETRY=0
2299 RFM69:CSMA:RSSI=-98
2314 RFM69:SWR:ACK,FROM=0,SEQ=5,RSSI=-84
2320 RFM69:ATC:ADJ TXL,cR=-84,tR=-78..-82,TXL=6
2324 RFM69:PTX:LEVEL=7 dBm
2328 TSF:MSG:SEND,2-2-0-0,s=255,c=4,t=0,pt=6,l=10,sg=0,ft=0,st=OK:010003007807C8CF0300
2336 RFM69:SWR:SEND,TO=0,SEQ=5,RETRY=0
2342 RFM69:CSMA:RSSI=-97
2359 RFM69:SWR:ACK,FROM=0,SEQ=6,RSSI=-91
2365 RFM69:ATC:ADJ TXL,cR=-91,tR=-78..-82,TXL=7
2369 RFM69:PTX:LEVEL=8 dBm
2371 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
2566 RFM69:SAC:SEND ACK,TO=0,RSSI=-71
2572 RFM69:CSMA:RSSI=-106
2576 TSF:MSG:READ,0-0-2,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
2582 RFM69:SWR:SEND,TO=0,SEQ=7,RETRY=0
2588 RFM69:CSMA:RSSI=-99
2600 RFM69:SWR:ACK,FROM=0,SEQ=8,RSSI=-87
2607 RFM69:ATC:ADJ TXL,cR=-87,tR=-78..-82,TXL=8
2611 RFM69:PTX:LEVEL=9 dBm
2613 TSF:MSG:SEND,2-2-0-0,s=255,c=0,t=17,pt=0,l=11,sg=0,ft=0,st=OK:2.3.0-alpha
2623 RFM69:SWR:SEND,TO=0,SEQ=8,RETRY=0
2627 RFM69:CSMA:RSSI=-101
2646 RFM69:SWR:ACK,FROM=0,SEQ=9,RSSI=-92
2650 RFM69:ATC:ADJ TXL,cR=-92,tR=-78..-82,TXL=9
2656 RFM69:PTX:LEVEL=10 dBm
2658 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:0
4667 RFM69:SWR:SEND,TO=0,SEQ=9,RETRY=0
4671 RFM69:CSMA:RSSI=-99
4689 RFM69:SWR:ACK,FROM=0,SEQ=10,RSSI=-91
4694 RFM69:ATC:ADJ TXL,cR=-91,tR=-78..-82,TXL=10
4700 RFM69:PTX:LEVEL=11 dBm
4702 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=11,pt=0,l=12,sg=0,ft=0,st=OK:TEMP HUM dev
4710 RFM69:SWR:SEND,TO=0,SEQ=10,RETRY=0
4716 RFM69:CSMA:RSSI=-103
4734 RFM69:SWR:ACK,FROM=0,SEQ=11,RSSI=-93
4739 RFM69:ATC:ADJ TXL,cR=-93,tR=-78..-82,TXL=11
4745 RFM69:PTX:LEVEL=12 dBm
4747 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=12,pt=0,l=3,sg=0,ft=0,st=OK:0.1
4755 RFM69:SWR:SEND,TO=0,SEQ=11,RETRY=0
4761 RFM69:CSMA:RSSI=-103
4780 RFM69:SWR:ACK,FROM=0,SEQ=12,RSSI=-93
4784 RFM69:ATC:ADJ TXL,cR=-93,tR=-78..-82,TXL=12
4790 RFM69:PTX:LEVEL=13 dBm
4792 TSF:MSG:SEND,2-2-0-0,s=0,c=0,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
4800 RFM69:SWR:SEND,TO=0,SEQ=12,RETRY=0
4804 RFM69:CSMA:RSSI=-96
4825 RFM69:SWR:ACK,FROM=0,SEQ=13,RSSI=-93
4829 TSF:MSG:SEND,2-2-0-0,s=1,c=0,t=6,pt=0,l=0,sg=0,ft=0,st=OK:
4837 RFM69:SWR:SEND,TO=0,SEQ=13,RETRY=0
4841 RFM69:CSMA:RSSI=-101
4859 RFM69:SWR:ACK,FROM=0,SEQ=14,RSSI=-91
4864 TSF:MSG:SEND,2-2-0-0,s=202,c=0,t=33,pt=0,l=0,sg=0,ft=0,st=OK:
4872 RFM69:SWR:SEND,TO=0,SEQ=14,RETRY=0
4876 RFM69:CSMA:RSSI=-97
4894 RFM69:SWR:ACK,FROM=0,SEQ=15,RSSI=-93
4898 TSF:MSG:SEND,2-2-0-0,s=203,c=0,t=33,pt=0,l=0,sg=0,ft=0,st=OK:
4907 RFM69:SWR:SEND,TO=0,SEQ=15,RETRY=0
4911 RFM69:CSMA:RSSI=-100
4929 RFM69:SWR:ACK,FROM=0,SEQ=16,RSSI=-96
4933 TSF:MSG:SEND,2-2-0-0,s=204,c=0,t=16,pt=0,l=0,sg=0,ft=0,st=OK:
4941 RFM69:SWR:SEND,TO=0,SEQ=16,RETRY=0
4947 RFM69:CSMA:RSSI=-103
4964 RFM69:SWR:ACK,FROM=0,SEQ=17,RSSI=-94
4968 TSF:MSG:SEND,2-2-0-0,s=201,c=0,t=30,pt=0,l=0,sg=0,ft=0,st=OK:
4976 MCO:REG:REQ
4978 RFM69:SWR:SEND,TO=0,SEQ=17,RETRY=0
4982 RFM69:CSMA:RSSI=-97
4999 RFM69:SWR:ACK,FROM=0,SEQ=18,RSSI=-91
5003 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=26,pt=1,l=1,sg=0,ft=0,st=OK:2
5203 RFM69:SAC:SEND ACK,TO=0,RSSI=-71
5210 RFM69:CSMA:RSSI=-96
5214 TSF:MSG:READ,0-0-2,s=255,c=3,t=27,pt=1,l=1,sg=0:1
5220 MCO:PIM:NODE REG=1
5224 MCO:BGN:STP
5736 MCO:BGN:INIT OK,TSP=1
6739 RFM69:SWR:SEND,TO=0,SEQ=19,RETRY=0
6746 RFM69:CSMA:RSSI=-101
6770 RFM69:SWR:ACK,FROM=0,SEQ=20,RSSI=-84
6774 TSF:MSG:SEND,2-2-0-0,s=1,c=1,t=0,pt=7,l=5,sg=0,ft=0,st=OK:26.56
7282 RFM69:SWR:SEND,TO=0,SEQ=20,RETRY=0
7288 RFM69:CSMA:RSSI=-98
7303 RFM69:SWR:ACK,FROM=0,SEQ=21,RSSI=-94
7309 TSF:MSG:SEND,2-2-0-0,s=0,c=1,t=1,pt=7,l=5,sg=0,ft=0,st=OK:36.0
7315 RFM69:SWR:SEND,TO=0,SEQ=21,RETRY=0
7321 RFM69:CSMA:RSSI=-96
7337 RFM69:SWR:ACK,FROM=0,SEQ=22,RSSI=-91
7344 TSF:MSG:SEND,2-2-0-0,s=202,c=1,t=37,pt=2,l=2,sg=0,ft=0,st=OK:-94
7852 RFM69:SWR:SEND,TO=0,SEQ=22,RETRY=0
7856 RFM69:CSMA:RSSI=-98
7874 RFM69:SWR:ACK,FROM=0,SEQ=23,RSSI=-92
7878 TSF:MSG:SEND,2-2-0-0,s=203,c=1,t=37,pt=2,l=2,sg=0,ft=0,st=OK:-72
8087 RFM69:SWR:SEND,TO=0,SEQ=23,RETRY=0
8091 RFM69:CSMA:RSSI=-100
8103 RFM69:SWR:ACK,FROM=0,SEQ=24,RSSI=-90
8108 TSF:MSG:SEND,2-2-0-0,s=204,c=1,t=23,pt=2,l=2,sg=0,ft=0,st=OK:100
9117 RFM69:SWR:SEND,TO=0,SEQ=24,RETRY=0
9123 RFM69:CSMA:RSSI=-99
9140 RFM69:SWR:ACK,FROM=0,SEQ=25,RSSI=-93
9144 TSF:MSG:SEND,2-2-0-0,s=201,c=1,t=38,pt=7,l=5,sg=0,ft=0,st=OK:3.26
9152 MCO:SLP:MS=600000,SMS=1,I1=255,M1=255,I2=255,M2=255
9158 RFM69:SWR:SEND,TO=0,SEQ=25,RETRY=0
9164 RFM69:CSMA:RSSI=-100
9175 RFM69:SWR:ACK,FROM=0,SEQ=26,RSSI=-60
9181 RFM69:ATC:ADJ TXL,cR=-60,tR=-78..-82,TXL=13
9185 RFM69:PTX:LEVEL=12 dBm
9189 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=32,pt=5,l=4,sg=0,ft=0,st=OK:500
9697 TSF:TDI:TSL
9697 RFM69:RSL

SimoneBnc, 2018-03-17:
Node log with ATC FIX 

0 MCO:BGN:INIT NODE,CP=RPONA---,VER=2.3.0-alpha
4 TSM:INIT
6 TSF:WUR:MS=10000
8 RFM69:INIT
8 RFM69:INIT:PIN,CS=10,IQP=2,IQN=0
12 RFM69:PTX:LEVEL=5 dBm
16 TSM:INIT:TSP OK
18 TSF:SID:OK,ID=2
20 TSM:FPAR
20 RFM69:SWR:SEND,TO=255,SEQ=0,RETRY=0
26 RFM69:CSMA:RSSI=-103
32 TSF:MSG:SEND,2-2-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
532 RFM69:SAC:SEND ACK,TO=0,RSSI=-66
536 RFM69:CSMA:RSSI=-105
542 TSF:MSG:READ,0-0-2,s=255,c=3,t=8,pt=1,l=1,sg=0:0
546 TSF:MSG:FPAR OK,ID=0,D=1
2039 TSM:FPAR:OK
2039 TSM:ID
2041 TSM:ID:OK
2043 TSM:UPL
2045 RFM69:SWR:SEND,TO=0,SEQ=2,RETRY=0
2050 RFM69:CSMA:RSSI=-106
2072 RFM69:SWR:ACK,FROM=0,SEQ=3,RSSI=-86
2076 RFM69:ATC:ADJ TXL,cR=-86,tR=-78..-82,TXL=5
2080 RFM69:PTX:LEVEL=6 dBm
2084 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=OK:1
2277 RFM69:SAC:SEND ACK,TO=0,RSSI=-66
2281 RFM69:CSMA:RSSI=-104
2287 TSF:MSG:READ,0-0-2,s=255,c=3,t=25,pt=1,l=1,sg=0:1
2293 TSF:MSG:PONG RECV,HP=1
2295 TSM:UPL:OK
2297 TSM:READY:ID=2,PAR=0,DIS=1
2301 RFM69:SWR:SEND,TO=0,SEQ=4,RETRY=0
2308 RFM69:CSMA:RSSI=-107
2322 RFM69:SWR:ACK,FROM=0,SEQ=5,RSSI=-90
2326 RFM69:ATC:ADJ TXL,cR=-90,tR=-78..-82,TXL=6
2332 RFM69:PTX:LEVEL=7 dBm
2334 TSF:MSG:SEND,2-2-0-0,s=255,c=4,t=0,pt=6,l=10,sg=0,ft=0,st=OK:010003007807C8CF0300
2344 RFM69:SWR:SEND,TO=0,SEQ=5,RETRY=0
2349 RFM69:CSMA:RSSI=-101
2367 RFM69:SWR:ACK,FROM=0,SEQ=6,RSSI=-93
2371 RFM69:ATC:ADJ TXL,cR=-93,tR=-78..-82,TXL=7
2377 RFM69:PTX:LEVEL=8 dBm
2379 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
2574 RFM69:SAC:SEND ACK,TO=0,RSSI=-66
2578 RFM69:CSMA:RSSI=-105
2584 TSF:MSG:READ,0-0-2,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
2590 RFM69:SWR:SEND,TO=0,SEQ=7,RETRY=0
2594 RFM69:CSMA:RSSI=-103
2609 RFM69:SWR:ACK,FROM=0,SEQ=8,RSSI=-90
2613 RFM69:ATC:ADJ TXL,cR=-90,tR=-78..-82,TXL=8
2619 RFM69:PTX:LEVEL=9 dBm
2621 TSF:MSG:SEND,2-2-0-0,s=255,c=0,t=17,pt=0,l=11,sg=0,ft=0,st=OK:2.3.0-alpha
2629 RFM69:SWR:SEND,TO=0,SEQ=8,RETRY=0
2635 RFM69:CSMA:RSSI=-107
2654 RFM69:SWR:ACK,FROM=0,SEQ=9,RSSI=-94
2658 RFM69:ATC:ADJ TXL,cR=-94,tR=-78..-82,TXL=9
2662 RFM69:PTX:LEVEL=10 dBm
2666 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:0
4675 RFM69:SWR:SEND,TO=0,SEQ=9,RETRY=0
4679 RFM69:CSMA:RSSI=-106
4696 RFM69:SWR:ACK,FROM=0,SEQ=10,RSSI=-92
4702 RFM69:ATC:ADJ TXL,cR=-92,tR=-78..-82,TXL=10
4706 RFM69:PTX:LEVEL=11 dBm
4710 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=11,pt=0,l=12,sg=0,ft=0,st=OK:TEMP HUM dev
4718 RFM69:SWR:SEND,TO=0,SEQ=10,RETRY=0
4724 RFM69:CSMA:RSSI=-107
4741 RFM69:SWR:ACK,FROM=0,SEQ=11,RSSI=-89
4745 RFM69:ATC:ADJ TXL,cR=-89,tR=-78..-82,TXL=11
4751 RFM69:PTX:LEVEL=12 dBm
4755 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=12,pt=0,l=3,sg=0,ft=0,st=OK:0.1
4761 RFM69:SWR:SEND,TO=0,SEQ=11,RETRY=0
4767 RFM69:CSMA:RSSI=-109
4786 RFM69:SWR:ACK,FROM=0,SEQ=12,RSSI=-92
4792 RFM69:ATC:ADJ TXL,cR=-92,tR=-78..-82,TXL=12
4796 RFM69:PTX:LEVEL=13 dBm
4800 TSF:MSG:SEND,2-2-0-0,s=0,c=0,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
4806 RFM69:SWR:SEND,TO=0,SEQ=12,RETRY=0
4812 RFM69:CSMA:RSSI=-103
4831 RFM69:SWR:ACK,FROM=0,SEQ=13,RSSI=-92
4835 TSF:MSG:SEND,2-2-0-0,s=1,c=0,t=6,pt=0,l=0,sg=0,ft=0,st=OK:
4843 RFM69:SWR:SEND,TO=0,SEQ=13,RETRY=0
4849 RFM69:CSMA:RSSI=-98
4866 RFM69:SWR:ACK,FROM=0,SEQ=14,RSSI=-93
4872 TSF:MSG:SEND,2-2-0-0,s=202,c=0,t=33,pt=0,l=0,sg=0,ft=0,st=OK:
4878 RFM69:SWR:SEND,TO=0,SEQ=14,RETRY=0
4884 RFM69:CSMA:RSSI=-102
4900 RFM69:SWR:ACK,FROM=0,SEQ=15,RSSI=-94
4907 TSF:MSG:SEND,2-2-0-0,s=203,c=0,t=33,pt=0,l=0,sg=0,ft=0,st=OK:
4913 RFM69:SWR:SEND,TO=0,SEQ=15,RETRY=0
4919 RFM69:CSMA:RSSI=-104
4935 RFM69:SWR:ACK,FROM=0,SEQ=16,RSSI=-93
4941 TSF:MSG:SEND,2-2-0-0,s=204,c=0,t=16,pt=0,l=0,sg=0,ft=0,st=OK:
4947 RFM69:SWR:SEND,TO=0,SEQ=16,RETRY=0
4954 RFM69:CSMA:RSSI=-105
4970 RFM69:SWR:ACK,FROM=0,SEQ=17,RSSI=-93
4976 TSF:MSG:SEND,2-2-0-0,s=201,c=0,t=30,pt=0,l=0,sg=0,ft=0,st=OK:
4982 MCO:REG:REQ
4984 RFM69:SWR:SEND,TO=0,SEQ=17,RETRY=0
4990 RFM69:CSMA:RSSI=-104
5005 RFM69:SWR:ACK,FROM=0,SEQ=18,RSSI=-92
5011 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=26,pt=1,l=1,sg=0,ft=0,st=OK:2
5212 RFM69:SAC:SEND ACK,TO=0,RSSI=-67
5216 RFM69:CSMA:RSSI=-105
5222 TSF:MSG:READ,0-0-2,s=255,c=3,t=27,pt=1,l=1,sg=0:1
5228 MCO:PIM:NODE REG=1
5230 MCO:BGN:STP
5742 MCO:BGN:INIT OK,TSP=1
6748 RFM69:SWR:SEND,TO=0,SEQ=19,RETRY=0
6752 RFM69:CSMA:RSSI=-106
6764 RFM69:SWR:ACK,FROM=0,SEQ=20,RSSI=-78
6768 RFM69:ATC:ADJ TXL,cR=-78,tR=-78..-82,TXL=13
6774 RFM69:PTX:LEVEL=12 dBm
6776 TSF:MSG:SEND,2-2-0-0,s=1,c=1,t=0,pt=7,l=5,sg=0,ft=0,st=OK:20.16
7286 RFM69:SWR:SEND,TO=0,SEQ=20,RETRY=0
7290 RFM69:CSMA:RSSI=-106
7309 RFM69:SWR:ACK,FROM=0,SEQ=21,RSSI=-94
7313 RFM69:ATC:ADJ TXL,cR=-94,tR=-78..-82,TXL=12
7317 RFM69:PTX:LEVEL=13 dBm
7321 TSF:MSG:SEND,2-2-0-0,s=0,c=1,t=1,pt=7,l=5,sg=0,ft=0,st=OK:58.0
7329 RFM69:SWR:SEND,TO=0,SEQ=21,RETRY=0
7333 RFM69:CSMA:RSSI=-98
7354 RFM69:SWR:ACK,FROM=0,SEQ=22,RSSI=-94
7358 TSF:MSG:SEND,2-2-0-0,s=202,c=1,t=37,pt=2,l=2,sg=0,ft=0,st=OK:-94
7866 RFM69:SWR:SEND,TO=0,SEQ=22,RETRY=0
7870 RFM69:CSMA:RSSI=-103
7888 RFM69:SWR:ACK,FROM=0,SEQ=23,RSSI=-90
7892 TSF:MSG:SEND,2-2-0-0,s=203,c=1,t=37,pt=2,l=2,sg=0,ft=0,st=OK:-67
8101 RFM69:SWR:SEND,TO=0,SEQ=23,RETRY=0
8105 RFM69:CSMA:RSSI=-103
8118 RFM69:SWR:ACK,FROM=0,SEQ=24,RSSI=-93
8122 TSF:MSG:SEND,2-2-0-0,s=204,c=1,t=23,pt=2,l=2,sg=0,ft=0,st=OK:100
9132 RFM69:SWR:SEND,TO=0,SEQ=24,RETRY=0
9138 RFM69:CSMA:RSSI=-101
9154 RFM69:SWR:ACK,FROM=0,SEQ=25,RSSI=-93
9160 TSF:MSG:SEND,2-2-0-0,s=201,c=1,t=38,pt=7,l=5,sg=0,ft=0,st=OK:3.26
9166 MCO:SLP:MS=600000,SMS=1,I1=255,M1=255,I2=255,M2=255
9175 RFM69:SWR:SEND,TO=0,SEQ=25,RETRY=0
9179 RFM69:CSMA:RSSI=-100
9199 RFM69:SWR:ACK,FROM=0,SEQ=26,RSSI=-93
9205 TSF:MSG:SEND,2-2-0-0,s=255,c=3,t=32,pt=5,l=4,sg=0,ft=0,st=OK:500
9711 TSF:TDI:TSL
9711 RFM69:RSL

tekka007, 2018-03-17:
You installed the fix on both node & GW?

pepeEL, 2018-03-17:
What fix ?

pepeEL, 2018-03-18:
please tell me where is this fix and how i can use it... ?

SimoneBnc, 2018-03-18:
Only on node.

pepeEL, 2018-03-18:
Ok but how use it this fix and whre i found it...

SimoneBnc, 2018-03-18:
It's for development purposes only, I'm testing with tekka to find out what is the problem

piotrraczynski, 2018-12-31:
Is there any update on this issue? I'm seeing the same issue using Rpi GW on 2.3.2. Poor TX RSSI (from node perspective) makes ATC unusable.

@SimoneBnc Could you please share the fix you mentioned earlier?

pepeEL, 2018-12-31:
For me is still problem. Untill not fix i use MySensors 2.2.0 and it works ok.

SimoneBnc, 2018-12-31:
Same here, I'm still in 2.2

pepeEL, 2018-12-31:
No looks good. Share new version with bug and share next versio also with bug...

piotrraczynski, 2019-01-12:
I confirm **2.2.0** works OK. Both RSSI on Rpi and ATC.

turboproc, 2019-04-16:
Hi, was this issue ever solved? Running MySensor version 2.3.2-beta  on a AVR 328p / RFM69CW node and a PI3 / RFMCW MQTT gateway I'm experiencing similar issues where the RSSI of messages received by the gateway remain very low (< -85dB) irrespective of the TX level of the node. Distance is only a few meters (< 5).

piotrraczynski, 2019-04-16:
Hi, I'm not sure I haven't checked it with 2.3.X. For production at my
house I'm still using 2.2.0 which works flawlessly for me in that matter.

Right now I'm busy working to get hoperf RFM73 working with mysensors (It's
a chinese 'clone' of the Nordic NRF24 but with some changes to it, anyway I
found it more convenient to use, especially that it's much smaller module
in size compared to Nordic nrf24...), which should work but it's currently
not.
--
Pozdrawiam,
Piotr Raczyński

turboproc, 2019-04-16:
OK, let me check how this works with version 2.2.0. If that works flawlessly at my place as well it's time to find the relevant difference between version 2.2.0 and 2.3.X.

pepeEL, 2019-04-16:
O dont know why developer not fing this bugs and not fix it for a long time.

SimoneBnc, 2019-04-16:
It is still beta for a reason I'd say 😀

pepeEL, 2019-04-16:
Ok but official release 2.3.1 also has this bug.

turboproc, 2019-04-19:
OK, did some trial and error and learned that release 2.2.0 works almost perfect, at least regarding this topic. So sticking for now on this release until I understand what can be gained with anything newer.
One further observersation (and I have to dive a bit deeper): I've the impression that the code doing the ATC control at the node is behaving weird which might be caused by the calculation of upper and lower limit of the target RSSI. Once I changed the code by a brutforce calculation of target RSSI + and - 2 (just hard code 68 resp 72 dB) the ATC became much better and more quiet, i.e. not adjsuting the TX level after every message.

speedy-colibri, 2020-06-17:
> Same here, I'm still in 2.2

I had same connection problems like you guys... With MySensors version 2.3.2, the GW and Nodes work okay for about have a day... the RSSI values ranging from -80 to -90, round about. Then I also switched to version 2.2.0. The RSSI values now read about -60. I hope this time the Nodes and GW operate reliably for much longer...

alexelite, 2021-05-04:
I found the proximate cause for the low Rx RSSI value when radio RFM69 is connected over SPI to Raspberry Pi. 
When the radio is connected to an MCU like ATmega328, the delay between the rising of DIO0 (PayloadReady interrupt) and the time the  `RFM69_interruptHandling`  is called and radio is put into stand by mode, is is in the order of 40us.
When the same radio is connected directly over Pi's SPI interface, the delay increases to about 3ms. From what I understand the radio does not stop evaluating the RSSI when packet is ready, because I read the RSSI register value multiple times after the Packetready interrupt, before changing the mode to standby, and the RSSI decreases over time.
I was able to replicate the low RSSI value on ATmega328 by adding a 1 to 4 ms delay before putting the radio into stand by.
So the shorter the delay is, the higher the RSSI is.

To test this, I used the radio's AutoMode, to switch from Rx to StandBy with the rising edge of PayloadReady `RFM69_REG_AUTOMODES = 0x85` and the RSSI is no longer low.  From -80...-90, now up to -27...-30 for same 1 meter distance from GW to node.

If you want to test this, you need to add `{RFM69_REG_AUTOMODES, 0x85},` in radio configuration function RFM69_setConfiguration(void) and comment this `&& (regIrqFlags2 & RFM69_IRQFLAGS2_PAYLOADREADY)` in RFM69_interruptHandling(void). All modifications in RFM69_new.cpp. With automode on, the PayloadReady flag is no longer set in register IRQFLAGS2, as it turns out.
I am testing this on the latest dev 2.4.0

In the `RFM69_handler(void)` function there is a comment `// radio is in STDBY` but at that time the radio is not in Stand By.

I do not know what is the root cause of this delay, as I wrote [here](https://forum.mysensors.org/topic/11665/rssi-value-variations-on-rpi-3-gateway-with-rfm69/5) before finding this topic and investigating even more, it happened that my GW worked perfectly once, without any hardware or software modification. So if a fluke made it work, there has to be a root cause.

piotrraczynski, 2021-11-02:
@alexelite nice debugging! I checked this topic after my controller SW updates to see if this one got resolved. It seems that almost, do you mind submitting a pull request?

alexelite, 2021-11-16:
@piotrraczynski, did you get a chance to test the fix? For me it works very good. It is the most stable GW after testing multiple alternatives like serial, esp8266-mqtt, esp32-mqtt.
Before and after the fix:
[image]

piotrraczynski, 2021-12-27:
Hi @alexelite not yet, I thought you planned to submit a PR for that. sure I can give it a try, let me first check your fix locally, BTW. what did you use for getting RSSI chart like here?

alexelite, 2022-01-03:
@piotrraczynski, CI failed because the length of one of the lines in the commit message was over the max stated in the rules.
I am not very good with github, so I don't know if I can do anything about the CI, but there are very old PRs that passed all test and have not been reviewed. Unfortunately development seems to have stalled.

About the graphs, I modified the gateway software so for every message received it also publishes an RSSI value, over MQTT. 
The actual graph is form a hacked [LowPowerLab RaspberryPi-Gateway](https://github.com/LowPowerLab/RaspberryPi-Gateway) made to decode mysensors messages also over MQTT. 
The code is [here](https://github.com/alexelite/MyThingsHub)

oumuamua119, 2023-06-22:
Hi all,

I've been having the same RSSI issue on my RPI gateway.

I used the RFM69_new.cpp file you refer above in my gateway, but the RPI presented an error when I ran "make".

Have you solved this problem using another file? Am I doing something wrong?

Thank you for any advice.

alexelite, 2023-06-22:
@oumuamua119, I did not change a thing in make. Can you post what error you got? The changes I made were very basic as I remember. It was more than 2 years ago when I submitted the pull request and maybe other things changed in mysensors.  I use the same version as I did then.

oumuamua119, 2023-06-23:
Hi @alexelite.

Thanks for the response! I repeated the steps I made before and realized I downloaded the wrong file (I did wget on the permalink...). Basic mistake...

I downloaded the right file and it is working perfectly.

Thanks!