mysensors/MySensors #1079: Linking errors with v2.2 and multiple files in Arduino project

papylhomme, 2018-03-10:
I'm trying to update my projects to 2.2, but compilation fails with linking errors.

All of my sketches are using multiple cpp/h files structured as follow:

- a `config.h` defining configuration
- a file for the sketch entry point which include `config.h` and `MySensors.h`
- pairs of cpp/h files, the header includes `config.h` and `MySensors.h`, the code includes the header

This results in the following errors:

    Linking everything together...
    "/home/papy/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p  -o 
    "/tmp/arduino_build_127564/MockMySensors.ino.elf" 
    "/tmp/arduino_build_127564/sketch/MockMySensors.ino.cpp.o" 
    "/tmp/arduino_build_127564/sketch/test.cpp.o" "/tmp/arduino_build_127564/libraries/SPI/SPI.cpp.o" 
    "/tmp/arduino_build_127564/core/core.a" "-L/tmp/arduino_build_127564" -lm
    /tmp/arduino_build_127564/sketch/test.cpp.o (symbol from plugin): In function `wakeUp1()':
    (.text+0x0): multiple definition of `wakeUp1()'
    /tmp/arduino_build_127564/sketch/MockMySensors.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
    /tmp/arduino_build_127564/sketch/test.cpp.o (symbol from plugin): In function `wakeUp1()':
    (.text+0x0): multiple definition of `_wakeUp1Interrupt'
    /tmp/arduino_build_127564/sketch/MockMySensors.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
    /tmp/arduino_build_127564/sketch/test.cpp.o (symbol from plugin): In function `wakeUp1()':
    (.text+0x0): multiple definition of `_wokeUpByInterrupt'
    /tmp/arduino_build_127564/sketch/MockMySensors.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
    /tmp/arduino_build_127564/sketch/test.cpp.o (symbol from plugin): In function `wakeUp1()':
    (.text+0x0): multiple definition of `wakeUp2()'
    /tmp/arduino_build_127564/sketch/MockMySensors.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
    /tmp/arduino_build_127564/sketch/test.cpp.o (symbol from plugin): In function `wakeUp1()':
    (.text+0x0): multiple definition of `_wakeUp2Interrupt'
    /tmp/arduino_build_127564/sketch/MockMySensors.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
    /tmp/arduino_build_127564/sketch/test.cpp.o (symbol from plugin): In function `wakeUp1()':
    ...

papylhomme, 2018-03-10:
Minimal example to reproduce: https://github.com/papylhomme/MySensorsTest

wi1k1n, 2022-02-27:
Hey, I have the same problem. Did you manage to find the solution?

virtual-maker, 2022-03-02:
> Hey, I have the same problem. Did you manage to find the solution?

@wi1k1n Which MySensors version do you use? This issue is related to ver. 2.2 which is outdated now. Did you tried to use current ver. 2.3.2 or 2.4.0-alpha?

A working example to reproduce your issue would be fine.
Unfortunately the mini example link from @papylhomme is pointing to 404 :-(

wi1k1n, 2022-03-03:
I'm using v2.3.2 with PlatformIO. This error occurred when I was trying to make several classes for my sensors with some inheritance (basically the including of <MySensors.h> in multiple header files caused this problem).

I've already made some kind of stupid workaround on it in my project (simply placing everything in one main.cpp to avoid multiple inclusion of MySensors.h), but for now I cannot recreate this issue again (strange, apparently I did something else, when I got this problem). I'll probably catch this issue again when doing some refactoring (in the near future). I'll update here if I face this problem again or not.

Thanks for answering!

matkor, 2022-03-03:
I am not sure how I solved above problem but I see NOTE in my code that #include'ing <MySensors.h> in multiple header / cpp files leads to linking problems above. 

Also I see that I am including 
```
#include <core/MySensorsCore.h>
```
in all my library headers where needed, and 
```
#include <MySensors.h>
#include <MyOwnLibraryExtendingMySensors.h>
```
only in main project file (ino).

MichaelLhommeOFP, 2022-03-03:
Sorry for the missing link. I haven't played with MySensors for a while now and I remember the problem only vaguely, @matkor solution seems to be the way to go

wi1k1n, 2022-03-03:
Ah, I see. I'll try it. @matkor , @MichaelLhommeOFP thanks a lot for helping out!

virtual-maker, 2022-03-03:
> Ah, I see. I'll try it. @matkor , @MichaelLhommeOFP thanks a lot for helping out!

@wi1k1n If you find a working solution to this problem it would be very nice to share it here.
Maybe then this old issue can be closed. Thank you

lyuboasenov, 2022-11-29:
I ran into this problem when trying to refactor the nrf24Doctor. The device couldn't measure the current consumption. I tried reviewing the code but failed. So I went on to refactor the code to make it more readable, at least to me.

Pointed by the answers of @wi1k1n and @matkor managed to work around the re-definition issue in MySensors - [.](https://github.com/lyuboasenov/nRF24Doctor/tree/platform-io).

The problem is that inside MySensors, there is a lot of inclusion of implementations rather than abstractions. 

My workaround is defining a component that wraps MySensors usage into a component - [.](https://github.com/lyuboasenov/nRF24Doctor/blob/platform-io/include/Radio.h). And only including MySensors library once into the implementation of this component - [.](https://github.com/lyuboasenov/nRF24Doctor/blob/platform-io/src/Radio.cpp)

I hope this can help.
L.