mysensors/MySensors #1095: Send Message, pass ID

soloam, 2018-03-26:
Hi all, please consider this:

It would be nice to, in the send function, that we could pass a extra value after the ack request, that value would be a numeric value that the gw would return back when replying with the ack. That way we could pass a unique id with the message, allowing to filter if the messages that we are acknowledging is the expected one.

Something like:

`id++;
send(msg2.set(state?false:true), true,id);`

Thank You

mfalkvidd, 2018-03-26:
Thanks for the suggestion. A sequence number or similar has been discussed. That eould make it possible to drop duplicate messages as well.

Until such feature is implemented, there might be a workaround for you: the ack message contains a copy of the original message. So it is possible to compare the payload to see if the ack is for the expected message. This workaround does not cover all use cases though.

soloam, 2018-03-28:
Thank you @mfalkvidd , yes that helps, can you tell me how can I get that copy of the message, that will work as a workaround until the feature is available.

Thank You and nice work

mfalkvidd, 2018-03-28:
The message is available in the receive function. Documentation: https://www.mysensors.org/apidocs/group__MySensorsCoregrp.html#gadbcc445e605952488109e37792e3fe04

closed 2022-01-20 by soloam