mysensors/MySensors #1096: Update GatewayESP8266 and GatewayESP8266MQTTClient wiring

mfalkvidd, 2018-03-26:
The wiring instructions provided in comments in GatewayESP8266 and GatewayESP8266MQTTClient  example sketches do not match the wiring instructions at https://www.mysensors.org/build/connect_radio

Suggestion: Remove the wiring instructions entirely from both sketches and point to https://www.mysensors.org/build/connect_radio only.

This discrepancy was reported by forum user RWoerz in https://forum.mysensors.org/topic/4800/building-a-wifi-gateway-using-esp8266/58

thucar, 2018-04-03:
I believe the wiring instructions in those examples should be carried over to the "Connecting the radio" page. The instructions are correct, but for a bare ESP8266 module whereas the current instructions on the Radio page are for the NodeMCU/Wemos boards.

I'm using mostly ESP8266-12E's on their own and always found myself glancing at the wiring instructions in the examples to refresh my memory.

mfalkvidd, 2018-04-03:
Thanks for the feedback on ESP-12E. I am not familiar with it, so I just left the comments in the sketch.
Now I've googled ESP-12E and I realized that I bought a few some time ago, but never got around to use them since the Wemos D1 mini is so much easier to use :-)

Just to make sure we're talking about the same instructions, these are the instructions **I** believe  we're discussing:
```
 * Not all ESP8266 modules have all pins available on their external interface.
 * This code has been tested on an ESP-12 module.
 * The ESP8266 requires a certain pin configuration to download code, and another one to run code:
 * - Connect REST (reset) via 10K pullup resistor to VCC, and via switch to GND ('reset switch')
 * - Connect GPIO15 via 10K pulldown resistor to GND
 * - Connect CH_PD via 10K resistor to VCC
 * - Connect GPIO2 via 10K resistor to VCC
 * - Connect GPIO0 via 10K resistor to VCC, and via switch to GND ('bootload switch')
```

To me, adding these instructions to "connecting the radio" feels a bit weird, since none of the instructions concern the radio(s).

Would a new section at https://www.mysensors.org/build/esp8266_gateway be a good place?

closed 2018-04-07 by mfalkvidd