mysensors/MySensors #1098: MQTT gateway connects using random client ID

MySuser1, 2018-04-01:
I have a MySensors 2.1.1 MQTT gateway running on a Raspberry Pi 2B, Raspbian Stretch, NRF24 radio to a couple ATmega-based sensors, with a properly configured client id ("--my-mqtt-client-id=raspi4gw" in config file).

When looking at the logs of my Mosquitto MQTT server, I note that the gateway first connects as expected
  `New client connected from 192.168.1.47 {raspi7}  as raspi4gw (c1, k15).` 
and then 2 seconds later connects again, using a seemingly random client ID:
  `New client connected from 192.168.1.47 {raspi7}  as Qf?GDHJOJR\WxBq=FwwxLLG4_aarMT<^AKuFS7emYI;IKlFa[4PwPW26?SiLgej0 (c1, k60).`

Any explanation ?

mfalkvidd, 2018-04-01:
@MySuser1 could you post a debug log from the gateway when this happens?

MySuser1, 2018-04-01:
The full configure line was
```
./configure             \
 --my-gateway=mqtt      \
 --my-controller-url-address=ha-server    \
 --my-mqtt-client-id=raspi4gw             \
 --my-mqtt-publish-topic-prefix=my/2/stat  \
 --my-mqtt-subscribe-topic-prefix=my/cmnd \
 --spi-driver=SPIDEV    \
 --spi-spidev-device=/dev/spidev1.2 \
 --my-rf24-cs-pin=36    \
 --my-rf24-ce-pin=37    \
 --my-rf24-irq-pin=33

```
How do I create a debug log from the gateway? I am running mysgw as a service

mfalkvidd, 2018-04-01:
Start it manually using sudo ./bin/mysgw -d

or follow instructions at https://www.mysensors.org/build/raspberry#troubleshooting

mfalkvidd, 2018-04-02:
I've tried to create a minimal case to reproduce the problem, but have not been successful yet. Could you try the following (change IP if you don't have mqtt on localhost):
`./configure --my-gateway=mqtt --my-controller-url-address=127.0.0.1 --my-mqtt-client-id=raspi4gw --my-mqtt-publish-topic-prefix=my/2/stat  --my-mqtt-subscribe-topic-prefix=my/cmnd --my-transport=none`

on 2.2.0, I get
```
1522690330: New connection from 127.0.0.1 on port 1883.
1522690330: New client connected from 127.0.0.1 as raspi4gw (c1, k15).
```
and nothing else

Gateway debug output 2.2.0:
```
mysgw: Starting gateway...
mysgw: Protocol version - 2.2.0
mysgw: MCO:BGN:INIT GW,CP=R-NGL---,VER=2.2.0
mysgw: MCO:BGN:STP
mysgw: MCO:REG:NOT NEEDED
mysgw: MCO:BGN:INIT OK,TSP=NA
mysgw: GWT:RMQ:MQTT RECONNECT
mysgw: connected to 127.0.0.1
mysgw: GWT:RMQ:MQTT CONNECTED
mysgw: GWT:TPS:TOPIC=my/2/stat/0/255/0/0/17,MSG SENT
```

MySuser1, 2018-04-02:
Apparently, this only happens once after the computer restarts. Restarting the _service_ alone does not cause the connection with a random client-ID (watched the logs for several hours))

After restarting the Raspberry Pi, when the mysgw service is started (at 19:22), `/var/log/daemon.log` on the Raspberry Pi shows this:

```
Apr  2 19:22:40 raspi7 systemd[1]: Created slice User Slice of admin.
Apr  2 19:22:40 raspi7 systemd[1]: Starting User Manager for UID 1001...
Apr  2 19:22:41 raspi7 systemd[1]: Started Session c10 of user admin.
Apr  2 19:22:41 raspi7 systemd[14178]: Listening on GnuPG cryptographic agent (access for web browsers).
Apr  2 19:22:41 raspi7 systemd[14178]: Listening on GnuPG cryptographic agent and passphrase cache.
Apr  2 19:22:41 raspi7 systemd[14178]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Apr  2 19:22:41 raspi7 systemd[14178]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Apr  2 19:22:41 raspi7 systemd[14178]: Reached target Sockets.
Apr  2 19:22:41 raspi7 systemd[14178]: Reached target Paths.
Apr  2 19:22:41 raspi7 systemd[14178]: Reached target Timers.
Apr  2 19:22:41 raspi7 systemd[14178]: Reached target Basic System.
Apr  2 19:22:41 raspi7 systemd[14178]: Reached target Default.
Apr  2 19:22:41 raspi7 systemd[14178]: Startup finished in 71ms.
Apr  2 19:22:41 raspi7 systemd[1]: Started User Manager for UID 1001.
Apr  2 19:22:51 raspi7 systemd[1]: Stopping Authorization Manager...
Apr  2 19:22:51 raspi7 systemd[14178]: Stopped target Default.
Apr  2 19:22:51 raspi7 systemd[14178]: Stopped target Basic System.
Apr  2 19:22:51 raspi7 systemd[14178]: Stopped target Paths.
Apr  2 19:22:51 raspi7 systemd[14178]: Stopped target Sockets.
Apr  2 19:23:38 raspi7 systemd[1]: Time has been changed
Apr  2 19:23:38 raspi7 systemd[1]: apt-daily.timer: Adding 2h 34min 34.371963s random time.
Apr  2 19:23:38 raspi7 systemd[1]: apt-daily-upgrade.timer: Adding 55min 29.504664s random time.
Apr  2 19:24:35 raspi7 systemd[1]: dev-serial1.device: Job dev-serial1.device/start timed out.
Apr  2 19:24:35 raspi7 systemd[1]: Timed out waiting for device dev-serial1.device.
Apr  2 19:24:35 raspi7 systemd[1]: dev-serial1.device: Job dev-serial1.device/start failed with result 'timeout'.
Apr  2 19:28:08 raspi7 systemd[1]: mysgw.service: Start operation timed out. Terminating.
Apr  2 19:28:08 raspi7 systemd[1]: Failed to start LSB: MySensors Gateway.
Apr  2 19:28:08 raspi7 systemd[1]: mysgw.service: Unit entered failed state.
Apr  2 19:28:08 raspi7 systemd[1]: mysgw.service: Failed with result 'timeout'.
Apr  2 19:28:08 raspi7 systemd[1]: Reached target Multi-User System.
Apr  2 19:28:08 raspi7 systemd[1]: Starting Update UTMP about System Runlevel Changes...
Apr  2 19:28:08 raspi7 systemd[1]: Started Update UTMP about System Runlevel Changes.
Apr  2 19:28:08 raspi7 systemd[1]: Startup finished in 3.139s (kernel) + 5min 4.769s (userspace) = 5min 7.909s.

```
At the same time, the log on the machine running the mosquitto server ( a Debian Linux VM) shows this (the IP address of the Raspberry Pi running mysgw is 192.168.1.47)
```
Apr  2 19:22:52 ha-server mosquitto[21425]: Socket error on client raspi4gw, disconnecting.
Apr  2 19:22:53 ha-server mosquitto[21425]: Socket error on client Qf?GDHJOJR\WxBq=FwwxLLG4_aarMT<^AKuFS7emYI;IKlFa[4PwPW26?SiLgej0, disconnecting.
Apr  2 19:23:16 ha-server mosquitto[21425]: New connection from 192.168.1.47 on port 1883.
Apr  2 19:23:16 ha-server mosquitto[21425]: New client connected from 192.168.1.47 as raspi4gw (c1, k15).
Apr  2 19:23:19 ha-server mosquitto[21425]: New connection from 192.168.1.47 on port 1883.
Apr  2 19:23:19 ha-server mosquitto[21425]: New client connected from 192.168.1.47 as S;mFhQmFMbjWVMtXVjFH11e;]4::W>qj9e7rvdxKNjbd>MCUxI]0:I;gN5qeDiVM (c1, k60).

```
The debug output of mysgw shows nothing at that time, a little later , it starts receiving NRF24 messages
```
Apr  2 18:56:43 raspi7 mysgw: TSF:SAN:OK
Apr  2 19:06:43 raspi7 mysgw: TSM:READY:NWD REQ
Apr  2 19:06:43 raspi7 mysgw: TSF:MSG:SEND,0-0-255-255,s=255,c=3,t=20,pt=0,l=0,sg=0,ft=0,st=OK:
Apr  2 19:11:43 raspi7 mysgw: TSF:SAN:OK
Apr  2 19:27:17 raspi7 mysgw: TSF:MSG:READ,109-109-0,s=41,c=1,t=0,pt=2,l=2,sg=0:18
Apr  2 19:27:17 raspi7 mysgw: Sending message on topic: my/2/stat/109/41/1/0/0
Apr  2 19:27:17 raspi7 mysgw: TSF:MSG:READ,109-109-0,s=11,c=1,t=16,pt=2,l=2,sg=0:0
Apr  2 19:27:17 raspi7 mysgw: Sending message on topic: my/2/stat/109/11/1/0/16
Apr  2 19:27:17 raspi7 mysgw: TSF:MSG:READ,109-109-0,s=12,c=1,t=16,pt=2,l=2,sg=0:1
Apr  2 19:27:17 raspi7 mysgw: Sending message on topic: my/2/stat/109/12/1/0/16
Apr  2 19:28:14 raspi7 mysgw: TSF:MSG:READ,110-110-0,s=61,c=1,t=23,pt=3,l=2,sg=0:14
Apr  2 19:28:14 raspi7 mysgw: Sending message on topic: my/2/stat/110/61/1/0/23
Apr  2 19:28:14 raspi7 mysgw: TSF:MSG:READ,110-110-0,s=21,c=1,t=16,pt=2,l=2,sg=0:0
Apr  2 19:28:14 raspi7 mysgw: Sending message on topic: my/2/stat/110/21/1/0/16

```

mfalkvidd, 2018-04-02:
My guess is that, for some reason, two instances of mysgw are started at boot.
Could you check the process list (`ps -ef | grep mysgw` or similar) after boot, to determine of more than one instance is started?

This is the output on my Pi:
```
pi@raspberrypi:~ $ ps -ef | grep mysgw
root       956     1  0 Mar23 ?        01:49:07 /usr/local/bin/mysgw
pi       29283 28673  0 19:52 pts/2    00:00:00 grep --color=auto mysgw

```

mfalkvidd, 2018-04-02:
You could also do `sudo systemctl disable mysgw.service` and restart to see if you still get one connection.

MySuser1, 2018-04-02:
I did (as root)
 `systemctl disable mysgw.service`
and also (belts and suspenders)
 `update-rc.d mysgw disable`
then rebooted the Pi,  now mysgw shouldn't be running at all, and it isn't
```
# ps -ef | grep mysgw
root       677   670  0 20:52 pts/0    00:00:00 grep mysgw
```
started mysgw manually
```
# mysgw -b -d &
[1] 705
```
Checked the log on the mosquitto machine: ok, only 1 connection made, as raspi4gw

Then stopped mysgw, enabled the service again, and started the service
```
 killall mysgw
 update-rc.d mysgw enable
 service mysgw start
```
Log ion mosquitto machine shows just 1 connection, but the shell hangs after the last line. Checking status in a 2nd shell
```
# service mysgw status
● mysgw.service - LSB: MySensors Gateway
   Loaded: loaded (/etc/init.d/mysgw; generated; vendor preset: enabled)
   Active: activating (start) since Mon 2018-04-02 20:56:31 CEST; 3min 12s ago
     Docs: man:systemd-sysv-generator(8)
Cntrl PID: 752 (mysgw)
   CGroup: /system.slice/mysgw.service
           ├─752 /bin/sh /etc/init.d/mysgw start
           └─758 /usr/local/bin/mysgw -b -d

Apr 02 20:56:31 raspi7 mysgw[758]: MCO:REG:NOT NEEDED
Apr 02 20:56:31 raspi7 mysgw[758]: MCO:BGN:STP
Apr 02 20:56:31 raspi7 mysgw[758]: MCO:BGN:INIT OK,TSP=1
Apr 02 20:56:31 raspi7 mysgw[758]: Attempting MQTT connection...
Apr 02 20:56:31 raspi7 mysgw[758]: connected to 192.168.1.91
Apr 02 20:56:31 raspi7 mysgw[758]: MQTT connected
Apr 02 20:56:31 raspi7 mysgw[758]: Sending message on topic: my/2/stat/0/255/0/0/18
Apr 02 20:56:31 raspi7 mysgw[758]: Message arrived on topic: my/cmnd/113/31/1/0/2
Apr 02 20:56:31 raspi7 mysgw[758]: !TSF:RTE:113 UNKNOWN
Apr 02 20:56:31 raspi7 mysgw[758]: !TSF:MSG:SEND,0-0-113-113,s=31,c=1,t=2,pt=0,l=1,sg=0,ft=0,st=NACK:0
```
^C in the 1st shell, then checking if mysgw is running
```
# ps -ef | grep mysgw
root       752     1  0 20:56 ?        00:00:00 /bin/sh /etc/init.d/mysgw start
root       758   752  0 20:56 ?        00:00:01 /usr/local/bin/mysgw -b -d
root       895   670  0 21:00 pts/0    00:00:00 grep mysgw
```
Stopping the service doesn't work
```
# service mysgw stop
# ps -ef | grep mysgw
root       758     1  0 20:56 ?        00:00:01 /usr/local/bin/mysgw -b -d
root       920   670  0 21:02 pts/0    00:00:00 grep mysgw
```
This is all with MySensors 2.1.1, btw.
Oh, and this is running on a Raspberry Pi that was originally set up with Raspbian Wheezy, and then migrated to Raspbian Stretch using "apt-get dist-upgrade" ... maybe there is some confusion with init.d vs the new init system ...

I will try to disable, uninstall, rebuild, and make install again

mfalkvidd, 2018-04-02:
You could also try running `sudo netstat -nap | grep 1883` on the Raspberry Pi to list all processes with established connections to the MQTT port.

MySuser1, 2018-04-02:
No success.
I disabled and removed the service, even renamed /etc/init.d/mysgw to something else, then did "make install" again, rebooted the Pi.
Now the service is running
```
# service mysgw status
● mysgw.service - MySensors Gateway daemon
   Loaded: loaded (/etc/systemd/system/mysgw.service; enabled; vendor preset: en
   Active: active (running) since Mon 2018-04-02 21:18:33 CEST; 13min ago
 Main PID: 310 (mysgw)
   CGroup: /system.slice/mysgw.service
           └─310 /usr/local/bin/mysgw -d
```
and 1 instance of mysgw is running
```
# ps -ef | grep mysgw
root       310     1  0 21:18 ?        00:00:03 /usr/local/bin/mysgw -d
root       773   769  0 21:30 pts/0    00:00:00 grep mysgw
```
But the mosquitto server was contacted twice
```
Apr  2 21:18:27 ha-server mosquitto[21425]: Socket error on client raspi4gw, disconnecting.
Apr  2 21:18:29 ha-server mosquitto[21425]: Socket error on client EM6C86NV`p1^E4sDNW4Xq0G65xk9YHG^e=a]36tSwu9C13GOKKg3Kn9A^d:>lA\H, disconnecting.
Apr  2 21:18:52 ha-server mosquitto[21425]: New connection from 192.168.1.47 on port 1883.
Apr  2 21:18:52 ha-server mosquitto[21425]: New client connected from 192.168.1.47 as raspi4gw (c1, k15).
Apr  2 21:18:54 ha-server mosquitto[21425]: New connection from 192.168.1.47 on port 1883.
Apr  2 21:18:54 ha-server mosquitto[21425]: New client connected from 192.168.1.47 as OEuBKpf9]o<^bSmm_Kp1Ai2oD=m7>P?]U<pqlMjPCwneQRZpnAr62de6aI>pYMT6 (c1, k60).
Apr  2 21:35:25 ha-server mosquitto[21425]: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
```
only 1 mysgw left log messages after the Pi restarted
```
Apr  2 21:18:27 ha-server mosquitto[21425]: Socket error on client raspi4gw, disconnecting.
Apr  2 21:18:29 ha-server mosquitto[21425]: Socket error on client EM6C86NV`p1^E4sDNW4Xq0G65xk9YHG^e=a]36tSwu9C13GOKKg3Kn9A^d:>lA\H, disconnecting.
Apr  2 21:18:52 ha-server mosquitto[21425]: New connection from 192.168.1.47 on port 1883.
Apr  2 21:18:52 ha-server mosquitto[21425]: New client connected from 192.168.1.47 as raspi4gw (c1, k15).
Apr  2 21:18:54 ha-server mosquitto[21425]: New connection from 192.168.1.47 on port 1883.
Apr  2 21:18:54 ha-server mosquitto[21425]: New client connected from 192.168.1.47 as OEuBKpf9]o<^bSmm_Kp1Ai2oD=m7>P?]U<pqlMjPCwneQRZpnAr62de6aI>pYMT6 (c1, k60).
Apr  2 21:35:25 ha-server mosquitto[21425]: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
```

MySuser1, 2018-04-02:
Oops, so sorry, problem solved, wasn't a mysgw error after all.

I had another MQTT client (a Python script) run as a service, too, and the constructor for the MQTT class didn't specify a client ID, so I guess paho-mqtt picked a really long random string instead. That service, of course, got started at almost the same time as mysgw ...
Once I did `sudo netstat -nap | grep 1883` , I saw the other process talking to port 1883, and realized what was happening.

I owe you a virtual beer ...

mfalkvidd, 2018-04-02:
No worries :-) Great work!

I'll close this issue then.

If you don't have an account already, you are very welcome to head over to https://forum.mysensors.org - a lot of nice and knowledgeable people hang around there.

closed 2018-04-02 by mfalkvidd