mysensors/MySensors #1102: Add pull request template

mfalkvidd, 2018-04-04:
To make it easier for first time contributors to understand why Gitler/Jenkins really is their friend (and not the pain in the a** that a first impression might give), let's add a pull request template that gently introduces our PR process/flow.

Documentation: https://help.github.com/articles/creating-a-pull-request-template-for-your-repository/

Make sure the template contains a prominent link to https://www.mysensors.org/download/contributing

fallberg, 2018-04-04:
Even better, use this: https://help.github.com/articles/setting-guidelines-for-repository-contributors/

The PR template will basically set a template for the commit message after we squash the PR.

mfalkvidd, 2018-04-04:
We already have that :-) https://github.com/mysensors/MySensors/blob/development/CONTRIBUTING.md

fallberg, 2018-04-04:
Then we should be done I guess :) And anyone that disagree can be instructed to RTFM.