mysensors/MySensors #1103: Rename "soft ack" to "echo"

mfalkvidd, 2018-04-05:
The send() functions (send(), present(), sendSketchInfo(), sendBatteryLevel() and maybe more) have a parameter called ack, which often is confused with what might be called "hardware ack" (see https://forum.mysensors.org/post/34263 for one of many lengthy discussions).

Suggestion: rename the parameter to echo. That should make it harder to confuse with the hardware ack, and reflects more accurately what will actually happen (the final destination will echo back the contents of the message, triggering the receive() function on the original node with a copy of the message, with message.isAck() set)

flatsiedatsie, 2019-05-30:
Should .isAck() now be renamed to .isEcho() in node code?

mfalkvidd, 2019-05-30:
@flatsiedatsie yes
[image]

Temporary staging url for the documentation before this PR is merged: https://ci.mysensors.org/job/MySensors/job/MySensors/job/PR-1292/7/Doxygen_20HTML/MyMessage_8h.html#a8f286dd914ce8dda5a5795d9cbb74b44

closed 2019-06-06 by mfalkvidd