mysensors/MySensors #1109: Consolidate wifi defines

mfalkvidd, 2018-04-08:
MySensors dev currently uses the following defines for wifi:
```
MY_ESP32_BSSID
MY_ESP32_HOSTNAME
MY_ESP32_PASSWORD
MY_ESP32_SSID
MY_ESP8266_BSSID
MY_ESP8266_HOSTNAME
MY_ESP8266_PASSWORD
MY_ESP8266_SSID
```
The board is not very relevant, and by consolidating the defines we would only need to document them once. They could look like this:
```
MY_BSSID
MY_HOSTNAME
MY_PASSWORD
MY_SSID
```
- [ ] Add backwards compatibility to support MY_ESP8266_* defines in existing sketches (ifdef MY_ESP8266_SSID "xyz" #define MY_SSID "xyz")
- [ ] Move MY_ESP8266_* keywords to the deprecated section in keywords.txt so they are no longer highlighted
- [ ] Change ESP32 code to only use the new defines (no backwards compatibility needed since ESP32 support is brand new
- [ ] Add Doxygen comments to the new defines

marceloaqno, 2018-04-08:
I think MY_PASSWORD could cause some confusion with signing simple password feature.
How about this?
MY_HOSTNAME
MY_WIFI_BSSID
MY_WIFI_PASSWORD
MY_WIFI_SSID

thucar, 2018-04-17:
Would it make sense to also use the above for MY_GATEWAY_TINYGSM when using an Esp8266 as a modem?

mfalkvidd, 2018-05-18:
Esp8266 defines updated by https://github.com/mysensors/MySensors/pull/1110

closed 2018-05-18 by mfalkvidd