mysensors/MySensors #1112: Error when do "make" with --master with RFM69

sineverba, 2018-04-13:
Raspberry PI1, branch master.

I configure with this:  `./configure --my-gateway=ethernet --my-transport=rfm69 --my-rfm69-frequency=433 --my-is-rfm69hw`

After I launch "make" I get:

`In file included from ./MySensors.h:343:0,
                 from examples_linux/mysgw.cpp:83:
./drivers/RFM69/new/RFM69_new.cpp:663:12: warning: ‘void RFM69_ATCmode(bool, int16_t)’ defined but not used [-Wunused-function]
 LOCAL void RFM69_ATCmode(const bool onOff, const int16_t targetRSSI)`

mfalkvidd, 2018-04-14:
@sineverba the warning is something that should be fixed eventually, but it is just a warning and won't break anything.

For the errors at end of make, I am not getting them. But it looks like the error message has been copied and then pasted into the terminal window, which causes the shell to try to execute the warning text which won't work.

This is the output when I use the same configure:
```
HEAD is now at 64be32a... MySensors 2.2.0 release
pi@raspberrypi:~/MySensors $ ./configure --my-gateway=ethernet --my-transport=rfm69 --my-rfm69-frequency=433 --my-is-rfm69hw
[SECTION] Detecting target machine.
  [OK] machine detected: SoC=BCM2836, Type=rpi2, CPU=armv7l.
[SECTION] Detecting SPI driver.
  [OK] SPI driver detected:BCM.
[SECTION] Detecting init system.
  [OK] init system detected: systemd.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.
pi@raspberrypi:~/MySensors $ make
gcc -MT build/drivers/Linux/log.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/log.c -o build/drivers/Linux/log.o
g++ -MT build/drivers/Linux/noniso.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/noniso.cpp -o build/drivers/Linux/noniso.o
g++ -MT build/drivers/Linux/GPIO.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/GPIO.cpp -o build/drivers/Linux/GPIO.o
g++ -MT build/drivers/Linux/SPIDEV.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/SPIDEV.cpp -o build/drivers/Linux/SPIDEV.o
g++ -MT build/drivers/Linux/Print.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/Print.cpp -o build/drivers/Linux/Print.o
g++ -MT build/drivers/Linux/EthernetClient.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/EthernetClient.cpp -o build/drivers/Linux/EthernetClient.o
g++ -MT build/drivers/Linux/SerialPort.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/SerialPort.cpp -o build/drivers/Linux/SerialPort.o
g++ -MT build/drivers/Linux/Stream.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/Stream.cpp -o build/drivers/Linux/Stream.o
g++ -MT build/drivers/Linux/IPAddress.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/IPAddress.cpp -o build/drivers/Linux/IPAddress.o
g++ -MT build/drivers/Linux/compatibility.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/compatibility.cpp -o build/drivers/Linux/compatibility.o
g++ -MT build/drivers/Linux/interrupt.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM69 -DMY_RFM69_NEW_DRIVER -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DMY_IS_RFM69HW -DMY_RFM69_FREQUENCY=RFM69_433MHZ  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c drivers/Linux/interrupt.cpp -o build/drivers/Linux/interrupt.o
[... cut ...]
g++ -pthread  -o bin/mysgw build/drivers/Linux/log.o build/drivers/Linux/noniso.o build/drivers/Linux/GPIO.o build/drivers/Linux/SPIDEV.o build/drivers/Linux/Print.o build/drivers/Linux/EthernetClient.o build/drivers/Linux/SerialPort.o build/drivers/Linux/Stream.o build/drivers/Linux/IPAddress.o build/drivers/Linux/compatibility.o build/drivers/Linux/interrupt.o build/drivers/Linux/SerialSimulator.o build/drivers/Linux/SoftEeprom.o build/drivers/Linux/EthernetServer.o build/examples_linux/mysgw.o build/drivers/BCM/bcm2835.o build/drivers/BCM/BCM.o build/drivers/BCM/SPIBCM.o build/drivers/BCM/Wire.o build/drivers/BCM/RPi.o
pi@raspberrypi:~/MySensors $
```

mfalkvidd, 2018-04-18:
Since there seems to be no real problem, I'm closing this. Feel free to reopen if there is something I've missed.

closed 2018-04-18 by mfalkvidd