mysensors/MySensors #1115: ESP8266/ESP32 locked on wifi connect

av1024, 2018-04-25:
If user pass invalid settings for WiFi, device will locked before `setup()` and never returns control to other code.

Some way to control over WiFi connection required. This may be at least a predefined timeout. A more complex way is external flag controlled by "hardware" timer and/or WiFi event handlers.

Both ESP8266 and ESP32 code in `gatewayTransportInit()` copied from trivial sample code
(latest development git, compiled with esp8266 2.3.0):

https://github.com/mysensors/MySensors/blob/e0988e6d9e368f1b619641c1e8e2328dcb809bf1/core/MyGatewayTransportEthernet.cpp#L133-L136

and

https://github.com/mysensors/MySensors/blob/e0988e6d9e368f1b619641c1e8e2328dcb809bf1/core/MyGatewayTransportEthernet.cpp#L153-L156

1. This loop will never ends if (for example) wrong password provided.
2. `wait` potentially may produce unexpected results because transport not ready here (see #946)
3. User code in `setup` and `loop` never be executed so no way to reconfig or restart device

gbrd, 2019-11-28:
It would be useful to improve that for wifi "node" (pseudo gateway with only node 0). 
I have some wall switches with local relay that will not work event locally if wifi is down.
(I cannot switch on the light even with "local"/main wall switch if my box is down)

gbrd, 2019-11-30:
For interested people, I found a workaround : 
on ESP8266 gateway: 
add ```#define MY_CORE_ONLY``` to avoid init of wifi "without control"
in setup initiate wifi "manually"

```
void setup() {
  Serial.begin(115200); // not done with MY_CORE_ONLY
  before(); // before is not called with MY_CORE_ONLY (but presentation() does after wifi successfull connect)
  Serial.println("setup");

  Serial.println("start wifi");
  WiFi.mode(WIFI_STA);
  WiFi.hostname(MY_HOSTNAME);
  WiFi.config(_ethernetGatewayIP, _gatewayIp, _subnetIp);
  WiFi.setAutoReconnect(false);
  WiFi.setAutoConnect(false);
  Serial.print("getAutoReconnect=");
  Serial.println(WiFi.getAutoReconnect());

(void)WiFi.begin(MY_WIFI_SSID, MY_WIFI_PASSWORD, 0, MY_WIFI_BSSID);

  Serial.println("setup END");
}
```
After that, if access point is down, we have only one loop() each 5 seconds (not found why). 
But if in that loop() call i call ```WiFi.disconnect(true);``` (true is important). 
It disconnects wifi and loop() is called normally. 

Thus it is possible to implement some watchdog that calls ```WiFi.disconnect(true);``` when there is a problem == when interval between 2 loops calls is too long (maybe around 50 ms) and local stuff cannot work. 
And retry to connect wifi periodically (with watchdog disabled for some seconds)

gbrd, 2019-11-30:
Would you be ready to accept a pull request for this "issue" ? (not sure I will have enough time, but just to know for now)