mysensors/MySensors #1122: Add a license file to github

flatsiedatsie, 2018-04-30:
With a license file it becomes easier to find and sort for projects with a certain open source license.

mandarcthorat, 2018-11-25:
As MySensors is an open source hardware and software community, it would be great if this code is released under permissive license(Ex. MIT or BSD3).

mfalkvidd, 2018-11-26:
MySensors current license text: 
https://www.mysensors.org/about/mysensors#copyright

flatsiedatsie, 2018-11-26:
@mfalkvidd Thanks, but this issue is about creating a license file in the Github code itself.

mfalkvidd, 2018-11-26:
@flatsiedatsie yes I am aware of that. But it seemed from mandarcthorat's comment that _which_ license to use was up for discussion.

mandarcthorat, 2018-11-27:
@mfalkvidd  Thanks for sharing the relevant link.