mysensors/MySensors #1128: Document board limitations for MY_RX_MESSAGE_BUFFER_FEATURE

mfalkvidd, 2018-05-11:
According to a note on https://github.com/mysensors/MySensors/blob/2994d83e53cda2867a80ff413575938ecd6d3ab2/drivers/RF24/RF24.cpp#L487 MY_RX_MESSAGE_BUFFER_FEATURE is "unsafe to use" on ESP8266. This should probably be included in the [Doxygen documentation](https://www.mysensors.org/apidocs-beta/group__RF24SettingGrpPub.html#gafe2f4565dfa8473846dfc82d417c8147).

Not sure if any other notes are needed. By reading the code, _my impression_ is that MY_RX_MESSAGE_BUFFER_FEATURE will work on Linux. There is no note about SAMD.

mfalkvidd, 2018-05-11:
Maybe also add a compilation error or warning

Yveaux, 2018-05-20:
The limitation for ESP is caused by incomplete support of SPI transactions in Arduino ESP port at the time of writing that comment line. We need to check for ESP (and SAMD, Linux,...) if SPI transactions are completely supported.

Yveaux, 2018-08-01:
The Arduino AVR core SPI library contains following remark: `SPI_HAS_TRANSACTION means SPI has beginTransaction(), endTransaction(), usingInterrupt(), and SPISetting(clock, bitOrder, dataMode)`
So checking for this define should be sufficient to detect if a certain architecture supports safe SPI access from interrupts.

However, studying the implementations of SPI libraries for various architectures reveals the following:

- Fully supported

  - 🥇 AVR: 1.6.21, ref https://github.com/arduino/ArduinoCore-avr/blob/3d09a51bd20f095e1354243e6d8addedfcb6001e/libraries/SPI/src/SPI.h#L21
  - 🥇 SAMD: 1.6.19, ref https://github.com/arduino/ArduinoCore-samd/blob/1.6.19/libraries/SPI/SPI.h#L30

- Defines SPI_HAS_TRANSACTION, but usingInterrupt() is stubbed

  - 👎 STM32: 1.3.0, ref: https://github.com/stm32duino/Arduino_Core_STM32/blob/1.3.0/libraries/SPI/src/SPI.h#L23
beginTransaction() and endTransaction() are not protected from recursive use
  - 👎 nRF52: 0.5.1, ref https://github.com/sandeepmistry/arduino-nRF5/blob/0.5.1/libraries/SPI/SPI.h#L31
beginTransaction() and endTransaction() are not protected from recursive use
  - 👍 Linux: MySensors 2.3.0, ref https://github.com/mysensors/MySensors/blob/2.3.0/drivers/Linux/SPIDEV.h#L190
beginTransaction() and endTransaction() use a mutex to protect from recursive use

- Defines SPI_HAS_TRANSACTION, but usingInterrupt() is not implemented
  - 👎👎 ESP8266: 2.4.1, ref https://github.com/esp8266/Arduino/blob/2.4.1/libraries/SPI/SPI.h#L27
This issue suggests otherwise: https://github.com/esp8266/Arduino/issues/1943

Summing up
- AVR, SAMD and probably Linux work out of the box.
- STM32 and nRF52 need protection in the MySensors stack. Simply disabling all interrupts could be an easy workaround until usingInterrupt() is correctly implemented.
- ESP8266 usingInterrupt() is not present, but same workaround as STM32 and nRF52 could also work here.

At the time of writing only AVR is known to work with MY_RX_MESSAGE_BUFFER_FEATURE. More testing is required.

Yveaux, 2018-08-01:
Regarding the compiler error remark, a compile error will be generated on ESP8266, https://github.com/mysensors/MySensors/blob/2.3.0/drivers/RF24/RF24.h#L130

Yveaux, 2018-08-01:
Arduino SPI access from interrupts on ESP8266 might become troublesome, see https://github.com/esp8266/Arduino/issues/3917

mfalkvidd, 2018-08-04:
I suggest we add a note in doxygen and link to this issue, as a start. I'll create a PR in the next few days.

mfalkvidd, 2019-06-03:
Reopening based on Yveaux's comment in https://github.com/mysensors/MySensors/pull/1296
I don't see a realistic way to handle that the information is outdated, but we should amend the documentation to something like "STM32, nRF52, ESP8266 and SoftSPI do not work."

closed 2019-07-13 by tekka007