mysensors/MySensors #1139: No minimal code to drive setup and loop in the example GatewaySerial.ino

tismoj, 2018-05-31:
I've noticed that the setup and loop functions doesn't have anything on it. I doubt that it is gonna run from just defines and includes. And How do I know that it is working? from the console?

Note, I am currently at https://github.com/tobof/openhab2-addons/wiki/Configuration, I have already installed the openHAB Add-on MySensors.

mfalkvidd, 2018-05-31:
presentation, loop and setup should be empty unless you want to attach local sensors (as mentioned in the comments in each function)

You can see what they gateway does by looking at the debug prints in the serial monitor.

tismoj, 2018-05-31:
Ah wow, actually I got it working before I had a chance to read your answer (I just followed through with this https://github.com/tobof/openhab2-addons/wiki/Configuration), surprising that there was a way to run some code outside of the setup and loop, so does that mean just by including MySensors.h header and the define MY_GATEWAY_SERIAL, the GatewaySerial's main code gets compiled and invoked in the MySensors library, certainly makes the .ino files simpler and more foolproof. Many thanks for the quick reply, honestly, timezone-wise, I thought you guys were already asleep, but I guess you never sleep, huh :)

mfalkvidd, 2018-05-31:
I'm in Sweden and it's another beautiful day here :)

Thanks for the feedback. I'll close this issue then.

closed 2018-05-31 by mfalkvidd