mysensors/MySensors #1150: OTA updates slow on RPI

Swiftnesses, 2018-06-16:
Good day,

After a full day of testing I can confirm that OTA node updates via RPI are extremely slow (1-2 stream messages per few seconds). Reverting back to either the serial or ethernet gateway resolves the issue (constant stream of messages).

All configs were the same, i.e version, channel, power etc etc.

mfalkvidd, 2018-06-16:
Could you share the configure command used for the gateway?

Swiftnesses, 2018-06-17:
./configure --my-transport=nrf24 --my-rf24-channel=125 --my-rf24-pa-level=RF24_PA_MAX --my-rf24-irq-pin=15 --extra-cxxflags="-DMY_OTA_LOG_RECEIVER_FEATURE"

Swiftnesses, 2018-06-18:
Guys, it's not just the Pi, it's all gateways - the issue coincided with me updating to 2.3.0 branch. Rolling back to 2.2.0 immediately fixes it. @tekka007

Swiftnesses, 2018-06-18:
Update: After another few hours sitting on the floor with wires going crazy, it seems it's related to this commit:

https://github.com/mysensors/MySensors/commit/2994d83e53cda2867a80ff413575938ecd6d3ab2

Replacing this file with the one from 2.2.0 causes ITA updates to resume full speed!

Swiftnesses, 2018-06-18:
Confirmed, with the file replaced, my Gateway (also tried other variants) is now working perfect!

Swiftnesses, 2018-06-19:
Another update - while replacing the file fixes the issue with the slow packets on the PA+LNA radio - it also seems to cause the gateway to fail after sometime - I assume that's because a simple file replace isn't the right way of fixing this, reverting back to 2.2.0 for now.

marceloaqno, 2018-06-20:
@Swiftnesses thanks for the detailed feedback.

Swiftnesses, 2018-09-10:
Any news on this guys?

tekka007, 2018-09-10:
@Swiftnesses Yes, #1189 fixes an issue with certain RF24 PA + LNA modules. Please try with the current dev branch

Swiftnesses, 2018-09-11:
@tekka007 Just tried it, seems the discover button in MYSController no longer works with the dev branch?

Swiftnesses, 2018-09-11:
Not working well for me at all, discover doesn't work and request presentation only provides version and one of the sensors (node has 3). Consistent behaviour.

tekka007, 2018-09-12:
I_DISCOVER messaging fixed in #1205

Swiftnesses, 2018-09-12:
@tekka007 That fixed it, a reboot of the nodes also fixed the presentation reporting issue!

closed 2018-09-15 by tekka007