mysensors/MySensors #1157: RFM69 gateway and default TX power

user2684, 2018-06-28:
If I'm understanding correctly, when MY_GATEWAY_FEATURE is defined, ATC is disabled which makes perfectly sense since every node will report a different RSSI and there is no need to change the tx level at every transmission. However, the default tx power will be used (MY_RFM69_TX_POWER_DBM) which is 5 and whereas ATC will increase that value if needed, this is not happening on the gateway which will stay at 5 forever despite not running on battery and so without any need to save power. Isn't this basically cutting at least by a half  the range of the entire network in a default configuration? Wouldn't be more convenient to keep the default to 20 if MY_GATEWAY_FEATURE is defined? I remember there was some beta version in the past with the tx level set to max to start with.
Thanks!

manutremo, 2018-12-04:
I realize this was asked 6 months ago but - I guess you can always use 

`#define MY_RFM69_TX_POWER_DBM (value)`

to overlay the default?

user2684, 2018-12-04:
Yes you can indeed but for users not knowing the library so in depth would be difficult to realize the entire radio network is running at a shorter range because of the default value set, if my understanding is correct of course, and they would not easily realize there is an easy way to increase the range at no cost, this was my original point

ericvb, 2020-12-10:
@mfalkvidd I know you can adapt the website articles ;-)

This is already an old topic, but I think this information is still valid with the MySensors library 2.3.2.
It should be mentioned as a remark in the article at https://www.mysensors.org/build/select_gateway.

I'm struggling now already for weeks to extend the range of my gateway using a RFM69HW board. My original gateway did have a simple helical antenna. I changed this antenna by a SMA external antenna with the hope to extend the range, but nada, no luck.

And now I'm reading that we can extend the range of the gateway by setting the parameter MY_RFM69_TX_POWER_DBM to a higher level?
Even for normal nodes, who have, by default, ATC enabled, we can extend the range by setting this parameter MY_RFM69_TX_POWER_DBM to a higher value then the default 5dBm. This value is used as the starting value of the ATC algorithm to set the used power level, if I interpret the MySensors library code correctly.

There is so much information in the forum topics, but because of the huge amount of topics, this information becomes untraceable.

mfalkvidd, 2020-12-10:
@ericvb yes I can. Just let me know what to adapt and where.

ericvb, 2020-12-12:
@mfalkvidd, okay, I propose to add a remark section to the article: https://www.mysensors.org/build/select_gateway

> For gateways using the radio transport RFM69 or RFM95, when MY_GATEWAY_FEATURE is defined, the ATC mode of the gateway is disabled which makes perfectly sense since every node will report a different RSSI and there is no need to change the tx level at every transmission. 
>
> However, the default tx power will then be used by the gateway (MY_RFM69_TX_POWER_DBM which is 5 and MY_RFM95_TX_POWER_DBM which is 13) and whereas ATC will increase that value if needed, this is not happening on the gateway which will stay at 5 for RFM69 (or 13 for RFM95) forever despite not running on battery and so without any need to save power. 
> 
> So if you are having range problems with your gateway, set the define MY_RFMxx_TX_POWER_DBM to a higher level.
> 
> Finally, pay attention to the hardware version of your radio.
> For example for RFM69, even though 20dBm may be specified, RFM69 is limited to +13dBm, only RFM69HW can reach real +20dBm. The opposite is also true, only RFM69 can use power values lower than -2dBm, RFM69HW is capped lower than that.

mfalkvidd, 2020-12-12:
@ericvb ok, I've added your note.

I am worried we might mislead people, because setting a higher tx level (=screaming louder) can make things worse. But I don't have enough experience to give conclusive information so I trust your judgement.

ericvb, 2020-12-14:
@mfalkvidd thanks for adding the note.
What you mention is correct, so I propose to add this also to the note.

> So if you are having range problems with your gateway, set the define MY_RFMxx_TX_POWER_DBM to a higher level.
> Be aware that setting this power setting to a higher level, will cause your node or gateway to 'screaming louder'. For nodes very close to the gateway, things could go even worse. So test with little steps.

mfalkvidd, 2020-12-15:
Thanks. I've updated.