mysensors/MySensors #1158: ESP8266 isn't connecting to SSID with hyphen in name

alc6379, 2018-07-02:
Just like the title says, I'm running MySensors 2.3.0. The SSID I'm trying to connect to (for example, let's call it TEST-ATT) has a hyphen in it. When I attempt to connect, I get: 

> no TEST found, reconnect after 1s
> reconnect

If I make an SSID that's just called TEST, it works just fine.

Hardware: NodeMCU ESP-12E

If there's anything else that you might need to help troubleshoot, let me know.

mfalkvidd, 2018-07-03:
@alc6379 could you post your #define MY_WIFI_SSID so we can reproduce?

alc6379, 2018-07-03:
sure thing, my actual line is 

`[code]
#define MY_WIFI_SSID "ROOSTER-ATT"

[/code]`

The actual error I get is 

> no ROOSTER found, reconnect after 1s
> reconnect

I'm using the GatewayESP8266MQTTClient example included with the library.

mfalkvidd, 2018-07-03:
Strange. I am not able to reproduce. What board version are you using?

My setup:
Based on GatewayESP8266MQTTClient
commented out #define MY_RADIO_NRF24
changed MY_WIFI_SSID:
`#define MY_WIFI_SSID "Mr-IoT.com"`
(and corresponding password)
changed ip address to controller:
`#define MY_CONTROLLER_IP_ADDRESS 192, 168, 1, 101`
Activated wifi debug from the Tools menu

ESP8266 board version: 2.4.1

Result:
```
SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
19662 MCO:BGN:INIT GW,CP=R-NGE---,VER=2.3.0
bcn 0
del if1
usl
mode : sta(18:fe:34:ca:de:c2)
add if0
wifi evt: 8
wifi evt: 2
20455 GWT:TPC:CONNECTING...
20984 GWT:TPC:CONNECTING...
21513 GWT:TPC:CONNECTING...
22042 GWT:TPC:CONNECTING...
22571 GWT:TPC:CONNECTING...
scandone
state: 0 -> 2 (b0)
23769 GWT:TPC:CONNECTING...
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with Mr-IoT.com, channel 3
dhcp client start...
wifi evt: 0
24447 GWT:TPC:CONNECTING...
24976 GWT:TPC:CONNECTING...
25505 GWT:TPC:CONNECTING...
26034 GWT:TPC:CONNECTING...
26563 GWT:TPC:CONNECTING...
ip:192.168.1.106,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 3
27092 GWT:TPC:CONNECTING...
27121 GWT:TPC:IP=192.168.1.106
27154 MCO:BGN:STP
27173 MCO:REG:NOT NEEDED
27199 MCO:BGN:INIT OK,TSP=NA
27229 GWT:TPC:IP=192.168.1.106
27262 GWT:RMQ:MQTT RECONNECT
27305 GWT:RMQ:MQTT CONNECTED
27335 GWT:TPS:TOPIC=mygateway1-out/0/255/0/0/17,MSG SENT
pm open,type:2 0

```

alc6379, 2018-07-03:
It's an ESP-12E NodeMCU Amica board. The chipset says

> Model ESP8266MOD
> Vendor AI-THINKER

I'm not sure if it's an actual one or a clone. I'll swap to another board I have and see if I'm getting similar results. The weird thing is that I'm pretty sure I've used this same board before with this SSID, and it's worked. I'm using the "NodeMCU 1.0(ESP-12E)" board in the Arduino IDE. I'll try switching to "Generic 8266 Module" and see if it makes any difference.

mfalkvidd, 2018-07-03:
Sorry. I meant the Arduino IDE ESP8266 board version, not the hardware version.
This is what my compilation output says when it starts building:
```
Using board 'd1_mini' from platform in folder: C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1
Using core 'esp8266' from platform in folder: C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1
```

alc6379, 2018-07-03:
OK, this is what I've got there: 

`Using board 'nodemcuv2' from platform in folder: /home/acavnar/.arduino15/packages/esp8266/hardware/esp8266/2.4.1
Using core 'esp8266' from platform in folder: /home/acavnar/.arduino15/packages/esp8266/hardware/esp8266/2.4.1`

mfalkvidd, 2018-07-03:
ok. That means we're using the same board version. I wonder what else could be different.

mfalkvidd, 2018-07-03:
I tried with nodemcuv2, still works
```
Using board 'nodemcuv2' from platform in folder: C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1
Using core 'esp8266' from platform in folder: C:\Users\Micke\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1
```

```
SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
5166 MCO:BGN:INIT GW,CP=R-NGE---,VER=2.3.0
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with Mr-IoT.com, channel 3
dhcp client start...
wifi evt: 0
ip:192.168.1.106,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 3
scandone
6037 GWT:TPC:CONNECTING...
6065 GWT:TPC:IP=192.168.1.106
6097 MCO:BGN:STP
6114 MCO:REG:NOT NEEDED
6140 MCO:BGN:INIT OK,TSP=NA
6169 GWT:TPC:IP=192.168.1.106
6201 GWT:RMQ:MQTT RECONNECT
6250 GWT:RMQ:MQTT CONNECTED
6279 GWT:TPS:TOPIC=mygateway1-out/0/255/0/0/17,MSG SENT
pm open,type:2 0
```

alc6379, 2018-07-03:
Odd. I'm going to try a different board, and now suddenly the RFM69 module I'm trying to use has quit working. I'll get those things sorted and report back. If it works with this other NodeMCU board, I'll just close this one out and write the issue off as a one-off with a weird board.

alc6379, 2018-07-03:
OK... so now it's REALLY odd. 

I didn't change anything in my code. I wiped out the entire memory on the original board I used, now everything is working. I'll go ahead and close this, since evidently it was a fluke.

closed 2018-07-03 by alc6379